espressif/arduino-esp32 #5613: Missing esp32_gpioMux values in esp32-hal-gpio.c for the ESP32-C3 target (r2.0.0-rc2)

chrisbabcock, 2021-08-30:
digitalPinIsValid() and digitalPinCanOutput() will fail for some pins (ex. GPIO3) due to missing initialization of esp32_gpioMux values for the CONFIG_IDF_TARGET_ESP32C3 in esp32-hal-gpio.c.

mehulsharma21, 2021-12-02:
@chrisbabcock 
Hi, Please copy the below table into file esp32-hal-gpio.c

const DRAM_ATTR esp32_gpioMux_t esp32_gpioMux[SOC_GPIO_PIN_COUNT]={
#if CONFIG_IDF_TARGET_ESP32C3
{0x04, -1, 0, -1}, // ADC1_CH0
{0x08, -1, 1, -1}, // ADC1_CH1
{0x0c, -1, 2, -1}, // ADC1_CH2 | FSPIQ
{0x10, -1, 3, -1}, // ADC1_CH3
{0x14, -1, 4, -1}, // MTMS | ADC1_CH4 | FSPIHD
{0x18, -1, 5, -1}, // MTDI | ADC2_CH0 | FSPIWP
{0x1c, -1, -1, -1}, // MTCK | FSPICLK
{0x20, -1, -1, -1}, // MTDO | FSPID
{0x24, -1, -1, -1}, //
{0x28, -1, -1, -1}, //
{0x2c, -1, -1, -1}, // FSPICSO
{0x30, -1, -1, -1}, //
{0x34, -1, -1, -1}, // SPIHD
{0x38, -1, -1, -1}, // SPIWP
{0x3c, -1, -1, -1}, // SPICSO
{0x40, -1, -1, -1}, // SPICLK
{0x44, -1, -1, -1}, // SPID
{0x48, -1, -1, -1}, // SPIQ
{0x4c, -1, -1, -1}, // USB-
{0x50, -1, -1, -1}, // USB+
{0x54, -1, -1, -1}, // U0RXD
{0x58, -1, -1, -1}, // U0TXD

VojtechBartoska, 2022-04-07:
@chrisbabcock is this still valid?

VojtechBartoska, 2022-04-22:
Hello, due to no answer I'm closing this issue as expired.

If you still need help, please reopen it.

Thanks for understanding!

closed 2022-04-22 by VojtechBartoska