espressif/arduino-esp32 #5618: Malformed board options in arduino-cli for atmegazero_esp32s2 and deneyapmini

manchoz, 2021-08-31:
### Hardware:
Board:							atmegazero_esp32s2, deneyapmini
Core Installation version:			2.0.0/2.0.0-rc2/Upstream
IDE name:						Arduino CLI
Flash Frequency:					N/A
PSRAM enabled:                                     N/A
Upload Speed:					N/A
Computer OS:                                         All the Arduino CLI supported platforms

### Description:

`boards.txt` has errors on `atmegazero_esp32s2` and `deneyapmini` board options.

`arduino-cli` output on any board-related operation, e.g. `arduino-cli board listall`:

```Error initializing instance: loading platform release esp32:esp32@2.0.0: loading boards: skipping loading of boards esp32:esp32:atmegazero_esp32s2, esp32:esp32:deneyapmini: malformed custom board options```

atanisoft, 2021-08-31:
@me-no-dev looks like both of these boards still reference `SerialMode` which should now be `CDCOnBoot`

manchoz, 2021-09-01:
Changing `SerialMode` to `CDCOnBoot` fixes the issue, thanks @atanisoft!

@me-no-dev let me know if you need a PR from me.

me-no-dev, 2021-09-01:
@manchoz please do send a PR :)

VojtechBartoska, 2022-03-30:
Hello @manchoz, is this still valid?

manchoz, 2022-04-01:
@VojtechBartoska no, it has been fixed in https://github.com/espressif/arduino-esp32/pull/5673

closed 2022-04-01 by manchoz