espressif/arduino-esp32 #5620: Serial trigger 1 byte is possivel????

tv4you2016, 2021-09-01:
### Hardware:
|||
|:---|---|
|<B>Board</B>|ESP32 Dev Module|
|<B>Version/Date</B>|2.0.0|
|<B>IDE name</B>|Arduino IDE|
|<B>Flash Frequency</B>|80Mhz|
|<B>PSRAM enabled</B>|no|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|

### Description:
Hey Good afternoon,

I have a doubt, and it is possible to make a trigger for each byte received by Serial.
I'm using the following code but it doesn't.

```cpp

#include <Arduino.h>

#define PIN_NOT_RE 18   // LOW = Recived
#define PIN_DE 19  // HIGH = SEND

void setup() {
  // put your setup code here, to run once:

  Serial.begin(14400);
  Serial.setRxBufferSize(1);
  Serial.println(Serial.getTimeout()); // print the default value
  Serial.setTimeout(1);
  Serial.println(Serial.getTimeout()); // print the new value
  pinMode(4, OUTPUT);
  pinMode(PIN_NOT_RE, OUTPUT);
  pinMode(PIN_DE, OUTPUT);

  delay(5000);

  digitalWrite(4, HIGH);

  digitalWrite(PIN_NOT_RE, LOW);
  digitalWrite(PIN_DE, LOW);
}

void loop() {

}

void serialEvent() {

  if (Serial.available()) {
    digitalWrite(4, LOW);
    // get the new byte:
    char inChar = (char)Serial.read();
    digitalWrite(4, HIGH);
  }

}

```
[image]

SuGlider, 2021-09-02:
@tv4you2016 ,

Please turn the debug level to verbose and check the messages, but be aware that some messages may be sent at  baud rate 115200 because  this is the default rate at boot time.

I can tell that ```Serial.setRxBufferSize(1);``` doesn't work given that the minimum RxBufferSize is 129 bytes.
RxBufferSize must be set before starting Serial port (Serial.begin(...))
Look at https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp#L275-L289

Regarding your specific question, I've tested it with this sketch, and it worked fine:

```
#include <Arduino.h>

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("Ready to receive bytes on UART.");
  Serial.println("It will output one byte per line with its HEX value.");
  Serial.flush(); 
}

void loop() {
}

void serialEvent() {
  if (Serial.available()) {
    char inChar = (char)Serial.read();
    Serial.print(inChar);
    Serial.print(" - 0x");
    Serial.println(inChar, HEX);
  }
}

```

rottaran, 2021-09-07:
`serialEvent` is just called inbetween loop(), see <https://github.com/espressif/arduino-esp32/blob/92ce408f4c71d235f0961a265e8d202f9fd570d5/cores/esp32/main.cpp#L47>. Neither interrupts nor full buffers involved.

SuGlider, 2021-09-08:
> `serialEvent` is just called inbetween loop(), see
> 
> https://github.com/espressif/arduino-esp32/blob/92ce408f4c71d235f0961a265e8d202f9fd570d5/cores/esp32/main.cpp#L47
> 
> . Neither interrupts nor full buffers involved.

Yes, this is correct. Arduino has no API to trigger UART events. 
All it has is the serailEvent called in between loop() - but this is exactly the Arduino specification.
https://www.arduino.cc/en/Tutorial/BuiltInExamples/SerialEvent

SuGlider, 2022-01-12:
A potential solution is being implemented.
It's the ```Serial.onReceive(function)```

As soon as any data arrives UART RX, ```function()``` will called.

closed 2022-01-17 by me-no-dev