espressif/arduino-esp32 #5621: MQTT Support in arduino-esp32

econeale, 2021-09-01:
### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|Adafruit Huzzah 32|
|<B>Version/Date</B>|today's master|
|<B>IDE name</B>|Arduino IDE|
|<B>Flash Frequency</B>|40Mhz|
|<B>PSRAM enabled</B>|no|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Mac OSX|

### Description:
I would like to use the ESP-IDF MQTT component in my Arduino IDE based project. Are there plans for adding the MQTT component to the arduino-esp32 project? 

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
N/A

### Debug Messages:
N/A

me-no-dev, 2021-09-01:
https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/esp32/include/mqtt/esp-mqtt/include
Did you try to use it?

econeale, 2021-09-01:
Womp womp. I looked through the whole SDK three times and missed that. I'll take a look right now! Thanks!

torntrousers, 2021-09-02:
if you get something working using the IDF MQTT would you share you code? TIA!

econeale, 2021-09-02:
@torntrousers Sure! I'm halfway, give or take, through transitioning from PubSubClient. I'll post some snippets when I'm done.

wladwnt, 2021-09-02:
I have ESP-IDF MQTT working with ESP32- Arduino 1.0.6 (very stable compared to many Arduino MQTT Libs I tried), but the same code does not work with 2.0.0. Apparently some breaking changes from IDF 3.3 to IDF 4.4.

torntrousers, 2021-09-02:
Share the code you have @wladwnt ?

AlexBorn, 2021-09-02:
Nothing special, based on ESP-IDF MQTT over TCP example.

Call  mqtt_setup() in your setup() function. Then call esp_mqtt_client_publish(mqtt_client, topic, payload, 0, 0, 0)each time you need to publlish some topics.

In the following code **settings** structure is what I hav ein my program to store all settings, you have to replace them according to your needs:

```c++
// include ESP-IDF MQTT Client Lib
#include "mqtt_client.h"

//declare MQTT configstructure, will be initialised in mqtt_setup() function
static esp_mqtt_client_config_t mqtt_cfg;

// this will be called if MQTT is connected, here we have to subscribe to topics we need. Change it acccording to your needs
void onMqttConnect(esp_mqtt_client_handle_t client) {
  // subscribe to topics required
  char *topic=(char *)malloc(MAX_MQTT_TOPIC_LEN+4);
  if(topic!=nullptr) {
      for(int i=0;i<NUMBER_DATAPOINTS;i++) {
          strcpy(topic,settings.mqtt_topicprefix);
          strcat(topic,"/");
          strcat(topic,settings.param_mqtttopic[i]);
          strcat(topic,"/set");
          esp_mqtt_client_subscribe(client, topic, 0);
        }
  free(topic);
 }
}

// this is event handler that we will register in mqtt_setup(). You will need at least two events: MQTT_EVENT_CONNECTED and MQTT_EVENT_DATA. more possible events see ESP-IDF example
static esp_err_t mqtt_event_handler(esp_mqtt_event_handle_t event)
{
    esp_mqtt_client_handle_t mqttclient = event->client;
    int msg_id;
    switch (event->event_id) {
        case MQTT_EVENT_CONNECTED:
            onMqttConnect(mqttclient);
            break;
        case MQTT_EVENT_DATA:
            // Write your own onMqttMessage function that react on topics received
            onMqttMessage(event->topic, event->topic_len, (const unsigned char *) event->data, event->data_len, 0, 0, 0);
            break;
        default:
            break;
    }
    return ESP_OK;
}

void mqtt_setup(){
 // init config structure
    mqtt_cfg.event_handle = mqtt_event_handler;
    mqtt_cfg.host=(const char *)settings.mqtt_server_ip;
    char *uri_line=(char *)malloc(100);
    sprintf(uri_line,"mqtt://%s:%u",settings.mqtt_server_ip,settings.mqtt_port);
    mqtt_cfg.uri=uri_line;
    mqtt_cfg.port=settings.mqtt_port;
    mqtt_cfg.username=(const char *)settings.mqtt_user;
    mqtt_cfg.password=(const char *)settings.mqtt_pwd;
 // setup MQTT
    mqtt_client = esp_mqtt_client_init(&mqtt_cfg);
}
```

wladwnt, 2021-09-02:
@AlexBorn : Thank you for sharing the code! I have also very simple implementation like you, but It does not work with Arduino-ESP32 2.0.0 directly. Compilation result is OK. I have not checked what is the reason, maybe other stuff in my quite complex full programm influences it. Let me know if you try 2.0.0. Thanks!

torntrousers, 2021-09-04:
Thanks for that code @AlexBorn . From that I've come up with this code which is complete to connect to the public Mosquitto test MQTT server, so any of you could run it. It compiles but doesn't work and never receives the MQTT_EVENT_CONNECTED event. Any help to get it to work?

```
#include "Arduino.h"
#include "mqtt_client.h"

// Test Mosquitto server, see: https://test.mosquitto.org
char * mqtt_server_ip = "5.196.95.208";
uint32_t mqtt_port = 1883;

// test Mosquitto doesn;t need a user/pwd
char * mqtt_user = "";
char * mqtt_pwd = "";

char * subscribeTopic = "foo";

esp_mqtt_client_handle_t mqtt_client;
static esp_mqtt_client_config_t mqtt_cfg;

void setup() {
  Serial.begin(115200); Serial.println();
  Serial.print("setup, ESP.getSdkVersion(): "); Serial.println(ESP.getSdkVersion());

  mqtt_setup();
}

void loop() {
  Serial.println("loop");
  delay(2000);
}

void onMqttConnect(esp_mqtt_client_handle_t client) {
    esp_mqtt_client_subscribe(client, subscribeTopic, 0);
    Serial.print("subscribed to: "); Serial.println(subscribeTopic);
}

static esp_err_t mqtt_event_handler(esp_mqtt_event_handle_t event) {
    Serial.print("mqtt_event_handler"); 

    esp_mqtt_client_handle_t mqttclient = event->client;
    int msg_id;
    switch (event->event_id) {
        case MQTT_EVENT_CONNECTED:
            Serial.print("mqtt_event_handler: MQTT_EVENT_CONNECTED"); 
            onMqttConnect(mqttclient);
            break;
        case MQTT_EVENT_DATA:
            Serial.print("mqtt_event_handler: MQTT_EVENT_DATA"); 
            break;
        default:
            break;
    }
    return ESP_OK;
}

void mqtt_setup(){
    Serial.println("mqtt_setup");
    mqtt_cfg.event_handle = mqtt_event_handler;
    mqtt_cfg.host=(const char *)mqtt_server_ip;
    char *uri_line=(char *)malloc(100);
    sprintf(uri_line,"mqtt://%s:%u",mqtt_server_ip,mqtt_port);

    Serial.print("mqtt_setup: "); Serial.println(uri_line);

    mqtt_cfg.uri=uri_line;
    mqtt_cfg.port=mqtt_port;
    mqtt_cfg.username=(const char *)mqtt_user;
    mqtt_cfg.password=(const char *)mqtt_pwd;

    mqtt_client = esp_mqtt_client_init(&mqtt_cfg);
}
```

torntrousers, 2021-09-04:
Was missing esp_mqtt_client_start. This code works for me with ESP32/Arduino 1.0.6 and 2.0.0:
```
#include "Arduino.h"
#include <WiFi.h>
#include "mqtt_client.h"

char ssid[] = "<yourWifiSSID>";
char pass[] = "<yourWifiPswd>";

// Test Mosquitto server, see: https://test.mosquitto.org
char * server = "mqtt://test.mosquitto.org:1883";

char * subscribeTopic = "foo";
char * publishTopic = "baa";

esp_mqtt_client_config_t mqtt_cfg;
esp_mqtt_client_handle_t mqtt_client;

void setup() {
  Serial.begin(115200); Serial.println();
  Serial.print("setup, ESP.getSdkVersion(): "); Serial.println(ESP.getSdkVersion());

  initWifi();
  mqtt_setup();
}

int pubCount = 0;

void loop() {
  String msg = "Hello: " + String(pubCount++);
  esp_mqtt_client_publish(mqtt_client, publishTopic, msg.c_str(), 0, 0, 0);
  Serial.printf("published to topic %s: '%s'\n", publishTopic, msg);

  delay(2000);
}

void onMqttConnect(esp_mqtt_client_handle_t client) {
    Serial.println("onMqttConnect");
    esp_mqtt_client_subscribe(client, subscribeTopic, 0);
    Serial.print("subscribed to: "); Serial.println(subscribeTopic);
}

void onMqttData(esp_mqtt_event_handle_t event) {
    String topic = String(event->topic).substring(0, event->topic_len);
    String msg = String(event->data).substring(0, event->data_len);
    Serial.printf("onMqttData topic '%s' msg '%s'\n", topic, msg);
}

static esp_err_t mqtt_event_handler(esp_mqtt_event_handle_t event) {
    Serial.println("mqtt_event_handler"); 

    esp_mqtt_client_handle_t mqttclient = event->client;
    int msg_id;
    switch (event->event_id) {
        case MQTT_EVENT_CONNECTED:
            onMqttConnect(mqttclient);
            break;
        case MQTT_EVENT_DATA:
            onMqttData(event);
            break;
        default:
            break;
    }
    return ESP_OK;
}

void mqtt_setup(){
    Serial.print("mqtt_setup: "); Serial.println(server);

    mqtt_cfg.event_handle = mqtt_event_handler;
    mqtt_cfg.uri=server;

// test Mosquitto doesn't need a user/pwd
//    // mqtt_cfg.username=(const char *)mqtt_user;
//    // mqtt_cfg.password=(const char *)mqtt_pwd;

    mqtt_client = esp_mqtt_client_init(&mqtt_cfg);
    esp_mqtt_client_start(mqtt_client);
}

void initWifi() {
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  Serial.print("Connecting to "); Serial.print(ssid);
  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED) {
     delay(250);
     Serial.print(".");
  }  
  Serial.print("\nWiFi connected, IP address: "); Serial.println(WiFi.localIP());
}
```

zekageri, 2021-09-10:
Does it have secure support?

econeale, 2021-09-10:
The configuration object has an option for PEM certificates. That seems to work for ThingsBoard at least. 

https://github.com/espressif/arduino-esp32/blob/92ce408f4c71d235f0961a265e8d202f9fd570d5/tools/sdk/esp32/include/mqtt/esp-mqtt/include/mqtt_client.h#L186

zekageri, 2021-09-10:
Oh i see.  
It does have `const char *cert_pem;` and const `char *client_cert_pem;`
How could one generate a **pem** file and use it?

I can see that i could add the pem file to `esp_mqtt_client_config_t`  like

```C
esp_mqtt_client_config_t mqtt_cfg;
mqtt_cfg.cert_pem = myPemFile;
mqtt_cfg.client_cert_pem = myClientPemFile;
```

econeale, 2021-09-10:
I have my certs. as character array: 

const char* myPemFile = ""-----BEGIN CERTIFICATE-----\n" \
"MIIFazCCA1OgAwIBAgIRAIIQz7DSQONZRGPgu2OCi1CLQJ13hef4Y53CI\n" \
........
"-----END CERTIFICATE-----\n";

ThingsBoard, at least, has a walkthrough on how to generate the required certificates here: https://thingsboard.io/docs/user-guide/certificates/

lbernstone, 2021-09-10:
I'm a big fan of using string literals for certificates, since then you can just copy and paste:
```
const char* pemFile = R"literal(
-----BEGIN CERTIFICATE-----
...
-----END CERTIFICATE-----)literal";
```
You can also see an example of how to load certs from a webpage/NVS in my [example](https://github.com/lbernstone/setCerts)

closed 2021-09-14 by econeale