espressif/arduino-esp32 #5622: esp32-s2 chip using Preferences library to store string  but lost after the restart

Kevin-CEN, 2021-09-02:
I wanted to use the Preferences library to store strings, using the ESP32-S2 chip, and using the official example StartCounter. ino to test, I compiled and burned without any problems, but when ESP32-S2 ran the program, the numbers didn't add up so should have been lost after the restart.
This is the output of the MCU:
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026168
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x424
load:0x4004c000,len:0x844
load:0x40050000,len:0x2460
entry 0x4004c180

Current counter value: 1
Restarting in 10 seconds...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026168
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x424
load:0x4004c000,len:0x844
load:0x40050000,len:0x2460
entry 0x4004c180

Current counter value: 1
Restarting in 10 seconds...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026168
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x424
load:0x4004c000,len:0x844
load:0x40050000,len:0x2460
entry 0x4004c180

Current counter value: 1
Restarting in 10 seconds...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026168
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x424
load:0x4004c000,len:0x844
load:0x40050000,len:0x2460
entry 0x4004c180

Current counter value: 1
Restarting in 10 seconds...
############################
This is my platformio.ini configuration：
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
board_build.mcu = esp32s2
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#2.0.0-alpha1
############################

atanisoft, 2021-09-02:
@Kevin-CEN I think this may have been fixed by https://github.com/espressif/arduino-esp32/pull/5309 which should be available in the 2.0.0 release. Can you retest with the 2.0.0 release?

Kevin-CEN, 2021-09-02:
> @Kevin-CEN I think this may have been fixed by #5309 which should be available in the 2.0.0 release. Can you retest with the 2.0.0 release?
Thanks, but I'm not quite sure how to get back to version 2.0.0.
Esp32-s2 doesn't have very good Arduino support.

atanisoft, 2021-09-02:
@Kevin-CEN Adjust your platformio.ini to have:
```
framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#2.0.0
```
Most of the ESP32-S2 peripherals should have compatible Arduino code already, if you find an exception to this though do log an issue for tracking/consideration.

Kevin-CEN, 2021-09-02:
@atanisoft 
Thank you very much, but I don't seem to be able to download version 2.0.0
``` ====================[ Build | Production | esp32dev ]===========================
/Applications/CLion.app/Contents/bin/cmake/mac/bin/cmake --build /Users/kevin/CLionProjects/esp32dev-s2test/cmake-build-esp32dev --target Production -- -j 3
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)

Tool Manager: Installing git+https://github.com/espressif/arduino-esp32#2.0.0
git version 2.32.0
正克隆到 '/Users/kevin/.platformio/.cache/tmp/pkg-installing-hcsdb1p7'...
fatal: 无法访问 'https://github.com/espressif/arduino-esp32/'：LibreSSL SSL_connect: SSL_ERROR_SYSCALL in connection to github.com:443 
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', '2.0.0', 'https://github.com/espressif/arduino-esp32', '/Users/kevin/.platformio/.cache/tmp/pkg-installing-hcsdb1p7']
make[3]: *** [CMakeFiles/Production] Error 1
make[2]: *** [CMakeFiles/Production.dir/all] Error 2
make[1]: *** [CMakeFiles/Production.dir/rule] Error 2
make: *** [Production] Error 2  ```

Kevin-CEN, 2021-09-02:
@atanisoft 
Thank you very much, I should be using version 2.0.0, but I added '-alpha1', if not '-alpha1' will have this error
``` /Users/kevin/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1

Kevin-CEN, 2021-09-02:
@atanisoft 
Thank you very much, I should be using version 2.0.0, but I added '-alpha1', if not '-alpha1' will have this error
``` /Users/kevin/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1 ```

Kevin-CEN, 2021-09-02:
@atanisoft 
Thank you very much, I should be using version 2.0.0, but I added '-alpha1', if not '-alpha1' will have this error
```Linking .pio/build/esp32dev/firmware.elf
/Users/kevin/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1

atanisoft, 2021-09-02:
@Kevin-CEN That link error looks related to PIO breaking the builds, I guess you will need to wait for the 2.0.0 release to be picked up by PIO instead. Usually they will pick up a new stable release within a few days of release.

Jason2866, 2021-09-02:
Mhh, i get this
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3ffe6100,len:0x7c
load:0x4004c000,len:0x7e4
load:0x40050000,len:0x2598
entry 0x4004c15c

Current counter value: 7
Restarting in 10 seconds...

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400260ec
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3ffe6100,len:0x7c
load:0x4004c000,len:0x7e4
load:0x40050000,len:0x2598
entry 0x4004c15c

Current counter value: 8
Restarting in 10 seconds...

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400260ec
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3ffe6100,len:0x7c
load:0x4004c000,len:0x7e4
load:0x40050000,len:0x2598
entry 0x4004c15c

Current counter value: 9
Restarting in 10 seconds...

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400260ec
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3ffe6100,len:0x7c
load:0x4004c000,len:0x7e4
load:0x40050000,len:0x2598
entry 0x4004c15c

Current counter value: 10
```
Compiled with Platformio

Kevin-CEN, 2021-09-03:
@Jason2866 Yours seems to be all right.Can I have a look at your platformio.ini file？

Jason2866, 2021-09-03:
@Kevin-CEN, reused the setup from the Project Tasmota (i am a member).
The relevant part is this https://github.com/arendst/Tasmota/blob/development/platformio_tasmota_cenv_sample.ini#L47-L60

Kevin-CEN, 2021-09-03:
@Jason2866 Thank you very much.But this configuration file of yours seems too complicated, I just tried it, it will compile wrong.
My project is very simple, I just hope THAT ESP32S2 can normally use the Preferences library in the Arduino framework, is there any other way I can set it up?

Jason2866, 2021-09-03:
@Kevin-CEN  Okay, simplified :-)
```
[env:esp32-s2-saola-1]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
platform_packages = framework-arduinoespressif32 @ https://github.com/tasmota/arduino-esp32/releases/download/2.0.0/framework-arduinoespressif32-release_IDF4.4.tar.gz
board = esp32-s2-saola-1
framework = arduino
```

yangminglong, 2021-12-29:
Can you explain what went wrong with platform IO?

Jason2866, 2021-12-29:
Using a old setup/version of Arduino results in non working sketch for S2.

yangminglong, 2021-12-29:
> Using a old setup/version of Arduino results in non working sketch for S2.

Thank you very much. According to your instructions, I uninstalled platformIO from vscode and deleted .platformio folder. After that, I reinstalled platformIO in vscode. The following configuration is used to solve the problem that preferences cannot be saved.

`[env:esp32s2]`
`platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2idf/platform-espressif32-2.0.2.zip`
`board = esp32-s2-saola-1`
`framework = arduino`

VojtechBartoska, 2022-01-04:
@Kevin-CEN Can I consider this as solved?

closed 2022-04-07 by VojtechBartoska