espressif/arduino-esp32 #5625: A2DP sink missing samples

atecroby, 2021-09-02:
I'm using 2.0.0 (so IDF4.4) and it seems that A2DP sink still has an old problem: some missing samples on audio streaming (with a sinewave is very noticeable).

With IDF4.3 beta 2 (and above) the audio problem was not present.

I think it's the bug marked as "A2DP: fixed data stream underrun for A2DP sink".

me-no-dev, 2021-09-03:
Since A2DP is part of ESP-IDF, we can not fix it here. You can give the `idf-master` branch of this repo a try and see if it's any better (the branch contains an updated ESP-IDF libs that could have the fix inside)

atecroby, 2021-09-03:
Thank you @me-no-dev and sorry for wrong place.
I just give it a try with branch `idf-master` and the problem still here.

I'll try to rebuild with Arduino Lib Builder!

closed 2021-09-03 by atecroby