espressif/arduino-esp32 #5626: ESP32 ARDUINO BLE getValue function cannot read 0x00 binary from app

maozhifeng, 2021-09-02:
The ESP32 ARDUINO BLE getValue function cannot read the 0x00 data from the app.
For example, the mobile app sends: 0x01 0x02 0x00 0x03 0x05;
The esp32 getValue() function can only return 0x01 0x02 and the data after 0x00 and 0x00 will not be read. How can I solve this problem?

The mobile app I use is nRFconnect

chegewara, 2021-09-02:
It can read, if you know how, but to make yourlife easier there is another function:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLECharacteristic.h#L64

maozhifeng, 2021-09-02:
> 它可以读取，如果你知道如何，但为了让你的生活更轻松，还有另一个功能：[https](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLECharacteristic.h#L64) :
> [//github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLECharacteristic.h#L64](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLECharacteristic.h#L64)

> 它可以读取，如果你知道如何，但为了让你的生活更轻松，还有另一个功能：[https](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLECharacteristic.h#L64) :
> [//github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLECharacteristic.h#L64](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLECharacteristic.h#L64)

I have tried the getData function and still cannot return 0x00. It can only return the data in ascii code. Can you clearly tell me how to change it?

this is my test code:

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
        ble_rxValue = "";
        ble_rxValue = pCharacteristic->getValue().c_str();
        ble_data = pCharacteristic->getData();
        pCharacteristic->
        if (ble_rxValue.length() > 0) {
            for(int i=0;i<ble_rxValue.length();i++)
            {
                Serial.print((*(ble_data++)));
            }
            ble_get_sta = 1;
        }
    }
};

maozhifeng, 2021-09-02:
`class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
        unsigned char *ble_data = NULL;
        ble_rxValue = "";
        ble_rxValue = pCharacteristic->getValue().c_str();
        ble_data = pCharacteristic->getData();
        pCharacteristic->
        if (ble_rxValue.length() > 0) {
            for(int i=0;i<ble_rxValue.length();i++)
            {
                Serial.print((*(ble_data++)));
            }
        }
    }
};`

chegewara, 2021-09-02:
Its not the problem with ble library, just you dont understand what will happen when you do that:
`Serial.print(0x0);`

maozhifeng, 2021-09-02:
> 这不是 ble 库的问题，只是您不明白这样做时会发生什么：
> `Serial.print(0x0);`

The serial port still can't get the 0x00 I want:
Changed to Serial.write(0x00);

`class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
        ble_rxValue = "";
        ble_rxValue = pCharacteristic->getValue().c_str();
        ble_data = pCharacteristic->getData();
        if (ble_rxValue.length() > 0) {
            for(int i=0;i<ble_rxValue.length();i++)
            {
                Serial.write((byte)(*(ble_data++)));
            }
        }
    }
};`

maozhifeng, 2021-09-02:
> 我假设`Serial`是输出日志。您希望通过发送空字符串在日志中看到什么？

I can see the return value in the serial debugging tool, so that I can control the esp32 through commands. I need to transmit data through the high and low bits, so it is inevitable to use the 0x00 data

I sent 0x00 through the nrfconnetc software, and didn’t make the callback function work. I don’t know if it’s the phone software problem or esp32 directly filtered out 0x00.

chegewara, 2021-09-02:
I can repeat it again and again, there is no problem to send any value with BLE, you can send as many `0x0` as you want. The problem is what you will do with data when you receive it. Maybe try instead of `Serial.print` another function like `Serial.printf("0x%02x", *ble_data);`

maozhifeng, 2021-09-02:
> 我可以一次又一次地重复，用BLE发送任何值都没有问题，您可以发送任意数量的`0x0`。问题是当你收到数据时你将如何处理它。也许尝试而不是`Serial.print`另一个功能，如`Serial.printf("0x%02x", *ble_data);`

感谢您耐心的解答我的问题，不过我现在的问题并不在于esp32 BLE发送数据，而是BLE接收来自手机app的数据。

例如手机app发送给BLE数据为：0x01 0x02 0x00 0x03 0x05；
esp32 BLE的getData()函数只能收到0x01 0x02，而0x00和0x00之后的数据将不会被读取

maozhifeng, 2021-09-02:
> I can repeat it again and again, there is no problem to send any value with BLE, you can send as many `0x0` as you want. The problem is what you will do with data when you receive it. Maybe try instead of `Serial.print` another function like `Serial.printf("0x%02x", *ble_data);`

Thank you for your patience to answer my question, but my current problem is not that esp32 BLE sends data, but that BLE receives data from the mobile app.

For example, the data sent by the mobile app to BLE is: 0x01 0x02 0x00 0x03 0x05;
The getData() function of esp32 BLE can only receive 0x01 0x02, and the data after 0x00 and 0x00 will not be read

maozhifeng, 2021-09-02:
> > 我可以一次又一次地重复，用BLE发送任何值都没有问题，您可以发送任意数量的`0x0`。问题是当你收到数据时你将如何处理它。也许尝试而不是`Serial.print`另一个功能，如`Serial.printf("0x%02x", *ble_data);`
> 
> 感谢您耐心回答我的问题，但我目前的问题不是 esp32 BLE 发送数据，而是 BLE 从移动应用程序接收数据。
> 
> 比如手机app发送给BLE的数据是：0x01 0x02 0x00 0x03 0x05；
> esp32 BLE的getData()函数只能接收0x01 0x02，不会读取0x00和0x00之后的数据

The mobile app is a dedicated Bluetooth debugging software

chegewara, 2021-09-02:
> my current problem is not that esp32 BLE sends data, but that BLE receives data from the mobile app.

I understand:
```
class MyCallbacks: public BLECharacteristicCallbacks {
void onWrite(BLECharacteristic pCharacteristic) {
ble_rxValue = "";
ble_rxValue = pCharacteristic->getValue().c_str();
ble_data = pCharacteristic->getData();
pCharacteristic->
if (ble_rxValue.length() > 0) {
for(int i=0;i<ble_rxValue.length();i++)
{
Serial.printf("0x%02x", *(ble_data++));
}
ble_get_sta = 1;
}
}
};
```

maozhifeng, 2021-09-03:
> > 我目前的问题不是 esp32 BLE 发送数据，而是 BLE 从移动应用程序接收数据。
> 
> 我明白：
> 
> ```
> class MyCallbacks: public BLECharacteristicCallbacks {
> void onWrite(BLECharacteristic pCharacteristic) {
> ble_rxValue = "";
> ble_rxValue = pCharacteristic->getValue().c_str();
> ble_data = pCharacteristic->getData();
> pCharacteristic->
> if (ble_rxValue.length() > 0) {
> for(int i=0;i<ble_rxValue.length();i++)
> {
> Serial.printf("0x%02x", *(ble_data++));
> }
> ble_get_sta = 1;
> }
> }
> };
> ```

Thank you for your answer. The problem has been solved. The reason is that I used the c_str() function when I tried to convert the std::string type to the String type. This function would return C-style data, thus thinking that 0x00 was the terminator. I cannot receive data after 0x00. Thank you very much for your answer

closed 2021-09-03 by VojtechBartoska