espressif/arduino-esp32 #5627: ESP32 C3 does not seem to be able to maintain a long and stable BLE connection

Danjuanlab, 2021-09-02:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
I (48) boot: ESP-IDF v4.4-dev-2313-gc69f0ec32 2nd stage bootloader
I (49) boot: compile time 12:10:14
I (49) boot: chip revision: 3
I (49) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (55) qio_mode: Enabling default flash chip QIO
I (59) boot.esp32c3: SPI Speed      : 80MHz
I (63) boot.esp32c3: SPI Mode       : QIO
I (67) boot.esp32c3: SPI Flash Size : 4MB
I (71) boot: Enabling RNG early entropy source...
I (75) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (91) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (97) boot:  2 app0             OTA app          00 10 00010000 00140000
I (104) boot:  3 app1             OTA app          00 11 00150000 00140000
I (110) boot:  4 spiffs           Unknown data     01 82 00290000 00170000
I (117) boot: End of partition table
I (120) boot_comm: chip revision: 3, min. application chip revision: 0
I (126) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=1e880h (125056) map
I (152) esp_image: segment 1: paddr=0002e8a8 vaddr=3fc8de00 size=01770h (  6000) load
I (153) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=9f9f4h (653812) map

[22:42:27.043]收←◆I (247) esp_image: segment 3: paddr=000cfa1c vaddr=3fc8f570 size=00c08h (  3080) load
I (248) esp_image: segment 4: paddr=000d062c vaddr=40380000 size=0dda8h ( 56744) load
I (261) esp_image: segment 5: paddr=000de3dc vaddr=50000000 size=00010h (    16) load
I (265) boot: Loaded app from partition at offset 0x10000
I (265) boot: Disabling RNG early entropy source...
[   287][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz

Computer OS:                                        Windows 10

### Description:
I use the NimBLE library to build a keyboard demo, and use multiple tasks to control the LED lights. After writing the program below, all functions can run normally for a period of time, but after a period of standby (about 40-50 Minutes), when you click the button, the computer will not receive the simulated keyboard information. I found through debug log that ble sent a blank message to the computer, which is different from the ble log of the previous text message. At this point, when you disconnect the esp32 ble keyboard on the computer, the LED status on the development board is still ble connected, and it does not jump out of the judgment of "bleKeyboard.isConnected()". I have to restart the development board so that it can be recognized as disconnected again.

I use a regular ESP32 development board to run this program, but it can run normally for at least a few hours.

### Sketch:  
```cpp

#include <RGBLed.h>
#include <Ticker.h>
#include <OneButton.h>
#include <BleKeyboard.h>
BleKeyboard bleKeyboard;

#define fnPin 1     //funtion key
#define footPin 2  //foot pin

#define led_R 3
#define led_G 4
#define led_B 5

int batteryLevel = 100; //电池电量
bool isCharging = false; //当前充电中？
bool fakecharing = true;   //假充电。
[... 970 lines cut ...]
E (3052401) task_wdt: CPU 0: 
[22:32:23.926]收←◆loopTask

[22:32:28.887]收←◆E (3057393) task_wdt: Task watch
[22:32:28.910]收←◆dog got triggered. The following tasks did not reset the watchdog in time:
E (3057393) task_wdt:  - IDLE (CPU 0)
E (3057393) task_wdt: Tasks currently running:
E (3057393) task_wdt: CPU 0: led State

[22:32:33.888]收←◆E (3062385) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (3062385) task_wdt:  - IDLE (CPU 0)
E (3062385) task_wdt: Tasks currently running:
E (3062385) task_wdt: CPU 0: led State

[22:34:23.886]收←◆E (3172209) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (3172209) task_wdt:  - IDLE (CPU 0)
E (3172209) task_wdt: Tasks currently running:
E (3172209) task_wdt: CPU 0: loopTask

[22:34:28.886]收←◆E (3177201) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (3177201) task_wdt:  - IDLE (CPU 0)
E (3177201) task_wdt: Tasks currently running:
E (3177201) task_wdt: CPU 0: led State

```

chegewara, 2021-09-02:
Sorry, i dont see where it disconnect, but i see other thing. Did you try to fix this problem:
```
E (3002481) task_wdt:  - IDLE (CPU 0)
E (3002481) task_wdt: Tasks currently running:
E (3002481) task_wdt: CPU 0: led State
```

Danjuanlab, 2021-09-03:
Yes, from the log and the LED connection status program I wrote, the return value of bleKeyboard.isConnected() is always true, but at this time, if the keyboard signal is trigged by  button, there will be no ble setvalue() in the debug content. Then in this state, even if I disconnect and delete esp32 blekeyboard on the computer, the return value of bleKeyboard.isConnected() is still true.
I can't find any clues to solve this problem.

closed 2022-04-20 by VojtechBartoska