espressif/arduino-esp32 #5628: arduino-esp32 v2.0.0 & Arduino IDE 1.8.15

Xylopyrographer, 2021-09-03:
@ lbernstone
Will the latest v 2.0.0 arduino-esp32 install under Arduino IDE v1.8.15?

Launching Arduino IDE v1.8.15 and going to the Boards Manager does not show the v2.0.0 release.

Launching Arduino IDE  2.0.0-beta.11 and going to the Boards Manager, the v 2.0.0 release is there.

Tweak needed to the .json file?

Would rather use the stable 1.8.15 IDE than the beta to test run the just released v2.0.0 of arduino-esp32.

Thanks.

VojtechBartoska, 2021-09-03:
Hi @Xylopyrographer,

I tried to install 2.0.0 version in Arduino IDE 1.8.15 and it's working just fine. Can you share more info what's your problem?

To set it up, you can check this [tutorial](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html).

Xylopyrographer, 2021-09-03:
Hi Vojtěch:

I have any existing Arduino IDE, v 1.8.15 install on macOS 11.5.2. Selecting Tools—>Board—>Board Manager and filtering for esp I get:

As you can see, core v 2.0.0 does not show up.

If I launch Arduino IDE Version: 2.0.0-beta.11 on the same machine, and do the same Tools—>Board—>Board Manager and filtering for esp I get:

Where core v2.0.0 appears.

Should be a way to see/force core v2 under IDE v1.8.15?

Best regards,
    Xylo
===================

SuGlider, 2021-09-03:
One work around is to first uninstall (delete) ESP preveious version on Arduino Board Manager and then install 2.0.0..
I did that  for IDE 1.8.15 and it worked fine.

lbernstone, 2021-09-03:
1.8.15 shows the option for me to upgrade to 2.0.0 in boards manager.  Ubuntu 21.04

Xylopyrographer, 2021-09-03:
OK. Did some mucking about.

In both Arduino IDE v1.8.15 and 2.0ß11 under Preferences: Additional Boards Manager, the URLs were:
https://dl.espressif.com/dl/package_esp32_index.json,https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json <https://dl.espressif.com/dl/package_esp32_index.json,https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json>

In Arduino IDE v1.8.15, changed this to be only:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json <https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json>

Quit and restart v1.8.15 and now the option to upgrade to arduino-esp32 v2.0.0 core appears.

Guess that was enough of a kick to refresh things…

Thanks all.
===================

closed 2021-09-06 by VojtechBartoska