espressif/arduino-esp32 #5632: SDMMC inconsistent behaviour crashes

lunadm, 2021-09-05:
* Development Board: ESP32-CAM
* Module or chip used: ESP32-S
* IDF version: Arduino 2.0.0-release
* Operating System: Mac
* Power Supply: [USB/battery]
* Chip is ESP32-D0WD (revision 1)

Inconsistent behaviour/crashing when mounting sd cards 

I’ve been using the same approach to initialising the sd card on the esp32-cam since RC4  - 
(Under RC4 I found if I mounted the cards at 64x1024 they could be mounted and read by both Mac and windows - other allocations seems to cause issues for Mac OS.)

But with RC20  I’m experiencing issues which are inconsistent
Booting with no card in causes a crash.
256gb card and 128gb only boot in 1 bit mode Using SD_MMC.begin("/sdcard",true))
But will crash with SD_MMC.begin()
64g cards crash no matter what

The only way I could get 64gb cards not to crash is to mount the cards in strange way -
For example I have to set slot.config width=1 - (even though I want to run in 4bit)
This will allow the card to boot but generates a setup error
_E (1144) vfs_fat_sdmmc: host init failed (0x103)._

The cards will then continue to work on the esp- but are unmountable/not recognised on any OS.

Interestingly if I try to run in true 1 bit by adding  
set host.flags = SDMMC_HOST_FLAG_1BIT; 
The cards will not boot/crash

Card size	SD_MMC.begin()	SD_MMC.begin("/sdcard",true)
No card	Crash	Crash
16gb	works	works
64gb 	Crash	Crash
128gb 	Crash	Boots
256gb	Crash	Boots

me-no-dev, 2021-09-07:
this is an issue in the ESP-IDF part of the SDMMC driver. It will be fixed there and will then be pulled here.

P-R-O-C-H-Y, 2021-11-01:
Hi @lunadm.
Tested the SD_MMC example with 16, 32, 64, 128 Gb cards and all of them were working.

Chip: ESP32
IDF Version: Arduino 2.0.1 RC1

Can you try updating arduino-esp32 to latest release `2.0.1-rc1` ?

P-R-O-C-H-Y, 2021-11-22:
@lunadm Any updates?