espressif/arduino-esp32 #5635: MPR121 with HUZZAH32 : noise from cap sensor readings via I2C

tahunus, 2021-09-05:
### Hardware:
Board:							HUZZAH32 (ESP WROOM 32)
Core Installation version:			2.0.0 (Arduino core for ESP32)
IDE name:						Arduino IDE
Flash Frequency:					40Mhz (also tried 80Mhz with same results)
Partition Scheme:                                    no OTA (also tried Default with same results)
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
When running Adafruit's sketch to debug MPR121's "raw" data using cap.filteredData(i) & cap.baselineData(i), everything works just fine if using an Arduino UNO or an Arduino MEGA. When running it on a HUZZAH32, the filtered data keeps returning wild readings (65535, 32738, etc.) on several (not all, but almost never none) of the cap sensors. The HUZZAH32 is connected via USB to my laptop (same as the tests on the Arduinos). I2C ports on HUZZAH are 23 for SDA & 22 for SCL. I ran other sample programs (not involving MPR121) on the HUZZAH32 and it works fine (so the board is ok).

A brief sample of the monitor output from the 12 sensors´ filtered and base data looks like this:

Filt: 227 216 198 195 195 189 65535 205 65535 65535 65535 32738
Base: 224 216 196 192 192 188 212 1020 196 604 180 196
                                                                                                                     0x0
Filt: 65535 65535 197 195 65535 189 65535 205 65535 185 181 197
Base: 624 216 196 192 192 188 212 204 1020 184 180 196
                                                                                                                     0x0
Filt: 228 216 197 196 195 189 214 205 196 185 182 197
Base: 224 212 1020 192 1020 188 212 204 196 184 180 196
                                                                                                                     0x0

If I grab (i.e. touch) the USB cable feeding the HUZZAH32, all sensors go wild (it does not happen with the Arduinos), so there is clearly some issue with noise from the power line feeding the HUZZAH32. I tried with different USB cables with the same result.

Has anyone seen noise from the power line in the HUZZAH32 fed into the I2C channel? Or into the cap sensors of a MPR121?

Since I intend (eventually) to make this a wireless cap sensing solution (hence the need for a HUZZAH32), there won't be a USB cable. It will all be fed by a 1200mAh LiPO battery which may solve the noise problem from the power line. But, since there's is still a lot of coding ahead of this project, I'll need the USB tether.

Thinking it could be a speed compatibility issue (given the speed differences between Arduinos and HUZZAH32), I tested several clock speeds on the I2C channel (from 5k to 400k), changed crystal speed on the HUZZAH32 to 40MHz and 80MHZ, and added delays of up to 1 sec between readings, but got the same results (only sower or faster serial output).

I'm powering the MPR121 from 3V3. The same behavior is present when running the MPR121 from the USB (5V) supply (the MPR121 has a built-in regulator and pullup resistors in the I2C pins)

HUZZAH32:  https://learn.adafruit.com/adafruit-huzzah32-esp32-feather
MPR121 : https://learn.adafruit.com/adafruit-mpr121-12-key-capacitive-touch-sensor-breakout-tutorial?view=all

Thanks in advance for any insights!

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <Wire.h>
#include "Adafruit_MPR121.h"

#ifndef _BV
#define _BV(bit) (1 << (bit))
#endif

Adafruit_MPR121 cap1 = Adafruit_MPR121();

void setup() {
 Serial.begin(115200);
 Serial.println("MPR121");
  if (!cap1.begin(0x5A)) {Serial.println("5A Bad"); while (1); }
   else {Serial.println("5A Good"); }
 Wire.begin (23, 22, 4000);
}
 
void loop() {
  Serial.print("\t\t\t\t\t\t\t\t\t\t\t\t\t 0x"); Serial.println(cap1.touched(), HEX);
  Serial.print("Filt: ");
  for (uint8_t i=0; i<12; i++) {
    Serial.print(cap1.filteredData(i)); Serial.print("\t");
  }
  Serial.println();
  Serial.print("Base: ");
  for (uint8_t i=0; i<12; i++) {
    Serial.print(cap1.baselineData(i)); Serial.print("\t");
  }
  Serial.println();
  delay(1000);
}
```

### Debug Messages:
```
Filt: [  4196][I][esp32-hal-i2c.c:1156] i2cProcQueue(): Bus busy, reinit
227	216	197	196	195	190	214	205	196	185	181	196	
Base: 224	216	1020	192	192	1020	212	204	1020	[  4207][I][esp32-hal-i2c.c:1156] i2cProcQueue(): Bus busy, reinit
[  4207][W][esp32-hal-i2c.c:1437] i2cCheckLineState(): invalid state sda(23)=0, scl(22)=1
[  4214][D][esp32-hal-i2c.c:1445] i2cCheckLineState(): Recovered after 1 Cycles
1020	180	196	
													 0x0
Filt: 227	216	198	195	195	190	214	205	196	65500	182	197	
Base: 224	216	196	192	192	188	212	204	196	184	180	196	
													 0x0
Filt: 227	216	197	[  6288][D][esp32-hal-i2c.c:1371] i2cProcQueue():  Gross Timeout Dead start=0x185d, end=0x188f, =50, max=50 error=1
[  6288][E][esp32-hal-i2c.c:336] i2cDumpI2c(): i2c=0x3ffbdb68
[  6293][I][esp32-hal-i2c.c:337] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[  6299][I][esp32-hal-i2c.c:339] i2cDumpI2c(): lock=0x3ffb90dc
[  6305][I][esp32-hal-i2c.c:341] i2cDumpI2c(): num=0
[  6310][I][esp32-hal-i2c.c:342] i2cDumpI2c(): mode=1
[  6314][I][esp32-hal-i2c.c:343] i2cDumpI2c(): stage=3
[  6319][I][esp32-hal-i2c.c:344] i2cDumpI2c(): error=1
[  6324][I][esp32-hal-i2c.c:345] i2cDumpI2c(): event=0x3ffb9164 bits=200
[  6331][I][esp32-hal-i2c.c:346] i2cDumpI2c(): intr_handle=0x3ffb9198
[  6337][I][esp32-hal-i2c.c:347] i2cDumpI2c(): dq=0x3ffb9278
[  6342][I][esp32-hal-i2c.c:348] i2cDumpI2c(): queueCount=2
[  6347][I][esp32-hal-i2c.c:349] i2cDumpI2c(): queuePos=1
[  6352][I][esp32-hal-i2c.c:350] i2cDumpI2c(): errorByteCnt=0
[  6358][I][esp32-hal-i2c.c:351] i2cDumpI2c(): errorQueue=2
[  6363][I][esp32-hal-i2c.c:352] i2cDumpI2c(): debugFlags=0x00000000
[  6369][I][esp32-hal-i2c.c:329] i2cDumpDqData(): Debug Buffer not Enabled
[  6376][I][esp32-hal-i2c.c:372] i2cDumpInts(): Debug Buffer not Enabled
65535	[  6393][I][esp32-hal-i2c.c:1156] i2cProcQueue(): Bus busy, reinit
195	190	65535	205	196	185	182	65535	
Base: 224	216	196	1020	[  6399][I][esp32-hal-i2c.c:1156] i2cProcQueue(): Bus busy, reinit
[  6400][W][esp32-hal-i2c.c:1437] i2cCheckLineState(): invalid state sda(23)=0, scl(22)=1
[  6408][D][esp32-hal-i2c.c:1445] i2cCheckLineState(): Recovered after 3 Cycles
608	188	212	204	196	184	180	196	
													 0x0
Filt: 227	216	197	196	195	65535	214	205	196	185	182	196	
Base: 224	216	196	192	192	1020	212	1020	196	1020	180	1020	
													 0x0
Filt: 227	216	197	196	195	190	214	205	196	185	182	197	
Base: 224	[  8490][D][esp32-hal-i2c.c:1371] i2cProcQueue():  Gross Timeout Dead start=0x20f7, end=0x2129, =50, max=50 error=1
[  8490][E][esp32-hal-i2c.c:336] i2cDumpI2c(): i2c=0x3ffbdb68
[  8494][I][esp32-hal-i2c.c:337] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[  8501][I][esp32-hal-i2c.c:339] i2cDumpI2c(): lock=0x3ffb90dc
[  8507][I][esp32-hal-i2c.c:341] i2cDumpI2c(): num=0
[  8512][I][esp32-hal-i2c.c:342] i2cDumpI2c(): mode=1
[  8516][I][esp32-hal-i2c.c:343] i2cDumpI2c(): stage=3
[  8521][I][esp32-hal-i2c.c:344] i2cDumpI2c(): error=1
[  8526][I][esp32-hal-i2c.c:345] i2cDumpI2c(): event=0x3ffb9164 bits=200
[  8532][I][esp32-hal-i2c.c:346] i2cDumpI2c(): intr_handle=0x3ffb9198
[  8539][I][esp32-hal-i2c.c:347] i2cDumpI2c(): dq=0x3ffb9278
[  8544][I][esp32-hal-i2c.c:348] i2cDumpI2c(): queueCount=2
[  8549][I][esp32-hal-i2c.c:349] i2cDumpI2c(): queuePos=1
[  8554][I][esp32-hal-i2c.c:350] i2cDumpI2c(): errorByteCnt=-1
[  8560][I][esp32-hal-i2c.c:351] i2cDumpI2c(): errorQueue=0
[  8565][I][esp32-hal-i2c.c:352] i2cDumpI2c(): debugFlags=0x00000000
[  8571][I][esp32-hal-i2c.c:329] i2cDumpDqData(): Debug Buffer not Enabled
[  8578][I][esp32-hal-i2c.c:372] i2cDumpInts(): Debug Buffer not Enabled
1020	[  8595][I][esp32-hal-i2c.c:1156] i2cProcQueue(): Bus busy, reinit
608	192	192	188	212	204	196	184	180	196	
													 0x0
Filt: 227	216	197	196	195	190	214	205	65535	185	181	197	
Base: 224	216	196	192	192	188	212	612	196	184	180	196	
													 0x0
Filt: 227	216	197	196	195	190	214	204	196	185	182	197	
Base: 224	1020	[ 10622][I][esp32-hal-i2c.c:1156] i2cProcQueue(): Bus busy, reinit
[ 10623][W][esp32-hal-i2c.c:1437] i2cCheckLineState(): invalid state sda(23)=0, scl(22)=1
[ 10626][D][esp32-hal-i2c.c:1445] i2cCheckLineState(): Recovered after 9 Cycles
196	192	192	188	1020	204	608	184	1020	196	
													 0x0
Filt: 227	216	197	195	195	190	214	205	196	185	181	197	
Base: 224	216	196	192	192	188	212	204	196	184	180	196	

```

SuGlider, 2021-09-06:
@tahunus 
Is there any reason for having  ```Wire.begin (23, 22, 4000);``` using 4kpbs setup for I2C clock?
Could you please test it with ``` Wire.begin (23, 22, **400000**);```  or  just ``` Wire.begin (23, 22);``` and let me know?

SuGlider, 2021-09-06:
@tahunus 

Use Arduino IDE using "Adafruit ESP32 Feather" as your board. The default I2C pins in its setup are already SDA = 23 and SCL = 22.
Please remove the line wth ```Wire.begin (23, 22, 4000);``` you added to your sketch and run it again.

This may solve your issue.

you setup with "Adafruit ESP32 Feather" should look like this:

``` cpp

void setup() {
 Serial.begin(115200);
 Serial.println("MPR121");
 
   // just to check that SDA and SCL are correctly set as 23 and 22 when using the right board setup with Arduino IDE
   // remove it latter checking it is working fine

   if (SDA != 23 || SCL != 22) {
       Serial.println("ERROR :: I2C SDA and SCL are not correct. Please check you board choice on Arduino IDE.");
   }

  if (!cap1.begin(0x5A)) {Serial.println("5A Bad"); while (1); }
   else {Serial.println("5A Good"); }
 }

```

tahunus, 2021-09-06:
thanks @SuGlider.  

> @tahunus
> Is there any reason for having `Wire.begin (23, 22, 4000);` using 4kpbs setup for I2C clock?
> Could you please test it with ` Wire.begin (23, 22, 400000);` or just ` Wire.begin (23, 22);` and let me know?

Thinking it could be a speed compatibility issue (given the speed differences between Arduinos and HUZZAH32), I tested several clock speeds on the I2C channel (from 5k to 400k), changed crystal speed on the HUZZAH32 to 40MHz and 80MHZ, and added delays of up to 1 sec between readings, but got the same results (only slower or faster serial output).
So there are no changes with any of the combinations. 

>
> `Wire.begin(23, 22)` shall be **executed before** ` cap1.begin(0x5A)` in order to setup used I2C pins prior to initializing MPR121.
> But looking into Adafruit code, I can tell that this will make it reset to default I2C pins...

Correct. That's why I added it after the cap1,begin. In any case, with or without the Wire.begin statement I get the same result

> What board setup are you using with Arduino IDE?

Board: HUZZAH32 (ESP WROOM 32)
Core Installation version: 2.0.0 (Arduino core for ESP32)
IDE name: Arduino IDE
Flash Frequency: 40Mhz (also tried 80Mhz with same results)
Partition Scheme: no OTA (also tried Default with same results)
Upload Speed: 921600

Thanks!

tahunus, 2021-09-06:
@SuGlider :

> Use Arduino IDE using "Adafruit ESP32 Feather" as your board. The default I2C pins in its setup are already SDA = 23 and SCL = 22.

That is precisely the board spec I'm using. It's the one recommended by Adafruit (HUZZAH32's manufacturer) https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/using-with-arduino-ide

> Please remove the line wth `Wire.begin (23, 22, 4000);` you added to your sketch and run it again.
> This may solve your issue.

Done. Same result

> 
> you setup with "Adafruit ESP32 Feather" should look like this:
> 
> ```c++
> void setup() {
>  Serial.begin(115200);
>  Serial.println("MPR121");
>  
>    // just to check that SDA and SCL are correctly set as 23 and 22 when using the right board setup with Arduino IDE
>    // remove this test latter, after checking it works fine on Adafruit HUZZAH32
> 
>    if (SDA != 23 || SCL != 22) {
>        Serial.println("ERROR :: I2C SDA and SCL are not correct. Please check you board choice on Arduino IDE.");
>    }
> 
>   if (!cap1.begin(0x5A)) {Serial.println("5A Bad"); while (1); }
>    else {Serial.println("5A Good"); }
>  }
> ```

SDA & SCL are correctly identified as 23 & 22 according to the DEBUG dump from the Arduino IDE (sample below, bigger sample in the first post above):
[  4207][W][esp32-hal-i2c.c:1437] i2cCheckLineState(): invalid state sda(23)=0, scl(22)=1
[  4214][D][esp32-hal-i2c.c:1445] i2cCheckLineState(): Recovered after 1 Cycles

I think the issue has to do with the stability of ESP32's implementation of I2C. Looking at the DEBUG dump, there are a lot of "bus busy" and "invalid state" that, although recovered after several cycles, by then the reading has been reported as 65535.

Thanks!!

SuGlider, 2021-09-06:
@tahunus  Thanks for testing and reporting back here. 
In fact it seems to be related to issue #4729 

I2C is currently been refactored to be based on IDF in order to support ESP32, ESP32-S2 and ESP32-C3.
This issue and some other problems shall be solved in Arduino Core 2.0.1 release.

tahunus, 2021-09-07:
Thanks @SuGlider. I guess I'll have to wait. Meanwhile, I'll run some other scenarios and report back in the hopes of providing further insights.

SuGlider, 2021-09-14:
@tahunus , I just submitted a PR  #5664 that shall fix the issue you reported.
Could you please apply it to your environment and verify if it solves this issue?
Thanks.

tahunus, 2021-09-18:
@SuGlider , pardon my newby-ness: besides adding the link 
 https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
on the Arduino's Additional Boards Manager URL, how do I download the 4 files marked as changed? I've found the originals in my laptop on users>local>Arduino15>packages>ESP32>hardware>2.0.0>cores>ESP32 and have made a backup copy of them. Should I just copy & paste the code from PR #5664 in a local file for each file? Thanks, man!

SuGlider, 2021-09-18:
@tahunus Yes, Just copy the files.
I see you are already in 2.0.0, so all needed is to copy the 4 files to their respective folders.

tahunus, 2021-09-18:
@SuGlider  when compiling with the 4 new files, these compiler errors appear:

`Arduino: 1.8.15 (Windows 10), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None, Default"

In file included from C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/esp32-hal.h:79,

                 from C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/Arduino.h:36,

                 from sketch\MPR121test.ino.cpp:1:

C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/esp32-hal-i2c.h:34:5: error: expected '}' before 'I2C_ERROR_ADDR_NACK'

     I2C_ERROR_ADDR_NACK,

     ^~~~~~~~~~~~~~~~~~~

C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/esp32-hal-i2c.h:24:14: note: to match this '{'

 typedef enum {

              ^

C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/esp32-hal-i2c.h:37:5: error: 'typedef enum I2C_ERROR_ADDR_NACK I2C_ERROR_TIMEOUT' redeclared as different kind of symbol

     I2C_ERROR_TIMEOUT,

     ^~~~~~~~~~~~~~~~~

C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/esp32-hal-i2c.h:28:5: note: previous declaration 'I2C_ERROR_ADDR_NACK I2C_ERROR_TIMEOUT'

     I2C_ERROR_TIMEOUT,

     ^~~~~~~~~~~~~~~~~

C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/esp32-hal-i2c.h:40:5: error: 'typedef enum I2C_ERROR_ADDR_NACK I2C_ERROR_NO_BEGIN' redeclared as different kind of symbol

     I2C_ERROR_NO_BEGIN,

     ^~~~~~~~~~~~~~~~~~

C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/esp32-hal-i2c.h:33:5: note: previous declaration 'I2C_ERROR_ADDR_NACK I2C_ERROR_NO_BEGIN'

     I2C_ERROR_NO_BEGIN

     ^~~~~~~~~~~~~~~~~~

C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/esp32-hal-i2c.h:41:1: error: expected unqualified-id before '}' token

 } i2c_err_t;

 ^

C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/esp32-hal-i2c.h:41:3: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?

 } i2c_err_t;

   ^~~~~~~~~

   esp_err_t

C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/esp32-hal-i2c.h:48:1: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?

 i2c_err_t i2cWrite(i2c_t * i2c, uint16_t address, uint8_t* buff, uint16_t size, bool sendStop, uint16_t timeOutMillis);

 ^~~~~~~~~

 esp_err_t

C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/esp32-hal-i2c.h:49:1: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?

[... 77 lines cut ...]

 i2c_err_t i2cAddQueueRead(i2c_t *i2c, uint16_t i2cDeviceAddr, uint8_t *dataPtr, uint16_t dataLen, bool SendStop, EventGroupHandle_t event);

 ^~~~~~~~~

 esp_err_t

In file included from C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/Arduino.h:36,

                 from sketch\MPR121test.ino.cpp:1:

C:\Users\pedro\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32/esp32-hal.h:132:1: error: expected declaration before '}' token

 }

 ^

exit status 1

Error compiling for board Adafruit ESP32 Feather.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

SuGlider, 2021-09-19:
You can try copying the files from my GitHub:

https://github.com/SuGlider/arduino-esp32/blob/i2c-refactoring/cores/esp32/esp32-hal-i2c.c

https://github.com/SuGlider/arduino-esp32/blob/i2c-refactoring/cores/esp32/esp32-hal-i2c.h

https://github.com/SuGlider/arduino-esp32/blob/i2c-refactoring/libraries/Wire/src/Wire.cpp

https://github.com/SuGlider/arduino-esp32/blob/i2c-refactoring/libraries/Wire/src/Wire.h

Please check the correct folder for each one.

I hope it works better!

me-no-dev, 2021-09-20:
Hi could you please re-test with this new PR: https://github.com/espressif/arduino-esp32/pull/5683

tahunus, 2021-09-26:
@SuGlider  It works! I ran several scenarios and the noise in the I2C channel is gone. Operation is stable! Thanks!  (apologies for the long silence, my "real world" job was too demanding these last few days.

tahunus, 2021-09-26:
@me-no-dev It works! I ran several scenarios and the noise in the I2C channel is gone. Operation is stable! Thanks! (apologies for the long silence, my "real world" job was too demanding these last few days.

PR #5683 (2021-10-01): Implement Thread-Safe I2C based on ESP-IDF API
## Summary
This PR is a complete refactoring of  Wire Library and I2C HAL in order to use IDF instead of current Register manipulation approach.

It implements Thread-safe Arduino Wire.

Fix #5478 
Fix #5635 
Fix #5636 
Fix #4729 
Fix #5470 

## Impact
It solves many reported issues related to I2C.
closed 2021-09-26 by tahunus