espressif/arduino-esp32 #5636: [ESP32-s2] Crash when scanning I2C

TD-er, 2021-09-06:
### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
When scanning the I2C bus, the ESP32-S2 immediately crashes.

The crash suggests the pins are the same, but that's not how I set them.
The I2C pins are set via `Wire.begin` and not via the `setPins` call, since I have no explicit differences between ESP8266 and ESP32 code for I2C.

It crashes on line 1855: (`i2c_master_stop`)
https://github.com/espressif/arduino-esp32/blob/92ce408f4c71d235f0961a265e8d202f9fd570d5/cores/esp32/esp32-hal-i2c.c#L1846-L1859

The scanning procedure is as basic as it gets:
https://github.com/letscontrolit/ESPEasy/blob/a27b6a6fc1e50861b52c70bbf361f9ccc6c9587f/src/src/WebServer/I2C_Scanner.cpp#L305-L365

### Debug Messages:
```
E (89928) i2c: i2c_set_pin(860): scl and sda gpio numbers are the same
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4010641e  PS      : 0x00060f30  A0      : 0x80083080  A1      : 0x3ffd3290
A2      : 0x00000000  A3      : 0x00000002  A4      : 0x3ffc5e5a  A5      : 0x00000000  
A6      : 0x3ffd85f4  A7      : 0x00000032  A8      : 0x8010641e  A9      : 0x3ffd3250  
A10     : 0x00000000  A11     : 0x00000002  A12     : 0x00000900  A13     : 0x00000000
A14     : 0x00001800  A15     : 0x00000000  SAR     : 0x00000016  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x00000900  LEND    : 0x00000000  LCOUNT  : 0x40028855  

Backtrace:0x4010641b:0x3ffd32900x4008307d:0x3ffd32b0 0x40083191:0x3ffd32d0 0x400f12a5:0x3ffd32f0 0x40101553:0x3ffd3360 0x40194f45:0x3ffd33a0 0x4008b59e:0x3ffd33c0 0x4008d5be:0x3ffd3400 0x4008d6e7:0x3ffd3470 0x400cb4a6:0x3ffd34c0 0x400cb791:0x3ffd3500 0x400a0bfb:0x3ffd3540 0x4010843c:0x3ffd3560 
  #0  0x4010641b:0x3ffd32900 in i2cWrite at C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c.c:1855

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026ad0
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3ffe6100,len:0x7c
load:0x4004c000,len:0x7e4
load:0x40050000,len:0x2598
entry 0x4004c15c
```

chegewara, 2021-09-06:
`E (89928) i2c: i2c_set_pin(860): scl and sda gpio numbers are the same`
@TD-er any idea what that means?

TD-er, 2021-09-06:
> `E (89928) i2c: i2c_set_pin(860): scl and sda gpio numbers are the same`
> @TD-er any idea what that means?

Well actually no, as I'm sure I don't set the same pins.
Also I cannot find the error in the code, so it is hard to see why it gives this error.

chegewara, 2021-09-06:
Not even close, but what i know.

TD-er, 2021-09-06:
> Not even close, but what i know.

Care to elaborate?
You mean the linked issue is not even close?

TD-er, 2021-09-06:
Just updated the opening post, as the scanning seems to be the only issue here. When connecting an I2C device like an OLED display, it does seem to work just fine.
So this does prove the I2C pins are correct.

chegewara, 2021-09-06:
In that case log message is stupid, but the issue you found seems to be already fixed in your code:
https://github.com/espressif/arduino-esp32/issues/5470#issuecomment-897230856

```
    if(size) 
        i2c_master_write(cmd, buff, size, ACK_CHECK_EN);
```
Maybe there is something else we didnt find yet.

TD-er, 2021-09-06:
Yep, I double checked and your fix is already present in the code I compiled.

TD-er, 2021-09-06:
OK, I changed my code to include the frequency at `Wire.begin` and now it does show up the connected I2C device.
Also I don't call `Wire.setClock` anymore now. (for ESP32 code)

SuGlider, 2021-09-06:
@TD-er  - i2c is currently been refactored to be based on IDF in order to support ESP32, ESP32-S2 and ESP32-C3.
This issue and some other problems shall be solved in Arduino Core  2.0.1 release.

TD-er, 2021-09-06:
@SuGlider 
While you're at it, please also incorporate the clock stretch limit, as that's really being missed.

Do you have an estimate on when this refactoring may be done?

SuGlider, 2021-09-06:
@TD-er , yes, sure!
Thanks for taking time to debug and to find a work around for the issue!

SuGlider, 2021-09-14:
@TD-er  I just submitted a PR  #5664 that shall fix the issue you reported.
Could you please apply it to your environment and verify if it solves this issue?
Thanks.

TD-er, 2021-09-16:
@SuGlider I manually replaced the 4 files in your PR but it doesn't even compile (for ESP32-s2) with lots and lots of errors (mainly not recognized types, so probably some other changes elsewhere too)

Then I tried to set PlatformIO to use your branch as platform:
```
platform                  = https://github.com/SuGlider/arduino-esp32.git#i2c-refactoring
platform_packages         = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/424/framework-arduinoespressif32-master-cdcf92440.tar.gz
build_flags               = -DESP32_STAGE
```

But that doesn't work either:
```
Executing task in folder ESPEasy: C:\Users\gijsn\.platformio\penv\Scripts\platformio.exe run --target clean --environment custom_ESP32s2_4M316k <

Processing custom_ESP32s2_4M316k (board: esp32-s2-saola-1; framework: arduino; platform: https://github.com/SuGlider/arduino-esp32.git#i2c-refactoring)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Platform Manager: Installing git+https://github.com/SuGlider/arduino-esp32.git#i2c-refactoring
git version 2.28.0.windows.1
Cloning into 'C:\Users\gijsn\.platformio\.cache\tmp\pkg-installing-87zu1o39'...
remote: Enumerating objects: 4211, done.
remote: Counting objects: 100% (4211/4211), done.
remote: Compressing objects: 100% (2876/2876), done.
remote: Total 4211 (delta 1103), reused 2628 (delta 848), pack-reused 0 eceiving objects:  99% (4169/4211), 109.14 MiB | 13.83 MiB/s
Receiving objects: 100% (4211/4211), 110.99 MiB | 12.82 MiB/s, done.
Resolving deltas: 100% (1103/1103), done.
Updating files: 100% (6564/6564), done.
Error: Could not find one of 'platform.json' manifest files in the package
The terminal process "C:\Users\gijsn\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'clean', '--environment', 'custom_ESP32s2_4M316k'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

I'm always having these issues where I try to build with changes in the core libraries like this one where it just doesn't work.
Really frustrating.

SuGlider, 2021-09-16:
Did you update to Arduino Core 2.0.0 before replacing those 4 files?

SuGlider, 2021-09-16:
Check this. It may help you:
https://github.com/espressif/arduino-esp32/issues/5470#issuecomment-920263554

TD-er, 2021-09-16:
> Did you update to Arduino Core 2.0.0 before replacing those 4 files?

I honestly don't know, as I am using PlatformIO, with platform_packages collected by @Jason2866 

The platformio packages:
```
[core_esp32_3_3_2]
platform                  = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
platform_packages         = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/424/framework-arduinoespressif32-master-cdcf92440.tar.gz
build_flags               = -DESP32_STAGE
```
The framework "version" is a bit unclear as it states "0.0.0"

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s2-saola-1.html
PLATFORM: Espressif 32 (3.3.1+sha.33ff413) > Espressif ESP32-S2-Saola-1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 1.81MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51)
 - toolchain-riscv32-esp 8.4.0+2021r1
 - toolchain-xtensa-esp32s2 8.4.0+2021r1
```

Regarding ESP32, I completely lost track of which version we're now on as it is a gathered set of toolkits just to be able to build some ESP32-S2 builds that seem to boot after a rebuild.

It looks like it is based on platform-espressif32 version 3.3.1, which dates from before arduino-esp32 2.0.0 was tagged, so I guess it may not be the "2.0.0" right now.

Jason2866, 2021-09-16:
@TD-er  the build you use is at the commit of core 2.0.0

Jason2866, 2021-09-16:
@SuGlider @TD-er  First test looks good. A connected Sensor BME280 does work.
[image]

TD-er, 2021-09-16:
Sadly, doesn't compile for the ESP32-S2:

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s2-saola-1.html
PLATFORM: Espressif 32 (3.3.2) > Espressif ESP32-S2-Saola-1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 1.81MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51) 
 - toolchain-riscv-esp 1.80400.0 (8.4.0)
 - toolchain-xtensa32s2 1.80400.210211 (8.4.0)
Converting ESPEasy.ino
Error: This board doesn't support arduino framework!
```

```
[core_esp32_3_3_2]
platform                  = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
platform_packages         = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/424/framework-arduinoespressif32-master-cdcf92440.tar.gz
build_flags               = -DESP32_STAGE
```
```
[esp32_common]
extends                   = common, core_esp32_3_3_2
lib_deps                  = td-er/ESPeasySerial @ 2.0.6, adafruit/Adafruit ILI9341 @ ^1.5.6, Adafruit GFX Library, LOLIN_EPD, Adafruit BusIO, VL53L0X @ 1.3.0, SparkFun VL53L1X 4m Laser Distance Sensor @ 1.2.9, td-er/SparkFun MAX1704x Fuel Gauge Arduino Library @ ^1.0.1
lib_ignore                = ${esp32_always.lib_ignore}, ESP32_ping, IRremoteESP8266, HeatpumpIR
board_build.f_flash       = 80000000L
board_build.flash_mode    = dout
board_upload.maximum_size = 1900544
board_build.partitions    = esp32_partition_app1810k_spiffs316k.csv
extra_scripts             = post:tools/pio/post_esp32.py
                            ${extra_scripts_default.extra_scripts}
build_unflags             = -Wall
build_flags               = ${core_esp32_3_3_2.build_flags} 
                            ${mqtt_flags.build_flags} 
                            -DCONFIG_FREERTOS_ASSERT_DISABLE
                            -DCONFIG_LWIP_ESP_GRATUITOUS_ARP
                            -fno-strict-aliasing
                            -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_NONE
monitor_filters           = esp32_exception_decoder

; ESP32-S2
[env:custom_ESP32s2_4M316k]
extends                   = esp32_common
build_flags               = ${esp32_common.build_flags} 
                            -DPLUGIN_BUILD_CUSTOM
                            -DESP32S2
board                     = esp32-s2-saola-1
extra_scripts             = ${esp32_common.extra_scripts}
                            pre:tools/pio/pre_custom_esp32.py
platform                  = ${core_esp32_stage.platform}
platform_packages         = ${core_esp32_stage.platform_packages}
```

TD-er, 2021-09-16:
Changed the board to this:
```
board                     = esp32dev
```

With an error at linking:

```
Linking .pio\build\custom_ESP32s2_4M316k\ESP_Easy_mega_20210916_custom_ESP32s2_4M316k.elf
c:/users/gijsn/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: final link failed: bad value
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\custom_ESP32s2_4M316k\ESP_Easy_mega_20210916_custom_ESP32s2_4M316k.elf] Error 1
```

Jason2866, 2021-09-16:
Using wrong `xtensa-esp32s2-elf/8.4.0` version of the compiler/toolchain. Your platformio setup is screwed. Easy solution. Delete the whole hidden folder `.platformio`. All needed will rebuilded (downloaded)

TD-er, 2021-09-16:
Was the rest of my PIO platform selection OK?

Jason2866, 2021-09-16:
Yes, last setup is okay. I does not mess up older builds for me. I will add a package.json with a unique version number. I think the problem is caused from the `0.0.0` version number

TD-er, 2021-09-16:
Right now I have it running on the S2 and it seems to run fine.
Still have to "revert" my changes to prevent the crash I had before, but I guess my changes were a good change anyway as the setClock now has a comment that it should be called before `begin()` anyway.

Jason2866, 2021-09-16:
Added a version number. If you delete again you should get now `2.0.0+i2cfix`

TD-er, 2021-09-16:
Just a remark about the code changes, I will post them in the respective PR.

me-no-dev, 2021-09-20:
Hi could you please re-test with this new PR: https://github.com/espressif/arduino-esp32/pull/5683

TD-er, 2021-09-20:
@Jason2866 Can you make a new PIO platform_package ?

Jason2866, 2021-09-20:
@TD-er Here you are https://github.com/Jason2866/arduino-esp32/releases/download/v.2.0.0-PR5683/framework-arduinoespressif32_PR5683.zip

Jason2866, 2021-09-20:
Does work with Tasmota. The SuGlider version did run 4 days nonstop with Tasmota until now without issues.

TD-er, 2021-09-20:
@Jason2866 Thanks, will try to build it with ESPEasy.
The SuGlider version did indeed run well without any issues.
But I did not try to access it from multiple threads.

PR #5683 (2021-10-01): Implement Thread-Safe I2C based on ESP-IDF API
## Summary
This PR is a complete refactoring of  Wire Library and I2C HAL in order to use IDF instead of current Register manipulation approach.

It implements Thread-safe Arduino Wire.

Fix #5478 
Fix #5635 
Fix #5636 
Fix #4729 
Fix #5470 

## Impact
It solves many reported issues related to I2C.
closed 2021-10-01 by me-no-dev