espressif/arduino-esp32 #5638: Copysign and round are not in std:: namespace

assumptionsoup, 2021-09-06:
### Hardware:
Board:							Adafruit Feather esp32
Core Installation version:			3.2.0)
IDE name:						Platform.io
Flash Frequency:					N/A (just building)
PSRAM enabled:                                    N/A (just building)
Upload Speed:					N/A (just building)
Computer OS:                                        Windows 10

### Description:
At least as far as I can tell, copysign and round global and aren't in the std:: namespace where other libraries expect them to be.

I'm trying to use the [linalg library](https://github.com/sgorsten/linalg) with esp32 using platformio. I filled [this bug](https://github.com/platformio/platform-espressif32/issues/237) with platformio a few years back and I'm trying again now. They suggested that this was an espressif issue. 

### Sketch:
```cpp

#include <Arduino.h>
#include <string>
#include <cmath>

void setup() {
    round(10);  //works
    std::round(10);   // fails
}

void loop() {
}

```

### Debug Messages:
```
src\main.cpp: In function 'void setup()':
src\main.cpp:7:5: error: 'round' is not a member of 'std'
     std::round(10);
```

atanisoft, 2021-09-07:
Unfortunately this looks to be a bug in the toolchain and not an Arduino-esp32 specific bug and can be reproduced using only ESP-IDF code.

igrr, 2021-09-08:
@atanisoft could you share a reproducer for ESP-IDF, please? The following seems to compile successfully (except for unused-variable warnings) with the latest IDF:
```c++
#include <string>
#include <cmath>

extern "C" void app_main()
{
    float a = round(10);
    float b = std::round(10);
}
```

Looking at [platformio releases page](https://github.com/platformio/platform-espressif32/releases), release 3.2.0 corresponds to arduino-esp32 release 1.0.6. That release was using an older toolchain version based on GCC 5.2.
In the latest 2.0.0 release which uses GCC 8.4 this issue should be fixed (same for other math related issues, e.g. https://github.com/espressif/arduino-esp32/issues/4771).

It seems that Platform.io folks haven't released a new version which would include arduino-esp32 v2.0.0 release. When that happens, I'd ask @assumptionsoup to upgrade and try again.

atanisoft, 2021-09-08:
@igrr I haven't seen this even in arduino-esp32 1.0.6 but in theory it should have failed. I don't have a current reproducing case as I'm using esp-idf v4.3 / master and arduino-esp32 master.

VojtechBartoska, 2022-03-30:
Hello @assumptionsoup, I'm closing this issue as we weren't able to reproduce it and no more info was provided. If needed you can reopen it. Thanks for your contribution.

closed 2022-03-30 by VojtechBartoska