espressif/arduino-esp32 #5639: esptool.py in 2.0.0 seems to require macOS 10.15 minimum, fails on 10.13

obdevel, 2021-09-07:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			2.0.0
IDE name:						Arduino IDE 1.8.15
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					460800
Computer OS:                                        macOS 10.13.6

esptool.py seems to be built for macOS 10.15 minimum, so it fails on 10.13.x

```[75924] Error loading Python lib '/var/folders/03/s3pv8qk91w78m4_st_nbxl5m0000gn/T/_MEIOL3T85/libpython3.8.dylib': dlopen: dlopen(/var/folders/03/s3pv8qk91w78m4_st_nbxl5m0000gn/T/_MEIOL3T85/libpython3.8.dylib, 10): Symbol not found: ____chkstk_darwin
  Referenced from: /var/folders/03/s3pv8qk91w78m4_st_nbxl5m0000gn/T/_MEIOL3T85/libintl.8.dylib (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libSystem.B.dylib
 in /var/folders/03/s3pv8qk91w78m4_st_nbxl5m0000gn/T/_MEIOL3T85/libintl.8.dylib```

Jason2866, 2021-10-10:
This will probably not solved since github actions runners oldest version which can be used is macos-10.15

zfields, 2022-01-28:
> The fastest solution is to update your OS.

Apple won't let me update my Macbook Pro, because it's too old. I'm stuck at High Sierra. :disappointed: 

You may not have meant it this way, but what you said comes off as, "Buy a new machine, or you're too poor to use our hardware."

It would be nice if you could offer any helpful suggestion or workaround, since it is clear you are not intending to fix the issue.

dpharris, 2022-01-28:
You could try using vmware or virtualbox to load Linux, and use that, or
bootup into a Linux environment.

David

zfields, 2022-01-28:
I figured out my answer...

It turns out, `esptool` is a stand alone _compiled_ python "script". Fortunately, the repository is open-source so you can clone it, package it yourself and replace the `esptool` "binary".

> Arduino ESP32 used automatically built esptools by Github Actions running on MacOS 10.15.

Here is how they are building it in their CI:
https://github.com/espressif/esptool/blob/master/.github/workflows/build_esptool.yml

Here are the steps required to build it on your own machine:
1. [[Re]Install Homebrew](https://stackoverflow.com/questions/46459152/cant-chown-usr-local-for-homebrew-in-mac-os-x-10-13-high-sierra/46459153#46459153)\*

   ```none
   /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"
   ```

2. [[Re]Install Python PIP](https://stackoverflow.com/questions/41233594/installing-pip-on-macos-sierra)

   ```
   brew reinstall python
   ```

3. Add new Python installation to `PATH` (update to current version of Python3)

   ```
   export PATH="/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/bin:$PATH"
   ```

4. Install `pyinstaller` (use version 4.3 to [avoid code signing problems](https://stackoverflow.com/questions/68884906/pyinstaller-error-systemerror-codesign-failure-on-macos))

   ```
   pip install -Iv pyinstaller==4.3
   ```

5. Install `pyserial` for packaging in the `esptool` "binary"

   ```
   pip install pyserial
   ```

6. Clone `esptool` repository at version 3.1

   ```
   git clone https://github.com/espressif/esptool.git -b v3.1
   ```

7. Step into the repository, make a build directory

   ```
   cd esptool && mkdir esptool_py && cd esptool_py
   ```

8. Create the replacement `esptool` "binary"

   ```
   pyinstaller --onefile ../esptool.py
   ```

9. Copy aside and replace the broken "binary"

   ```
   mv /Users/${USER}/Library/Arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool /Users/${USER}/Library/Arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool_broken
   cp dist/esptool /Users/${USER}/Library/Arduino15/packages/esp32/tools/esptool_py/3.1.0/
   ```

_\* If you are having permissions issues when using `brew install` I would suggest reinstalling Homebrew over itself if you have High-Sierra; it basically "fixes it". Myself and several others had numerous problems with stale Homebrew installations on High-Sierra._

techstudio-design, 2022-01-28:
@zfields Thanks for this solution.

Jason2866, 2022-01-28:
A different solution would be using Platformio (with VSC). There is no need for a compiled esptool since Platformio uses the python version.

zfields, 2022-01-28:
FWIW, you should not trust me, or anyone else who provides you a binary from the internet, but here you go:
https://www.dropbox.com/s/y09fcrt27sjzub8/esptool?dl=0

_I created/compiled this using Python 3.9 on High-Sierra (10.13), so you will likely need to satisfy both of those requirements in order for this to work. If it doesn't work, then just follow the steps above. They are relatively painless and guarantee that you will create a binary compatible with your system._

Jason2866, 2022-01-28:
It is a good idea to use esptool.py v3.2 with Arduino core 2.0.2 and pyserial has to be >=v3.5

technoblogy, 2022-01-28:
@zfields Thank you so much - it works, and you saved me a lot of work. Suggestion: perhaps espressif could provide an official version as a service to users who don't have the latest Macs.

Note for anyone else doing this: I got "permission denied" and had to do `sudo chmod 777 esptool`

obdevel, 2022-01-29:
Thanks from me (the OP) too. Saves having to spin up a Linux VM.
Works just fine on macOS 10.13.6 with core v 2.0.0, with the chmod +x fix.

marchingband, 2022-02-11:
@zfields you are my hero <3

technoblogy, 2022-04-16:
I managed to get this working with 2.0.0 on my High Sierra Mac by replacing esptool 3.1.0 with one kindly provided by @zfields, but after upgrading to 2.0.3-RC1 it seems to have stopped working again.

zfields, 2022-04-16:
When you upgrade it replaces the contents of the directory. Have you tried replacing `esptool` after upgrading?

technoblogy, 2022-04-16:
Thanks for your comment. Yes, I copied across the one you provided. I noticed that the folder name has changed from 3.1.0 to 3.3.0.

zfields, 2022-04-16:
Ughhh... I know a recompile following the steps above would work, but I'm not sure it's necessary. Unfortunately, I don't have time to look at it at the moment.

technoblogy, 2022-04-16:
No problem; I'm grateful for the response anyway.

It's a pity that espressif can't do it, to help people like me who don't have the latest hardware.

marchingband, 2022-10-18:
I am willing to worship the person who repeats this process for us poor souls wanting to use 2.0.3+ on our old macbooks. I have an esp32s3 right here and can't believe this is happening :'-( @zfields do you want to be my hero again?

marchingband, 2022-10-18:
Ok I made my own, which works with macos 10.13, this is esptool version 3.3.2
I moved it into the 3.3.0 folder as per the instructions above.
I got a python version error, but found that launching Arduino from the command line fixes that, and I can now compile for S3. As stated above, don't trust random files online but if you're crazy like me meh why not. Hope this helps someone!
https://file.io/ZTT5YYGrHwMm

technoblogy, 2022-10-18:
@marchingband thank you for helping us MacOS 10.13 users!

> I got a python version error

I assume you mean that when using the Arduino IDE you get:

    exec: "python3": executable file not found in $PATH

I confirmed that python3 is actually installed on my Mac by typing this in the Terminal:

    david$ which python3
    /Library/Frameworks/Python.framework/Versions/3.8/bin/python3

The solution I finally got to work is to edit the platform.txt file in: /Users/david/Library/Arduino15/packages/esp32/hardware/esp32/2.0.5

Find the line:

    tools.gen_esp32part.cmd=python3 "{runtime.platform.path}/tools/gen_esp32part.py"

Change python3 to the full path given in the "which python3" command earlier. In my case it becomes:

    tools.gen_esp32part.cmd=/Library/Frameworks/Python.framework/Versions/3.8/bin/python3 "{runtime.platform.path}/tools/gen_esp32part.py"

Then everything worked fine!

marchingband, 2022-10-18:
That's amazing 😀 thank you for reporting back 🤍 us vintage Mac users gotta stick together!

marchingband, 2022-10-19:
I can confirm that (with the updated link above) this esptool works for Arduino-esp32 versions 2.3 to 2.5, and using @technoblogy 's patch to platform.txt, works as it should from the IDE.

leetut, 2023-01-31:
step 4: 
lee$ pip install -Iv pyinstaller==4.3
-bash: pip: command not found

leetut, 2023-02-01:
usage: esptool [-h]
               [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}]
               [--port PORT] [--baud BAUD]
               [--before {default_reset,usb_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset,no_reset_stub}]
               [--no-stub] [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               [--connect-attempts CONNECT_ATTEMPTS]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
               ...
esptool: error: argument --chip/-c: invalid choice: 'esp32s3' (choose from 'auto', 'esp8266', 'esp32', 'esp32s2', 'esp32s3beta2', 'esp32s3beta3', 'esp32c3', 'esp32c6beta')
exit status 2
/Applications/Arduino.app/Contents/Java/arduino-builder returned 2
Error compiling for board ESP32S3 Dev Module.

leetut, 2023-02-01:
tried everything on this page, nothing works
```
bash: /Users/lee/Library/Arduino15/packages/esp32/tools/esptool_py/4.2.1/esptool: Permission denied
exit status 126
/Applications/Arduino.app/Contents/Java/arduino-builder returned 126
Error compiling for board ESP32S3 Dev Module.
```

ROCKLOTUS, 2023-04-23:
> I figured out my answer...
> 
> It turns out, `esptool` is a stand alone _compiled_ python "script". Fortunately, the repository is open-source so you can clone it, package it yourself and replace the `esptool` "binary".
> 
> > Arduino ESP32 used automatically built esptools by Github Actions running on MacOS 10.15.
> 
> Here is how they are building it in their CI: https://github.com/espressif/esptool/blob/master/.github/workflows/build_esptool.yml
> 
> Here are the steps required to build it on your own machine:
> 
> > _**NOTE:** If you already have Python3 installed, then skip to step 4_
> 
> 1. [[Re]Install Homebrew](https://stackoverflow.com/questions/46459152/cant-chown-usr-local-for-homebrew-in-mac-os-x-10-13-high-sierra/46459153#46459153)*
>    ```
>    /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"
>    ```
> 2. [[Re]Install Python PIP](https://stackoverflow.com/questions/41233594/installing-pip-on-macos-sierra)
>    ```
>    brew reinstall python
>    ```
> 3. Add new Python installation to `PATH` (update to current version of Python3)
>    ```
>    export PATH="/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/bin:$PATH"
>    ```
> 4. Install `pyinstaller` (use version 4.3 to [avoid code signing problems](https://stackoverflow.com/questions/68884906/pyinstaller-error-systemerror-codesign-failure-on-macos))
>    ```
>    pip install -Iv pyinstaller==4.3
>    ```
> 5. Install `pyserial` for packaging in the `esptool` "binary"
>    ```
>    pip install pyserial
>    ```
> 6. Clone `esptool` repository at version 3.1
>    ```
>    git clone https://github.com/espressif/esptool.git -b v3.1
>    ```
> 7. Step into the repository, make a build directory
>    ```
>    cd esptool && mkdir esptool_py && cd esptool_py
>    ```
> 8. Create the replacement `esptool` "binary"
>    ```
>    pyinstaller --onefile ../esptool.py
>    ```
> 9. Copy aside and replace the broken "binary"
>    ```
>    mv /Users/${USER}/Library/Arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool /Users/${USER}/Library/Arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool_broken
>    cp dist/esptool /Users/${USER}/Library/Arduino15/packages/esp32/tools/esptool_py/3.1.0/
>    ```
> 
> _* If you are having permissions issues when using `brew install` I would suggest reinstalling Homebrew over itself if you have High-Sierra; it basically "fixes it". Myself and several others had numerous problems with stale Homebrew installations on High-Sierra._

Thanks for your answer, but I fail for follow error message:
Pyserial is not installed for /Users/rock/Documents/Code/ESP32/esptool-master/esptool_py/dist/esptool/./esptool. Check the README for installation instructions.
Traceback (most recent call last):

Then I try another method, I found esptool.py work well, so I make a soft link to this file and copy to the destination folder, it works
ln -s esptool.py esptool
cp esptool /Users/${USER}/Library/Arduino15/packages/esp32/tools/esptool_py/3.1.0/

Then back to the Arduino software and setting
Upload Speed:"115200"
JTAG Adapter:"ESP USB Bridge"
Flash Mode:"DIO"

finally, Arduino IDE 2.1.0 works well.

mgesteiro, 2023-05-07:
I've updated the procedure to reflect last changes from the **ESP32 framework** (v2.0.9) which includes **esptool v4.5.1**. These commands should be executed from within a terminal:

0. Create a working directory and move inside:
`mkdir esptoolupdate`
`cd esptoolupdate`
1.  Download and uncompress the corresponding package (into the previous folder):
`wget https://github.com/espressif/esptool/archive/refs/tags/v4.5.1.tar.gz`
`tar xvf v4.5.1.tar.gz`
2. You should already have **python3** available in your computer (if not, install it using *macports*, *brew* or your preferred method). Now we create a virtual environment with just the necessary requirements just for this task:
`python3 -m venv ve`
`source ve/bin/activate`
`pip install -U pip pyserial pyinstaller`
3. Time to create the new program/binary:
`cd esptool-4.5.1`
`pyinstaller --distpath ./esptool-macos -F --icon=ci/espressif.ico --add-data="./esptool/targets/stub_flasher/*.json:/esptool/targets/stub_flasher/" esptool.py`
4. Check that everything worked:
`deactivate`
`./esptool-macos/esptool`
5. Replace the bad one with the newly created:
`cp esptool-macos/esptool ~/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1/`
6. You can close the terminal now, remove the working folder, restart your Arduino IDE and **ENJOY 😊**

**NOTE:** this procedure has been tested in a real computer with *MacOS Mojave (v10.14.6)* and *Python 3.11* installed via [Macports](https://www.macports.org/install.php#installing) (in my experience, brew is a pita with our older systems and macports provided a better general behaviour).

ziggy90127, 2023-09-21:
For Mojave 10.14.6, I made some alterations for my situation (you might have to change "4.5.1" to whatever version you have)(I installed python3.11 and pip-3.11 with macports):

pip-3.11 install pyinstaller
pip-3.11 install pyserial
git clone https://github.com/espressif/esptool.git
cd esptool && mkdir esptool_py && cd esptool_py
python3.11 -m PyInstaller --onefile ../esptool.py
mv /Users/${USER}/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1/esptool /Users/${USER}/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1/esptool_broken
cp dist/esptool /Users/${USER}/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1/

I didn't seem to have a code signing issue mentioned above.

jorgechamorro, 2024-06-05:
> 3\. pyinstaller --distpath ./esptool-macos -F --icon=ci/espressif.ico --add-data="./esptool/targets/stub_flasher/*.json:/esptool/targets/stub_flasher/" esptool.py

There's a typo, /esptool/targets/stub_flasher should have had a dot . before: ./esptool/targets/stub_flasher , other than it werks.

I've done this in my mac running osx 10.13.6, and worked fine (the file that ends into /esptool-macos works) despite it throwing some errors about code signing.

ginjo, 2024-08-04:
> For Mojave 10.14.6

For others who land here looking to get esp32 running in Arduino IDE on Mojave, here's what worked for me to overcome the sketch compilation failures. Note that I did not try any of the solutions mentioned earlier in this thread.

* Downgraded esp32 in Boards Manager to v1.0.6.
* Previously installed esptool manually via ```pip3 install esptool```, but I don't know if that has anything to do with my success. 
* Previously re-linked ```/usr/local/bin/python``` to wherever my python3 is installed, but again, no idea if this contributed to my success.

Environment
* OSX 10.14.6
* MacBook Pro (Retina, 13-inch, Early 2015)
* Arduino IDE 1.8.19
* DOIT esp32 devkit v1 with CP2101 chip
* Silab CP2101 serial driver for Mac
* Python 3.9.12
* Esptool 4.7.0

nikbry, 2024-09-03:
> For others who land here looking to get esp32 running in Arduino IDE on Mojave, here's what worked for me to overcome the sketch compilation failures. Note that I did not try any of the solutions [...]

Thanks so much it worked for me on OSX 10.14.6, just by downgrading the board library to 1.0.6 nothing else needed. Beware that v 1.0.6 may not support some of the newer ESP32 boards.

technoblogy, 2025-04-28:
@pielasze thanks for the suggestion. What version of macOS are you using?

pielasze, 2025-04-28:
10.14.6 Mojave

ginjo, 2025-04-29:
Just a follow up, as I posted above about how I got esptool working on Mojave. I honestly can't even remember doing that, and I now do all of my esp32 development & testing in a Docker container running the `esphome/esphome` image. Everything I need is in that image, and it remains consistent no matter what hardware or OS I'm running on. No more OS version hassles, hardware incompatibility, python installs, homebrew nightmares... The only gotcha is that using USB devices with Docker can be tricky. I usually run the esphome container on a cloud instance, using  the `web.esphome.com` serial tool for local esp32 uploading and output monitoring. This may not suit everyone, but for developing esphome solutions and components, it works great for me.

ginjo, 2025-05-01:
> Can you share the Dockerfile?

I'm using the official [esphome/esphome](https://hub.docker.com/r/esphome/esphome) Docker image.

The image is set up to run the `esphome` command by default. However if you want to launch the container as a bash session, so you can run the various tools manually, try this:

    docker run -it --rm --entrypoint '' esphome/esphome bash

technoblogy, 2025-06-06:
@prc24 please can you explain this in a bit more detail as it sounds like the perfect solution:

> My workaround was to replace Arduino provided "esptool" binary with executable bash script that simply calls current python script:

When I execute:

    cat ~/Library/Arduino15/packages/esp32/tools/esptool_py/4.9.dev3/esptool

I just get a lot of non-printing characters in the terminal, so i assume that was the wrong thing to do.

I've downloaded the source of the esptool script from:

    https://github.com/espressif/esptool/archive/refs/tags/v4.9.dev3.tar.gz

and expanded it into a folder:

esptool-4.9.dev3

Is the bash script a text file? What do I put in it and where do I put that?

Thanks!

technoblogy, 2025-06-06:
@prc24 Thank you very much!

technoblogy, 2025-06-07:
@prc24 Everything worked apart from the last step:

    sudo pip install esptool
    sudo: pip: command not found

I checked I have pip:

    david$ python3 -m pip --version
    pip 19.2.3 from /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pip (python 3.8)

At this point I'm stuck. Any suggestions? Thanks!

ziggy90127, 2025-06-09:
> [@prc24](https://github.com/prc24) Everything worked apart from the last step:
>
> ```
> sudo pip install esptool
> sudo: pip: command not found
> ```
>
> I checked I have pip:
>
> ```
> david$ python3 -m pip [...]

Just in case you don't know about command completion, you can type "pip" (or anything) and press the Tab key, and you'll be shown matching commands available:

$ pip                         # press  Tab
pip-3.12  pip3          # results
$ which pip3            # use the which command to see where they are
/usr/bin/pip3
$ which pip-3.12
/opt/local/bin/pip-3.12
$