espressif/arduino-esp32 #5641: Compiling error with ESP32 core version 1.x vs. 2.x / rtc_gpio_desc vs. rtc_io_desc

ClemensGruber, 2021-09-08:
We have some problems getting the established OneWire lib from @PaulStoffregen working with Arduino ESP32 core version 2.x, see: https://github.com/PaulStoffregen/OneWire/issues/100 . The lib is working with ESP32 core for Arduino version 1.x but can not compile with core version 2.x. 

```
In file included from /home/alex/Arduino/libraries/OneWire/OneWire.cpp:144:
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:161:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:161:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:189:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:189:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
```

We tried to replace `rtc_gpio_desc `with `rtc_io_desc` waht leads to no compiling error but it is still not working functionally, see https://github.com/PaulStoffregen/OneWire/issues/100#issuecomment-910609906

me-no-dev, 2021-09-08:
@igrr will`CONFIG_RTCIO_SUPPORT_RTC_GPIO_DESC` be supported by ESP-IDF indefinitely?

VojtechBartoska, 2022-03-30:
Hello @ClemensGruber, we will reevaluate this again and let you know.

VojtechBartoska, 2022-04-05:
Status update: After evaluation we are keeping this feature request open and it needs more investigation.

lbernstone, 2022-04-05:
Take a look at https://github.com/pstolarz/OneWireNg/blob/master/src/platform/OneWireNg_ArduinoIdfESP32.cpp, where he claims to support new versions of esp-idf along with variant chips.

closed 2025-01-16 by Parsaabasi