espressif/arduino-esp32 #5642: Cannot use custom pins for SPI with esp32-pico-v3

brahmajit, 2021-09-08:
### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32 Dev Module|node32|ttgo_lora|ESP32-S2-Saola|Custom w/ ESP32-S2-WROVER 16MB|
|<B>Version/Date</B>|1.0.4|2.0.0|0badbeef|11/jul/2017|today's master|
|<B>IDE name</B>|Arduino IDE|Atom + Platform.io|IDF component|VSCode|
|<B>Flash Frequency</B>|40Mhz|80Mhz|
|<B>PSRAM enabled</B>|yes|no|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|Mac OSX|Ubuntu|

### Description:
I'm trying to use custom pins for SPI on ESP32-pico-v3 (pico mini-02 to be exact). But getting `Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled`.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

/* The ESP32 has four SPi buses, however as of right now only two of
 * them are available to use, HSPI and VSPI. Simply using the SPI API 
 * as illustrated in Arduino examples will use VSPI, leaving HSPI unused.
 * 
 * However if we simply intialise two instance of the SPI class for both
 * of these buses both can be used. However when just using these the Arduino
 * way only will actually be outputting at a time.
 * 
 * Logic analyser capture is in the same folder as this example as
 * "multiple_bus_output.png"
 * 
 * created 30/04/2018 by Alistair Symonds
 */
#include <SPI.h>

// Define ALTERNATE_PINS to use non-standard GPIO pins for SPI bus

#ifdef ALTERNATE_PINS
  #define VSPI_MISO   2
  #define VSPI_MOSI   4
  #define VSPI_SCLK   0
  #define VSPI_SS     33

  #define HSPI_MISO   26
  #define HSPI_MOSI   27
  #define HSPI_SCLK   25
  #define HSPI_SS     32
#else
  #define VSPI_MISO   MISO
  #define VSPI_MOSI   MOSI
  #define VSPI_SCLK   SCK
  #define VSPI_SS     SS

  #define HSPI_MISO   12
  #define HSPI_MOSI   13
  #define HSPI_SCLK   14
  #define HSPI_SS     15
#endif

static const int spiClk = 1000000; // 1 MHz

//uninitalised pointers to SPI objects
SPIClass * vspi = NULL;
SPIClass * hspi = NULL;

void setup() {
  //initialise two instances of the SPIClass attached to VSPI and HSPI respectively
  vspi = new SPIClass(VSPI);
  hspi = new SPIClass(HSPI);
  
  //clock miso mosi ss

#ifndef ALTERNATE_PINS
  //initialise vspi with default pins
  //SCLK = 18, MISO = 19, MOSI = 23, SS = 5
  vspi->begin();
#else
  //alternatively route through GPIO pins of your choice
  vspi->begin(VSPI_SCLK, VSPI_MISO, VSPI_MOSI, VSPI_SS); //SCLK, MISO, MOSI, SS
#endif

#ifndef ALTERNATE_PINS
  //initialise hspi with default pins
  //SCLK = 14, MISO = 12, MOSI = 13, SS = 15
  hspi->begin();
#else
  //alternatively route through GPIO pins
  hspi->begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_SS); //SCLK, MISO, MOSI, SS
#endif

  //set up slave select pins as outputs as the Arduino API
  //doesn't handle automatically pulling SS low
  pinMode(VSPI_SS, OUTPUT); //VSPI SS
  pinMode(HSPI_SS, OUTPUT); //HSPI SS

}

// the loop function runs over and over again until power down or reset
void loop() {
  //use the SPI buses
  vspiCommand();
  hspiCommand();
  delay(100);
}

void vspiCommand() {
  byte data = 0b01010101; // junk data to illustrate usage

  //use it as you would the regular arduino SPI API
  vspi->beginTransaction(SPISettings(spiClk, MSBFIRST, SPI_MODE0));
  digitalWrite(VSPI_SS, LOW); //pull SS slow to prep other end for transfer
  vspi->transfer(data);  
  digitalWrite(VSPI_SS, HIGH); //pull ss high to signify end of data transfer
  vspi->endTransaction();
}

void hspiCommand() {
  byte stuff = 0b11001100;
  
  hspi->beginTransaction(SPISettings(spiClk, MSBFIRST, SPI_MODE0));
  digitalWrite(HSPI_SS, LOW);
  hspi->transfer(stuff);
  digitalWrite(HSPI_SS, HIGH);
  hspi->endTransaction();
}
```
### Debug Messages:

Not sure if it debug messages, but this is displayed on the serial monitor
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40080f31  PS      : 0x00060730  A0      : 0x800d0f3d  A1      : 0x3ffb1f30
A2      : 0x0000000e  A3      : 0x00000002  A4      : 0x3ffb8364  A5      : 0x00000001
A6      : 0x00000000  A7      : 0x00000004  A8      : 0x3f400674  A9      : 0xaaaaaaaa
A10     : 0xaaaaaaaa  A11     : 0x00000030  A12     : 0x3ffb82cc  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000009  EXCCAUSE: 0x0000001c
EXCVADDR: 0xaaaaaaaa  LBEG    : 0x400d1258  LEND    : 0x400d1265  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x40080f31:0x3ffb1f30 0x400d0f3a:0x3ffb1f50 0x400d0db2:0x3ffb1f70 0x400d0c2b:0x3ffb1f90 0x400d190a:0x3ffb1fb0 0x40085fa5:0x3ffb1fd0

Rebooting...
ets Jul 29 2019 12:21:46
```

### Decoded backtrace, using the ep32 backtrace decode tool

```
0x40080f31: __pinMode at C:\Users\some\path\ArduinoData\packages\esp32\hardware\esp32\1.0.5\cores\esp32\esp32-hal-gpio.c line 115
0x400d0f3a: spiAttachSCK at C:\Users\some\path\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.5\cores\esp32\esp32-hal-spi.c line 87
0x400d0db2: SPIClass::begin(signed char, signed char, signed char, signed char) at C:\Users\some\path\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.5\libraries\SPI\src\SPI.cpp line 57
0x400d0c2b: setup() at C:\Users\some\path\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.5\libraries\SPI\examples\SPI_Multiple_Buses/SPI_Multiple_Buses.ino line 67
0x400d190a: loopTask(void*) at C:\Users\some\path\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.5\cores\esp32\main.cpp line 32
0x40085fa5: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```

Basic examples like blinking, serial writing are all working.

Thanks in advanced.

me-no-dev, 2021-09-08:
Have you tried this with ESP32 Arduino v2.0.0?

brahmajit, 2021-09-08:
Yes same

brahmajit, 2021-09-08:
@me-no-dev when using v2.0.0 there is no board named pico kit, what could be used instead ?

me-no-dev, 2021-09-08:
https://github.com/espressif/arduino-esp32/blob/2.0.0/boards.txt#L602 or the generic "ESP32 Dev Board"

brahmajit, 2021-09-08:
@me-no-dev could you suggest something on how to tackle this issue ? I would like to dig a little deeper, but kind of stuck

me-no-dev, 2021-09-08:
first try a pin other than `0` for SCK (this is what the backtrace suggests) and also IO0 is never a good idea for data pin. It usually is pulled up and have a cap to GND, which makes it not suitable for high frequency signals.

brahmajit, 2021-09-08:
Testing using different pins, same error. The mcu keeps restarting. Anything that uses the SPI header (like reading/writing to SD card) fails too. All the sketches are from the official example.

Can any maintainer/maintainers provide some insights.

Jason2866, 2021-09-08:
Have you tried if it works with the actual IDF from branch master?

brahmajit, 2021-09-08:
@Jason2866 No, I've not. I'll give it a try.

jbkuma, 2021-09-24:
I'm using an alternate SPI mapping with no issue, using CLK 14, MOSI 15, MISO 2, CS 12.  I'm using this mapping because it matches the pins for SDIO.  I have not had any issue with this at all, or any other pin maps I've used.

brahmajit, 2022-04-01:
> Hello @brahmajit <https://github.com/brahmajit> , sorry for late reply.
> Are you able to test your issue on development version 2.0.3-RC1
> <https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1> to
> check if this is still valid? You can take a look on Docs
> <https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html>
> where is explained how to choose development release version in Arduino IDE.

Hi Vojtěch Bartoška.

I replied to another comment of yours, this issue is (fixed for me).

VojtechBartoska, 2022-04-01:
Great, closing as well.

closed 2022-04-01 by VojtechBartoska