espressif/arduino-esp32 #5644: Noise triggers input button events with pullup/pulldown

mpresso, 2021-09-08:
I am facing a trouble that seems to be generated by AC coupling noise. 

I get button pushes events triggered by electrical noise. My button is connected to PIN 21 of ESP32. I tried configuring the input with internal an external PULL-UP, also tried with internal/external PULL-DOWN, and also including debounce code and hardware circuit, and still having false button pushes.

This trouble happens in different ESP32 boards, NodeMCU 32s, DEVKITV1, DEVKITV4

I generate noise pluging and unplugging AC loads and switching on and off.

I noticed that noisy AC coupling trigger button1.event on the main loop. 
CODE:

void loop() {
  int co2ppm = mhz19.getCO2();          // Request CO2 (as ppm)
  int temp = mhz19.getTemperature();    // Request Temperature (as Celsius)

  if (button1.event) {
      Serial.printf("Button has been pressed for %u millis\n", button1.timePressed);
      if (button1.timePressed < 1000)
        CO2_base = co2ppm;
      else calibrate_mhz19 ();
      // button state updated inside a critical section
      portENTER_CRITICAL(&mux);
      button1.event = false;
      portEXIT_CRITICAL(&mux);
  }

..... }

Complete code is: https://github.com/medicionco2/proyecto_medidor_co2/blob/main/src/mhz19/ESP32/IoT/wifwaflib_MHZ19b_IoT_thingspeak.ino

I´ll really preciate if someone can help me, can share a similar experience or have a useful advice.

RudyFiero, 2021-09-22:
This is not a software/code issue, it isn't about the ESP32. This is about the external wiring. This is not the place for this kind of question. Try a forum with general electrical help. You will need to describe what and how you connected things in your setup. Pictures would also be a help.

There is lots that can be done wrong that would make a circuit susceptible to electrical disturbances. One thing you should do is use a lower value pull up (or down) resistor. A good initial value would be 1K Ohm.

VojtechBartoska, 2022-04-07:
Hello, closing this one as expired. As mentioned above, we are not able to help on this issue.

closed 2022-04-07 by VojtechBartoska