espressif/arduino-esp32 #5645: ESP32-Arduino Sketch and Encryption with ESP-idf

Mos135, 2021-09-08:
Hi
I have written a complicated project with Arduino-IDE for ESP32, GPS and Display modules and it was working perfect.
After that I have decided to enable encryption and protection, to protect my program. but I find out that it is not possible to enable it through Arduino-IDE. I read many forums and have tried to use ESP-idf
but the problem was, I got many errors from other module's libraries, while I didn't have the same problem with Arduino-IDE.
then I have tried to compile my code with Arduino-IDE and only write the encrypted flash in the app partition and I got the error that it can not initialize EEPROM and SPIFF.
then I have tried to find out what the problem is. I write a simple code to use SPIFF and EEPROM. when I compile it in ESP-idf and then encrypt it and write it to the flash, it works perfectly and I get this message:

I (0) cpu_start: App cpu up.
I (145) cpu_start: Pro cpu start user code
I (145) cpu_start: cpu freq: 160000000
I (145) cpu_start: Application information:
I (150) cpu_start: Project name: main
I (154) cpu_start: App version: 1
I (159) cpu_start: Compile time: Aug 27 2021 14:32:23
I (165) cpu_start: ELF file SHA256: c81c9b73a2a94ecd...
I (171) cpu_start: ESP-IDF: v4.3-dirty
I (176) heap_init: Initializing. RAM available for dynamic allocation:
I (183) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (189) heap_init: At 3FFB3840 len 0002C7C0 (177 KiB): DRAM
I (196) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (202) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (208) heap_init: At 4008BB88 len 00014478 (81 KiB): IRAM
I (216) spi_flash: detected chip: generic
I (219) spi_flash: flash io: dio
W (224) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (233) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (261) nvs: NVS partition "nvs" is encrypted.
�EEPROM is Empty
EEPROM was initialised
SPIFFS mounted
Flash encryption enabled

but when I compile it with Arduino-IDE and then encrypt it and write it to the app partition, I got this error.

entry 0x400805e8
�failed to initialise EEPROM
EEPROM is Empty
failed to initialise EEPROM
EEPROM is Empty
E (4004) SPIFFS: spiffs partition could not be found
An Error has occurred while mounting SPIFFS
Flash encryption enabled

can anyone help about this issue?
please note that when I disable encryption, in both situation code works perfectly. also for my original project.

chegewara, 2021-09-08:
What bootloader do you flash with encryption enabled?

Mos135, 2021-09-08:
> What bootloader do you flash with encryption enabled?

bootloader from ESP-idf after enabling encryption

chegewara, 2021-09-09:
Did you try this scenario:
- enable encryption with esp-idf,
-  flash bootloader from esp-idf,
- flash non encrypted binary from arduino

As far as i remember esp32 should encrypt binary on first run. This is only theory i remember from other issues and topics on forum.

me-no-dev, 2021-09-09:
Arduino itself is not geared towards encryption. You need to use it as IDF component if you want such features.
SPIFFS on other hand can not be encrypted. You need to use FATFS is you want to encrypt the file system as well.
I can not comment on the EEPROM failure. It's in NVS and could be the same or similar issue.

Mos135, 2021-09-13:
> Did you try this scenario:
> 
> * enable encryption with esp-idf,
> * flash bootloader from esp-idf,
> * flash non encrypted binary from arduino
> 
> As far as i remember esp32 should encrypt binary on first run. This is only theory i remember from other issues and topics on forum.

I have Done so but have same error:
entry 0x400805e8
E (260) partition: No MState: restoring default values !
failed to initialise EEPROM
EEPROM is Empty !!
failed to initialise EEPROM
EEPROM is Empty !!
E (2265) partition: No MD5 found in partition table
E (2266) partition: load_partitions returned 0x105
E (2266) SPIFFS: spiffs partition could not be found
An Error has occurred while mounting SPIFFS

Mos135, 2021-09-16:
I have found the problem:
I was always changing the partition table for the nvs and partition table locations. 
after changing all to the default partition table and adding nvs_key location everything working fine!!

mrdc, 2021-09-21:
@Mos135 

Hi, can you explain in brief your steps to make encryption work for projects which use Arduino?

Mos135, 2021-09-21:
1- Using a simple project from ESP-IDF (for example Hello-World)

2- adding Arduino Components as following:

https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

3- adding suitable sdkconfig from Arduino folder to the project folder (from ..\components\arduino\tools\sdk\esp32 for ESP32)

4- adding customized partition.csv as follow to the project:

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x140000,
nvs_key,  data, nvs_keys,0x285000,0x1000,
spiffs,   data, spiffs,  0x290000,0x170000,

5- enabling the encryption and customized partition table by

idf.py menuconfig

6- building the projekt by:

idf.py build

7- copying the built partition-table.bin  (from ..\build\partition_table) , the built bootloader.bin (from ..\build\bootloader) and  ota_data_initial.bin (from ..\build) to the esptool folder (..\esp-idf\components\esptool_py\esptool) 

8- convert the Arduino sketch to binary by: Arduino IDE-> Sketch-> Export compiled binary (main.bin)

9- copying the main.bin to the esptool folder

10- creating own key

espsecure.py generate_flash_encryption_key my_flash_encryption_key.bin

11 - write the key in the module  

espefuse.py --port PORT burn_key flash_encryption my_flash_encryption_key.bin

12- write all the files to the module:

esptool.py -p PORT -b 921600 --before default_reset --after no_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader.bin 0x8000 partition-table.bin 0xe000 ota_data_initial.bin 0x10000 main.bin 0x290000 spiffs.bin

13- it will encrypt the flash and restart

to rewrite the module for development, use the development mode in step 5 and encrypt the main.bin file:

espsecure.py encrypt_flash_data --keyfile my_flash_encryption_key.bin --address 0x10000 --output main_en.bin main.bin

and write this part only:

esptool.py -p PORT -b 921600 write_flash --flash_mode dio 0x10000 main_en.bin

for me it works perfectly

pedros89, 2021-10-31:
Very nice exmplanation @Mos135 ,  you really are making thing easy.
Could you kindly explain how you managed to add arduino as IDF component in first place. I am struggling for 2 days and two nights to do so with all the IDF version and I have not managed to build a sketch 

Could you kindly exapand a bit this two very generic points, which versions did you use? At the moment seems to me that everything I try to use arduino as IDF component produce errors.
> 2- adding Arduino Components as following:
> 
> https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html
> 
> 3- adding suitable sdkconfig from Arduino folder to the project folder (from ..\components\arduino\tools\sdk\esp32 for ESP32)

thank you

Once I mange to encrypt the ESP I swear I will make a guide that will explain everything in super details for the community so hopefully people will not loose days trying like I'm doing at the moment.

pedros89, 2021-11-04:
After many efforts I've succesfully built the Arduino as ESP32 component ([see guide here](https://github.com/espressif/arduino-esp32/issues/4702#issuecomment-960296078))

And I installed also python2 ([see guide](https://datascience.com.co/how-to-install-python-2-7-and-3-6-in-windows-10-add-python-path-281e7eae62a)) to generate the key.bin file to avoid a error ocurring using python3 ([see Git thread](https://github.com/espressif/esp-idf/issues/5786))
To generate the key python2 some submodules are needed but can be installed as follow:
```
python2 -m pip install cryptography
python2 -m pip install ecdsa
python2 -m pip install pyserial
```
Up to here was everything fine. I did everything as sketchily exmpained by @Mos135 and I managed to connect to ESP32 and write the  key and the bins but I got the following error at the very end

`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
`

[image]

VojtechBartoska, 2022-04-07:
Hello @Mos135, can we close this as solved?

VojtechBartoska, 2022-04-21:
Hello,

as there was no answer in more than 14 days, I'm closing the issue as expired to keep our backlog manageable.

If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-21 by VojtechBartoska

simogaspa84, 2023-01-16:
> if(esp_flash_encryption_enabled()==true){
>     Serial.println("Flash encryption enabled");
>   } else {
>     Serial.println("Flash encryption disabled");
>   }

Hi.. I need i bootloader for my specs.. 

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1D0000,
app1,     app,  ota_1,   0x1E0000,0x1D0000,
spiffs,   data, spiffs,  0x3B0000,0x50000,

is it the procedure complicated? 

Can you guide me ? 

Thanks a lot for your help

pedros89, 2023-01-16:
> 

Basically with esp Idf, I suggest you to use ESP-IDF 5.0 CMD console you just need to create an encrypted project and copy the bootloader programmed to suit your needs
-select the flash frequency 40 MHz and the flash size 4MB
-the crypt flag enabled as release (and ‘none’ set as bootloader log info)
-replace the file in the Arduino bootloader folder you find at:
C:\Users\YOURUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0\tools\sdk\esp32\bin

If you do not much about esp `menuconfig`  look at this video or similar ones to get familiar for the IDF `menuconfig` settings console in Windows (or choose another video if you are on Linux) From `menuconfig` you can configure an IDF project with encrypted bootloader, after you have to use the command `build` like in the video, just copy paste that bootloader that will be vreated in the project folder in your Arduino folder where the esp bootloaders are and you are done. Just be very careful because you will not be able to reflash esp anymore. You will brick them. So make a back up of the original bootloader and use the encrypted bootloader with care.

[video link menuconfig](https://youtu.be/zGN8dTM9IPU?t=339)

simogaspa84, 2023-01-16:
Hi @pedros89 .. 
is it possible to encrypt also NVS partition...  I need to create a new bootloader that is able to encrypt the app partition and also the nvs partition where the data are stored..
Do you know if it is possible ? 

Thanks a lot

pedros89, 2023-01-16:
> Hi @pedros89 .. is it possible to encrypt also NVS partition... I need to create a new bootloader that is able to encrypt the app partition and also the nvs partition where the data are stored.. Do you know if it is possible ?
> 
> Thanks a lot

Sorry, from my experience only esp app partition will be encrypted with the randomly generated key by the bootloader. I am not sure about NVS partition, but I think it will not. I'm still using the old esp Virtual EEPROM library and the 4KB allocated are not encrypted in my case. Also SPIFFs partition is not encrypted with this method. I don't know how to encrypt those. If you want you can save sensible information (text or numbers) as encrypted with RC4 or similar methods and use an unique key you have in flash to decrypt when you retrieve the info, in this way is like having it encrypted.

simogaspa84, 2023-01-17:
Hi @pedros89 .. thanks for your suggestion.. when i create the custom bootloader with the menuconfig do I hvae also to change the partion table to be used ? I need a special one which is set in this file .. 

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1D0000,
app1,     app,  ota_1,   0x1E0000,0x1D0000,
spiffs,   data, spiffs,  0x3B0000,0x50000,

pedros89, 2023-01-17:
> Hi @pedros89 .. thanks for your suggestion.. when i create the custom bootloader with the menuconfig do I have also to change the partion table to be used ? I need a special one which is set in this file ..
> 
> # Name, Type, SubType, Offset, Size, Flags
> nvs, data, nvs, 0x9000, 0x5000, otadata, data, ota, 0xe000, 0x2000, app0, app, ota_0, 0x10000, 0x1D0000, app1, app, ota_1, 0x1E0000,0x1D0000, spiffs, data, spiffs, 0x3B0000,0x50000,

yes of course if you have a custom partition you have to modify the .csv
like thay say here on idf website
[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#creating-custom-tables](modify csv custom partition table).

whoim2, 2023-11-07:
Hi all! I'm trying to repeat the encryption procedure for the Lolin S2 mini board.
What was done:
- installed idf tools
- hello_world example copied to desktop
- via ifd.py target esp32s2 set
- enter the menuconfig with setting parameters as in the screenshots
- default.csv from `AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\tools\partitions` folder was loaded into the project directory as partitions.csv
- commas at the end of each line were removed, otherwise there was an error
- idf.py build

Before that, I uploaded the flash files that the Arduino uses to a my directory and wrote a cmd file for the firmware (use parameters from arduino log window), checked that the board was being flashed and the program was working.

I replaced the bootloader with the one generated above in idf tools. It is flashed, I perform a reset (automatically impossible for this board) and wait, up to 10 minutes. Nothing changes, rebooting the board does not launch the program. What could I have missed?

[image]
[image]
[image]
[image]
[image]

pedros89, 2023-11-07:
- Try not to use your customised directory but the Arduino loading arrow copying the code in Arduino IDE and uploading it. Or use Arduino directory as below that in my case works.

`esptool.exe --chip esp32 --port COM11 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xe000 C:\Users\YOUR_USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0/tools/partitions/boot_app0.bin 0x1000 C:\Users\YOUR_USER\AppData\Local\Temp\arduino_build_WHATEVER_NUMBER/MYFIRMWARE.ino.bootloader.bin 0x10000 C:\Users\YOUR_USER\AppData\Local\Temp\arduino_build_WHATEVER_NUMBER/MYFIRMWARE.ino.bin 0x8000 C:\Users\YOUR_USER\AppData\Local\Temp\arduino_build_WHATEVER_NUMBER/MYFIRMWARE.ino.partitions.bin `

- If you have the `hard resetting` message at the end of the process it means the encrypted bootloader and firmware are uploaded.

- Remove power to your board.

- Give power to the board again and the enctyption process should start. Wait 5 min, In my case it takes 2 and 1/2 minutes.

- To see if the encryption happened give the following command

`espefuse.py -p COM11 summary`

If you find 128 with many question mark as key as in the picture it means the chip is encrypted
[image]

By the way Iit is weird that commas at the end of each line were removed in the partition .csv as default partition schemes have it.

whoim2, 2023-11-09:
> By the way it is weird that commas at the end of each line were removed in the partition .csv as default partition schemes have it.

Thanks for the answer! I don’t think it’s a directory issue, I copied the same files that would interfere with the Arduino theme and used the command line from their Arduino log when flashing the firmware. The process is completely identical only The sketch does not compile, it was exported earlier. Everything works with the Arduino bootloader. I can't view the bootloader messages because The port is USB and disappears after flashing and resetting. No signs of work. The firmware lights up the LED at startup. This doesn't happen. I can flash the original bootloader since I did not exclude this possibility when setting up a new bootloader and chose development. I'll try to replace the bootloader in the Arduino, ok. But I don't think this will help. The busy ones had to be deleted because esptools showed a csv partitions file error.

whoim2, 2023-11-09:
hm, at `AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\tools\sdk\esp32\bin` present ELF files, not BIN.
[image]

whoim2, 2023-11-09:
I regenerated the bootloader and replaced Arduino15\packages\esp32\hardware\esp32\2.0.11\tools\sdk\esp32s2\bin\bootloader_dio_80m.elf with a new one, with boot encryption options enabled:

[image]

Then, as usual, I uploaded the sketch to Arduino. The sketch works as usual, instantly, and the total shows that all keys are at 00..00.

I chose dio 80m because this is what is indicated for the lolin s2 mini board in Arduino15\packages\esp32\hardware\esp32\2.0.11\boards.txt:
```
lolin_s2_mini.build.flash_size=4MB
lolin_s2_mini.build.flash_freq=80m
lolin_s2_mini.build.flash_mode=dio
```
I tried to find in the output of the Arduino compiler where it gets the bootloader, but was not successful, here is the output log: https://pastebin.com/GB95ayV4

I think the bootloader is coming from somewhere else.

whoim2, 2023-11-09:
if enable bootlog in arduino and find this: `esptool.exe" --chip esp32s2 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\whoim\\AppData\\Local\\Temp\\arduino_build_563386\\radio.ino.bootloader.bin" "C:\\Users\\whoim\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11\\tools\\sdk\\esp32s2\\bin\\bootloader_qio_80m.elf`
im generate (set menuconfig, build) new bootloader in idf and replace bootloader_qio_80m.elf. Flash esp with arduino, reset, wait 10 minutes, reset again. No live ;(

This idf menuconfig settings:
[image]

[image]

[image]

[image]

whoim2, 2023-11-09:
My problem is solved in this checkbox. We need to take it off.
After that, I uploaded the binaries via ESPTOOLZ (as I described earlier, in my directory), the bootloader encrypted the partition and rebooted the board, everything worked.
Thanks.

[image]

pedros89, 2023-11-09:
> My problem is solved in this checkbox. We need to take it off. After that, I uploaded the binaries via esptool.exe (as I described earlier, in my directory), the bootloader encrypted the partition and rebooted the board, everything worked. Thanks.
> 
> [image]
> 
> Addition: I chose the development version and left the permission to upload via USB. Last time this allowed me to return to the Arduino bootloader, now after flashing the firmware with a regular bootloader the sketch does not work. When re-downloading with a bootloader that encrypts, the sketch also does not work, I waited a long time. As far as I understand, if the key is registered in the fuse, you can’t download anything else via USB, only OTG. I have a download port, but I can't download anything else, the sketch doesn't work. As I understand it, the board is bricked :) In other words, if you flash it twice in the same way via USB, the board will be lost if you don’t know how to get permission to work via fuse back. If anyone can tell me, thanks.

Good that you got it sorted. I did not remember about that checkbox, infact, I did not touch anything, I left everything on default in `menuconfig` except selecting the flash frequency and the flash size with the crypt flag enabled as release and ‘none’ set as bootloader log info .

I confirm that if you have the bootloader with the encryption in developer mode, before uploading a new firmware via USB you need to burn an efuse with the following command
`espefuse.py -p COM7 burn_efuse FLASH_CRYPT_CNT`
More details in the idf guide
[Burn efuse guide](https://docs.espressif.com/projects/esptool/en/latest/esp32/espefuse/burn-efuse-cmd.html)
You can burn up to 3 times, on the fourth you will brick the esp.

On the other hand if you are working in release mode only OTA updates are possible to update the firmware, no more USB

At the end of the encryption process you can also try to read the flash and see if it is encrypted with this command

Read Falsh
2MB of flash
`esptool.py -p COM7 -b 460800 read_flash 0 0x200000 flash_contents.bin`
and read 4MB of flash
`esptool.py -p COM10 -b 460800 read_flash 0 0x400000 flash_contents.bin`

whoim2, 2023-11-09:
I set SPI_BOOT_CRYPT_CNT (0b001 -> 0b011) and it worked! As I understand it, I have one attempt left, but I can already try to work via OTA (I have provided for it in the firmware). I just would like to return this particular board to Arduino, if possible)

sboicu, 2024-01-21:
> 3- adding suitable sdkconfig from Arduino folder to the project folder (from ..\components\arduino\tools\sdk\esp32 for ESP32)

where do you find the sdkconfig in Arduino for the project you want to build?

pedros89, 2024-01-21:
> > 3- adding suitable sdkconfig from Arduino folder to the project folder (from ..\components\arduino\tools\sdk\esp32 for ESP32)
> 
> where do you find the sdkconfig in Arduino for the project you want to build? @pedros89 maybe do you know?

Are you looking for ?
C:\Users\YOUR_USER\AppData\Local\Arduino15
AppData is actually an hidden folder
This is the Arduino folder.