espressif/arduino-esp32 #5654: RAM Data Corruption when RX(GPIO3) Pin is not Connected to USB TTL

mightChamp, 2021-09-10:
Hardware: ESP32-WROOM-32E Module 4MB (ESP32-D0WD-V3 (revision 3))
Software: Arduino IDE with ESP32 Board V1.0.6

Configuration:
Selected Board: ESP32 Dev Module (esp32)
PSRAM: Disable
Partition Scheme: Minimal SPIFFS(1.9MB APP With OTA/190KB SPIFFS)
CPU Frequency: 240 MHz
Flash Mode: QIO
Flash Frequency: 80MHz
Flash Size: 4MB(32Mb)
Upload Speed:921600

Issue:

I have a 321 byte of file in SPIFFS, I read it in a 10KB char array type buffer. and Print it on Serial Port and TCP/IP Port, 4 times Without any operation on this Buffer.
Case 1: USB TTL Connected
All 4 times Buffer Printed Perfectly on both Serial Port and TCP/IP Port.

Case 2:  USB TTL Not Connected
Data on TCP/IP: 
1st time buffer Prints Perfectly.
2nd Time buffer print garbage data print.
3rd Time mostly garbage data print,
4th time most time garbage data print.

Issue is not generated in smaller size sample code. So can't share sample Code.

My Code build Analysis:
Sketch uses 1230490 bytes (62%) of program storage space. Maximum is 1966080 bytes.
Global variables use 90032 bytes (27%) of dynamic memory, leaving 237648 bytes for local variables. Maximum is 327680 bytes.

I tried to connect USB TTL pin one by one, first Connect TX Pin(ESP32) Only, Issue is generated. but when connect only TX Pin there is not issue. After that I have pulled up RX Pin with 4.7K. then my issue is solved permanently.
 
So, I want to know why this behavior  occurs? What are causes this type of issue. and what's it Solution.

chegewara, 2021-09-10:
When you leave RX pin not connected, that pin is floating, and there may be some electrical noise.
That noise can be interpreted as a data, which may eventually, for reasons i dont know now, lead to crash.
The solution is to initialize UART port properly with `RX = -1`, which is the info for UART peripheral this pin is not used.

dipakbhutada, 2021-09-13:
I have to use RX pin, as it is used to enter some configuration commands, So I can't initialize UART port with RX = -1. Issue occurs only when USB TTL is not connected to ESP32. Is there any reason behind this issue?

mightChamp, 2021-09-13:
I have to use RX pin, as it is used to enter some configuration commands, So I can't initialize UART port with RX = -1. Issue occurs only when USB TTL is not connected to ESP32. Is there any reason behind this issue?

mightChamp, 2021-09-18:
Hello,

I disconnect both TX and RX pin of USB TTL and ESP32. But I test my issue in below cases :
1) Connect Only TX Pin(ESP32) to USB TTL RX - Have Issue
2) Connect Only RX Pin(ESP32) to USB TTL TX  - No Issue
3) TX and RX both Connected to USB TTL - No Issue
4) Disconnect RX and TX both and pullup RX Pin(ESP32) - No Issue

So looking to know is there any reason for this type of behaviour.

Thank You

RudyFiero, 2021-09-22:
>After that I have pulled up RX Pin with 4.7K. then my issue is solved permanently.

This is the solution. You NEED to have control of the state of the input. Otherwise what can happen is what chegewara said, and what you have experienced. This is not a Code issue, it is an electrical circuit implementation problem.

chegewara, 2021-09-22:
Today i encountered similar issue with TX and RX connected.
I have connected SIM800L to esp32 and ive been getting bunch of garbage together with data i expected. After adding pullup all problems gone.

VojtechBartoska, 2022-03-30:
Hello @mightChamp  , sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

closed 2022-04-20 by VojtechBartoska