espressif/arduino-esp32 #5656: Enable progress bar on ArduinoOTA upload

tbertels, 2021-09-10:
Is there a reason why PROGRESS is disabled in espota.py?
https://github.com/espressif/arduino-esp32/blob/master/tools/espota.py#L44
https://github.com/espressif/arduino-esp32/blob/master/tools/espota.py#L308

I tested it with it enabled, it works fine.
The progress is shown based on the size of the file sent, so it should always work.

[image]
vs
[image]

me-no-dev, 2021-09-15:
the reason is that ArduinoIDE's console does not support carriage return, so it will print a whole lot of lines as percentage grows.

closed 2021-09-15 by me-no-dev

tbertels, 2021-09-15:
It actually does now.
Here are captures with the platform.txt patch:
[image]
[image]

When uploading by USB instead of OTA however, it still does indeed output multiple lines.

me-no-dev, 2021-09-16:
hmm interesting. will check it out. I reopened the pull request