espressif/arduino-esp32 #5658: analogRead(A10) WIDTH ERR

touchgadget, 2021-09-11:
### Hardware:
Board:							ESP32S2 Saola, ESP32S2 Dev Module
Core Installation version:			2.0.0
IDE name:						Arduino IDE 1.8.16
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Ubuntu 18.04

### Description:
Analog input on ADC2 returns WIDTH ERR. A0-A9 work fine.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
void setup() {
  Serial.begin(115200);
  Serial.println(analogRead(A10));
}

void loop() {
}
```

### Debug Messages:
```
E (90) ADC: adc2_get_raw(578): WIDTH ERR: see `adc_bits_width_t` for supported bit width
[   559][E][esp32-hal-adc.c:174] __analogRead(): GPIO11: ESP_ERR_INVALID_ARG
```

touchgadget, 2021-09-13:
Setting the default analogWidth to 4 for ESP32S2 seems to solve this problem. adc2_get_raw rejects all other values.
```
diff --git a/portable/packages/esp32/hardware/esp32/2.0.0/cores/esp32/esp32-hal-adc.c b/portable/packages/esp32/hardware/esp32/2.0.0/cores/esp32/esp32-hal-adc.c
index 0530f1d..72d051f 100644
--- a/portable/packages/esp32/hardware/esp32/2.0.0/cores/esp32/esp32-hal-adc.c
+++ b/portable/packages/esp32/hardware/esp32/2.0.0/cores/esp32/esp32-hal-adc.c
@@ -46,7 +46,11 @@ static uint8_t __analogVRefPin = 0;
 #endif
 
 static uint8_t __analogAttenuation = 3;//11db
+#if CONFIG_IDF_TARGET_ESP32
 static uint8_t __analogWidth = 3;//12 bits
+#elif CONFIG_IDF_TARGET_ESP32S2
+static uint8_t __analogWidth = 4;   // 13 bits
+#endif
 static uint8_t __analogClockDiv = 1;
 static adc_attenuation_t __pin_attenuation[SOC_GPIO_PIN_COUNT];
 

```

touchgadget, 2021-09-28:
@chegewara, @caternuson, @atanisoft, @ladyada 

The following patch make more sense because the fix is specific to ESP32S2. It compiles cleanly with Board Types=[ESP32 Dev Module | ESP32S2 Dev Module | ESP32C3 Dev Module] and Core Debug Level=Debug.

```
diff --git a/portable/packages/esp32/hardware/esp32/2.0.0/cores/esp32/esp32-hal-adc.c b/portable/packages/esp32/hardware/esp32/2.0.0/cores/esp32/esp32-hal-adc.c
index 0530f1d..8619a40 100644
--- a/portable/packages/esp32/hardware/esp32/2.0.0/cores/esp32/esp32-hal-adc.c
+++ b/portable/packages/esp32/hardware/esp32/2.0.0/cores/esp32/esp32-hal-adc.c
@@ -46,7 +46,11 @@ static uint8_t __analogVRefPin = 0;
 #endif
 
 static uint8_t __analogAttenuation = 3;//11db
+#if CONFIG_IDF_TARGET_ESP32S2
+static uint8_t __analogWidth = 4;   // 13 bits
+#else
 static uint8_t __analogWidth = 3;//12 bits
+#endif
 static uint8_t __analogClockDiv = 1;
 static adc_attenuation_t __pin_attenuation[SOC_GPIO_PIN_COUNT];

```

ladyada, 2021-09-28:
great @touchgadget can you do a PR so @me-no-dev can merge it?

touchgadget, 2021-09-28:
@ladyada OK, I will submit a PR but it is fine if @caternuson updates #5711.

caternuson, 2021-09-28:
ok, i pushed those changes

closed 2022-03-30 by VojtechBartoska