espressif/arduino-esp32 #5661: Why not use I2C IDF drivers for Vanilla ESP32?

ianwillianb, 2021-09-12:
Hi guys, 
I noticed that all new espressif chips like, ESP32-S2, and ESP32-C3 are using the I2C driver from IDF in esp32-i2c-hal in the 2.0 release, but if the target platform is "Vanilla" ESP32 the stickbreaker implementation is in use.
Is there an specific reason why not to switch to a single implementation using IDF drivers?
Just of out curiosity. 🙂
Thanks for the amazing work.

VojtechBartoska, 2021-09-13:
@SuGlider Can you please take a look on this behavior regarding I2C driver?

SuGlider, 2021-09-14:
@ianwillianb , This is exactly what I'm working on.
I just submitted a PR #5664 that shall implement you're talking about.
Could you please apply it to your environment and check if it support your needs for I2C?
Thanks.

ianwillianb, 2021-09-14:
> @ianwillianb , This is exactly what I'm working on.
> I just submitted a PR #5664 that shall implement you're talking about.
> Could you please apply it to your environment and check if it support your needs for I2C?
> Thanks.

That's really nice to know🙂
I'll definitely it give a try this week.

closed 2021-09-14 by ianwillianb