espressif/arduino-esp32 #5662: DNS entries not saved at all

Universumgames, 2021-09-12:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			        3.3.2
IDE name:						Platform.io
Flash Frequency:					default
PSRAM enabled:                                       default
Upload Speed:					        115200
Computer OS:                                          Windows 10

### Description:
By default, without adding my own code handling IPs and DNS entries, I've noticed that the EPS32 is not storing any IPs after a DNS request. I've noticed this behaviour in my pihole where my ESP sends about every two seconds (300 requests in 10 minutes) a DNS request to it and I've wondered why this rather bad performant behaviour is implemented into the HTTPClient Module.

### Sketch: 
```cpp
#include <Arduino.h>
#include <HTTPClient.h>

void setup() {
//setup wifi connection
}

void loop() {
    HTTPClient http;
    http.begin("http://google.com");
    http.post("");
    http.end();
}
```

VojtechBartoska, 2022-03-30:
Hello @Universumgames , sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-20:
Hello,

as there was no answer in more than 14 days, I'm closing the issue as expired to keep our backlog manageable. 

If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-20 by VojtechBartoska

Universumgames, 2022-06-03:
I'm sorry I didn't answer you, I was unable to reprogramm my ESPs so far to test the new firmware. Sadly upgrading to the newest framework version did not change any of the DNS behaviour and for every request there is still a DNS request preceeding it.

lbernstone, 2022-06-03:
The ESP32 has 520KB of RAM.  Since memory is so precious, it does no caching by default.  If you would like to cache the ip address, then do so.  https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiClient.cpp#L297-L298