espressif/arduino-esp32 #5663: Arduino as a ESP-IDF component Build error at esp32c3

lybgo, 2021-09-13:
### Hardware:
Board:							ESP32C3 Dev Module
Core Installation version:			V4.3.1
IDE name:						Vscode
Flash Frequency:					default
PSRAM enabled:                                    default
Upload Speed:					default
Computer OS:                                        Windows 10

### Description:
I try by course of https://blog.csdn.net/flurry_rain/article/details/119361997

main.cpp code:
```
#include "Arduino.h"

void setup(){
    Serial.begin(115200);
}

uint32_t mCounter = 0;
void loop(){
    Serial.println(String("loop:") + mCounter);
    mCount++;
    vTaskDelay(1000);
}
```

### Debug Messages:
```
> Executing task: cmake --build . <

[2/102] Performing build step for 'bootloader'
ninja: no work to do.
[9/100] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/HWCDC.cpp.obj
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/HWCDC.cpp.obj 
D:\esp\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe  -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/arduino/variants/esp32c3 -I../components/arduino/cores/esp32 -I../components/arduino/libraries/ArduinoOTA/src -I../components/arduino/libraries/AsyncUDP/src -I../components/arduino/libraries/BLE/src -I../components/arduino/libraries/BluetoothSerial/src -I../components/arduino/libraries/DNSServer/src -I../components/arduino/libraries/EEPROM/src -I../components/arduino/libraries/ESP32/src -I../components/arduino/libraries/ESPmDNS/src -I../components/arduino/libraries/FFat/src -I../components/arduino/libraries/FS/src -I../components/arduino/libraries/HTTPClient/src -I../components/arduino/libraries/HTTPUpdate/src -I../components/arduino/libraries/LittleFS/src -I../components/arduino/libraries/NetBIOS/src -I../components/arduino/libraries/Preferences/src -I../components/arduino/libraries/RainMaker/src -I../components/arduino/libraries/SD_MMC/src -I../components/arduino/libraries/SD/src -I../components/arduino/libraries/SimpleBLE/src -I../components/arduino/libraries/SPIFFS/src -I../components/arduino/libraries/SPI/src -I../components/arduino/libraries/Ticker/src -I../components/arduino/libraries/Update/src -I../components/arduino/libraries/USB/src -I../components/arduino/libraries/WebServer/src -I../components/arduino/libraries/WiFiClientSecure/src -I../components/arduino/libraries/WiFi/src -I../components/arduino/libraries/WiFiProv/src -I../components/arduino/libraries/Wire/src 
-I../components/arduino/cores/esp32/libb64 -ID:/esp/esp-idf/components/newlib/platform_include -ID:/esp/esp-idf/components/freertos/include -ID:/esp/esp-idf/components/freertos/port/riscv/include -ID:/esp/esp-idf/components/esp_hw_support/include -ID:/esp/esp-idf/components/esp_hw_support/port/esp32c3/. -ID:/esp/esp-idf/components/esp_hw_support/port/esp32c3/private_include -ID:/esp/esp-idf/components/heap/include -ID:/esp/esp-idf/components/log/include -ID:/esp/esp-idf/components/lwip/include/apps -ID:/esp/esp-idf/components/lwip/include/apps/sntp -ID:/esp/esp-idf/components/lwip/lwip/src/include -ID:/esp/esp-idf/components/lwip/port/esp32/include -ID:/esp/esp-idf/components/lwip/port/esp32/include/arch -ID:/esp/esp-idf/components/soc/include -ID:/esp/esp-idf/components/soc/esp32c3/. -ID:/esp/esp-idf/components/soc/esp32c3/include -ID:/esp/esp-idf/components/hal/esp32c3/include -ID:/esp/esp-idf/components/hal/include -ID:/esp/esp-idf/components/esp_rom/include -ID:/esp/esp-idf/components/esp_rom/esp32c3 -ID:/esp/esp-idf/components/esp_common/include -ID:/esp/esp-idf/components/esp_system/include -ID:/esp/esp-idf/components/esp32c3/include -ID:/esp/esp-idf/components/driver/include -ID:/esp/esp-idf/components/driver/esp32c3/include -ID:/esp/esp-idf/components/esp_ringbuf/include -ID:/esp/esp-idf/components/efuse/include -ID:/esp/esp-idf/components/efuse/esp32c3/include -ID:/esp/esp-idf/components/riscv/include -ID:/esp/esp-idf/components/espcoredump/include 
-ID:/esp/esp-idf/components/esp_timer/include -ID:/esp/esp-idf/components/esp_ipc/include -ID:/esp/esp-idf/components/esp_pm/include -ID:/esp/esp-idf/components/vfs/include -ID:/esp/esp-idf/components/esp_wifi/include -ID:/esp/esp-idf/components/esp_wifi/esp32c3/include -ID:/esp/esp-idf/components/esp_event/include -ID:/esp/esp-idf/components/esp_netif/include -ID:/esp/esp-idf/components/esp_eth/include -ID:/esp/esp-idf/components/tcpip_adapter/include -ID:/esp/esp-idf/components/spi_flash/include -ID:/esp/esp-idf/components/mbedtls/port/include -ID:/esp/esp-idf/components/mbedtls/mbedtls/include -ID:/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -ID:/esp/esp-idf/components/mdns/include -ID:/esp/esp-idf/components/console -ID:/esp/esp-idf/components/esp_adc_cal/include -ID:/esp/esp-idf/components/wifi_provisioning/include -ID:/esp/esp-idf/components/protocomm/include/common -ID:/esp/esp-idf/components/protocomm/include/security -ID:/esp/esp-idf/components/protocomm/include/transports -ID:/esp/esp-idf/components/nvs_flash/include -ID:/esp/esp-idf/components/nghttp/port/include -ID:/esp/esp-idf/components/nghttp/nghttp2/lib/includes -ID:/esp/esp-idf/components/fatfs/diskio -ID:/esp/esp-idf/components/fatfs/vfs -ID:/esp/esp-idf/components/fatfs/src -ID:/esp/esp-idf/components/wear_levelling/include -ID:/esp/esp-idf/components/sdmmc/include -ID:/esp/esp-idf/components/app_update/include -ID:/esp/esp-idf/components/bootloader_support/include -ID:/esp/esp-idf/components/spiffs/include -ID:/esp/esp-idf/components/openssl/include -ID:/esp/esp-idf/components/esp_https_ota/include -ID:/esp/esp-idf/components/esp_http_client/include -march=rv32imc   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Wno-error=format= -nostartfiles -Wno-format -Og -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu++11 -fno-exceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v4.3-dirty\" -DESP_PLATFORM -DARDUINO=10812 -DARDUINO_ESP32C3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32C3_DEV\" -DARDUINO_VARIANT=\"esp32c3\" -DESP32 -MD -MT esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/HWCDC.cpp.obj -MF esp-idf\arduino\CMakeFiles\__idf_arduino.dir\cores\esp32\HWCDC.cpp.obj.d -o esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/HWCDC.cpp.obj -c ../components/arduino/cores/esp32/HWCDC.cpp
../components/arduino/cores/esp32/HWCDC.cpp:25:10: fatal error: hal/usb_serial_jtag_ll.h: No such file or directory
 #include "hal/usb_serial_jtag_ll.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[18/100] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/BLE/src/GeneralUtils.cpp.obj  
ninja: build stopped: subcommand failed.
终端进程“C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake --build .”已终止，退出代码: 1。

终端将被任务重用，按任意键关闭。
```

atanisoft, 2021-09-13:
Arduino 2.0 / master has dependencies on ESP-IDF v4.4 (master). It will not compile cleanly against IDF v4.3 as it appears you are using.

lybgo, 2021-09-13:
@atanisoft The latest version of ESP-IDF is V4.3.1. When is V4.4 expected to be released? Is the Master branch V4.4?

atanisoft, 2021-09-13:
@lybgo Use the master branch of ESP-IDF and it should work. I don't know the timeline for v4.4 release but am also eagerly awaiting it for other bug fixes.

lybgo, 2021-09-13:
@atanisoft Thank you very much. I'll have a try.

k3rn3lp4n1c-1337, 2021-09-30:
hi I had the same problem.
I then also checked out the master from the Idf ToolChain in version 4.4.
But now I get this error when compiling:
```
../components/arduino/cores/esp32/esp32-hal-timer.c: In function 'timerAttachInterrupt':
../components/arduino/cores/esp32/esp32-hal-timer.c:292:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_ena'
             TIMERG1.int_ena.val &= ~BIT(timer->timer);
                    ^
../components/arduino/cores/esp32/esp32-hal-timer.c:296:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_clr'
             TIMERG1.int_clr.val = BIT(timer->timer);
                    ^
../components/arduino/cores/esp32/esp32-hal-timer.c:299:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_ena'
             TIMERG0.int_ena.val &= ~BIT(timer->timer);
                    ^
../components/arduino/cores/esp32/esp32-hal-timer.c:303:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_clr'
             TIMERG0.int_clr.val = BIT(timer->timer);
                    ^
../components/arduino/cores/esp32/esp32-hal-timer.c:336:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_ena'
             TIMERG1.int_ena.val |= BIT(timer->timer);
                    ^
../components/arduino/cores/esp32/esp32-hal-timer.c:338:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_ena'
             TIMERG0.int_ena.val |= BIT(timer->timer);
                    ^
[979/1099] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-uart.c.obj
../components/arduino/cores/esp32/esp32-hal-uart.c: In function 'uartStartDetectBaudrate':
../components/arduino/cores/esp32/esp32-hal-uart.c:503:17: warning: unused variable 'hw' [-Wunused-variable]
     uart_dev_t *hw = UART_LL_GET_HW(uart->num);
                 ^~
[980/1099] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-rmt.c.obj
../components/arduino/cores/esp32/esp32-hal-rmt.c:118:25: warning: 'g_rmt_objlocks' defined but not used [-Wunused-variable]
 static xSemaphoreHandle g_rmt_objlocks[MAX_CHANNELS] = {
                         ^~~~~~~~~~~~~~
[984/1099] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/ArduinoOTA/src/ArduinoOTA.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

chegewara, 2021-09-30:
1) you have to use matching arduino-esp32 and esp-idf versions
2) best to do it is to use arduino-esp32 `idf-master` branch

k3rn3lp4n1c-1337, 2021-10-01:
Hi chegewara  thx for your help.
But i still have a problem, i hope you or someone else can help me with it.
This is how I proceeded:

I look at arduino-esp32 idf-master repo and see the idf commit is from 3e370c4296:
```
 "UNITY_INCLUDE_CONFIG_H",
        "WITH_POSIX",
        "_GNU_SOURCE",
        ("IDF_VER", '\\"v4.4-dev-3224-g23a03a3bdb\\"'),
        ("IDF_VER", '\\"v4.4-dev-3235-g3e370c4296\\"'),  <=== Here i see the idf version
        "ESP_PLATFORM",
        "ARDUINO_ARCH_ESP32",
        "ESP32",
```
Then i clone the esp-idf repo with:
` git clone --recursive https://github.com/espressif/esp-idf.git`
check the commit out and set the head to this:
```
 git checkout v4.4-dev-3235-g3e370c4296
git reset --soft
```
then i install run the install.sh script with esp32 and esp32c3
then i run the export.sh script
then i create an new project with the idf tool and set the target to esp32c3
then i go into the project folder and run this

```
mkdir -p components && \
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git submodule update --init --recursive && \
cd ../.. && \
idf.py menuconfig
```

then i compile the complete project with idf.py all

all runs fine but at the end i run into the exception i post :(

It is also possible that I am doing something wrong. 
I hope someone can help me :)

chegewara, 2021-10-01:
Try to switch arduino component to use branch `idf-master`.
`master` branch is build with different esp-idf commit.

k3rn3lp4n1c-1337, 2021-10-01:
Hi could you tell me how I can proceed exactly so per git command?

I am unfortunately overwhelmed right now :(
Thanks a lot

me-no-dev, 2021-10-01:
`cd components/arduino && git checkout idf-master`

closed 2021-10-01 by me-no-dev

k3rn3lp4n1c-1337, 2021-10-01:
I tried it, it runs a little further but gets a failure right here again :/
```
./components/arduino/cores/esp32/libb64/cdecode.c:64:23: warning: this statement may fall through [-Wimplicit-fallthrough=]
         *plainchar    = (fragment & 0x003) << 6;
         ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
../components/arduino/cores/esp32/libb64/cdecode.c:65:7: note: here
       case step_d:
       ^~~~
[944/1098] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-rmt.c.obj
../components/arduino/cores/esp32/esp32-hal-rmt.c:118:25: warning: 'g_rmt_objlocks' defined but not used [-Wunused-variable]
 static xSemaphoreHandle g_rmt_objlocks[MAX_CHANNELS] = {
                         ^~~~~~~~~~~~~~
[945/1098] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-uart.c.obj
../components/arduino/cores/esp32/esp32-hal-uart.c: In function 'uartStartDetectBaudrate':
../components/arduino/cores/esp32/esp32-hal-uart.c:503:17: warning: unused variable 'hw' [-Wunused-variable]
     uart_dev_t *hw = UART_LL_GET_HW(uart->num);
                 ^~
[952/1098] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/Esp.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

chegewara, 2021-10-01:
In that case you did something wrong with esp-idf or arduino clone/checkout.
This works to me (maybe not with most recent commits, but about 2 weeks ago).

Also, those logs are only warnings, where is the log error?

k3rn3lp4n1c-1337, 2021-10-01:
Hi, i testet it with this workaround from another user:
```
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git checkout tags/2.0.0 && \
git submodule update --init --recursive
```
And then i can compiled it for older esp32 modules.
But when i compiled it for the esp32c3 module i became this errors (copy the completed compiler flow):

```
idf.py all  
Executing action: all (aliases: build)
Running ninja in directory /root/test/build
Executing "ninja all"...
[15/1099] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[584/1099] Performing configure step for 'bootloader'
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file /root/test/sdkconfig
-- Adding linker script /root/esp-idf/components/soc/esp32c3/ld/esp32c3.peripherals.ld
-- Adding linker script /root/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script /root/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script /root/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script /root/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script /root/esp-idf/components/bootloader/subproject/main/ld/esp32c3/bootloader.ld
-- Adding linker script /root/esp-idf/components/bootloader/subproject/main/ld/esp32c3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32c3 esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table riscv soc spi_flash
-- Component paths: /root/esp-idf/components/bootloader /root/esp-idf/components/bootloader_support /root/esp-idf/components/efuse /root/esp-idf/components/esp32c3 /root/esp-idf/components/esp_common /root/esp-idf/components/esp_hw_support /root/esp-idf/components/esp_rom /root/esp-idf/components/esp_system /root/esp-idf/components/esptool_py /root/esp-idf/components/freertos /root/esp-idf/components/hal /root/esp-idf/components/log /root/esp-idf/components/bootloader/subproject/main /root/esp-idf/components/bootloader/subproject/components/micro-ecc /root/esp-idf/components/newlib /root/esp-idf/components/partition_table /root/esp-idf/components/riscv /root/esp-idf/components/soc /root/esp-idf/components/spi_flash
-- Configuring done
-- Generating done
-- Build files have been written to: /root/test/build/bootloader
[619/1099] Performing build step for 'bootloader'
[1/3] Linking C executable bootloader.elf
[2/3] Generating binary image from built executable
esptool.py v3.2-dev
Merged 1 ELF section
Generated /root/test/build/bootloader/bootloader.bin
[3/3] cd /root/test/build/bootloader/esp-idf/esptool_py && /root/.espressif/python_env/idf4.4_py3.8_env/bin/python /root/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /root/test/build/bootloader/bootloader.bin
Bootloader binary size 0x4e20 bytes. 0x31e0 bytes (39%) free.
[935/1099] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-timer.c.obj
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-timer.c.obj 
/root/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/arduino/variants/esp32c3 -I../components/arduino/cores/esp32 -I../components/arduino/libraries/ArduinoOTA/src -I../components/arduino/libraries/AsyncUDP/src -I../components/arduino/libraries/BLE/src -I../components/arduino/libraries/BluetoothSerial/src -I../components/arduino/libraries/DNSServer/src -I../components/arduino/libraries/EEPROM/src -I../components/arduino/libraries/ESP32/src -I../components/arduino/libraries/ESPmDNS/src -I../components/arduino/libraries/FFat/src -I../components/arduino/libraries/FS/src -I../components/arduino/libraries/HTTPClient/src -I../components/arduino/libraries/HTTPUpdate/src -I../components/arduino/libraries/LittleFS/src -I../components/arduino/libraries/NetBIOS/src -I../components/arduino/libraries/Preferences/src -I../components/arduino/libraries/RainMaker/src -I../components/arduino/libraries/SD_MMC/src -I../components/arduino/libraries/SD/src -I../components/arduino/libraries/SimpleBLE/src -I../components/arduino/libraries/SPIFFS/src -I../components/arduino/libraries/SPI/src -I../components/arduino/libraries/Ticker/src -I../components/arduino/libraries/Update/src -I../components/arduino/libraries/USB/src -I../components/arduino/libraries/WebServer/src -I../components/arduino/libraries/WiFiClientSecure/src -I../components/arduino/libraries/WiFi/src -I../components/arduino/libraries/WiFiProv/src -I../components/arduino/libraries/Wire/src -I../components/arduino/cores/esp32/libb64 -I/root/esp-idf/components/newlib/platform_include -I/root/esp-idf/components/freertos/include -I/root/esp-idf/components/freertos/include/esp_additions/freertos -I/root/esp-idf/components/freertos/port/riscv/include -I/root/esp-idf/components/freertos/include/esp_additions -I/root/esp-idf/components/esp_hw_support/include -I/root/esp-idf/components/esp_hw_support/include/soc - [... cut ...]
[944/1099] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-uart.c.obj
../components/arduino/cores/esp32/esp32-hal-uart.c: In function 'uartStartDetectBaudrate':
../components/arduino/cores/esp32/esp32-hal-uart.c:507:17: warning: unused variable 'hw' [-Wunused-variable]
     uart_dev_t *hw = UART_LL_GET_HW(uart->num);
                 ^~
[952/1099] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/Esp.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

chegewara, 2021-10-01:
Ok, please try `git show` in `components/arduino`. It looks to me you still are on `master` branch.

k3rn3lp4n1c-1337, 2021-10-01:
@chegewara  I wanted to say thanks again for trying to get it up and running with me!!!!

git show from components/arduino
```
commit 44c11981d2159565c551028f2498f65764607f74 (HEAD, tag: 2.0.0)
Author: Unexpected Maker <seon@unexpectedmaker.com>
Date:   Tue Aug 31 16:28:56 2021 +1000

    Added UM FeatherS2 Neo to boards.txt and added appropriate variants. (#5615)

diff --git a/boards.txt b/boards.txt
index abc77f9e..fb1854d5 100644
--- a/boards.txt
+++ b/boards.txt
@@ -993,7 +993,7 @@ feathers2.build.variant=um_feathers2
 feathers2.build.board=FEATHERS2
 
 feathers2.build.cdc_on_boot=1
-feathers2.build.msc_on_boot=1
+feathers2.build.msc_on_boot=0
 feathers2.build.dfu_on_boot=0
 feathers2.build.f_cpu=240000000L
 feathers2.build.flash_size=16MB
@@ -1008,10 +1008,10 @@ feathers2.menu.CDCOnBoot.cdc.build.cdc_on_boot=1
 feathers2.menu.CDCOnBoot.default=Disabled
 feathers2.menu.CDCOnBoot.default.build.cdc_on_boot=0
 
-feathers2.menu.MSCOnBoot.msc=Enabled
-feathers2.menu.MSCOnBoot.msc.build.msc_on_boot=1
 feathers2.menu.MSCOnBoot.default=Disabled
 feathers2.menu.MSCOnBoot.default.build.msc_on_boot=0
+feathers2.menu.MSCOnBoot.msc=Enabled
+feathers2.menu.MSCOnBoot.msc.build.msc_on_boot=1
 
 feathers2.menu.DFUOnBoot.default=Disabled
 feathers2.menu.DFUOnBoot.default.build.dfu_on_boot=0
@@ -1108,7 +1108,131 @@ feathers2.menu.DebugLevel.verbose=Verbose
 feathers2.menu.DebugLevel.verbose.build.code_debug=5
 
 ##############################################################
+feathers2neo.name=UM FeatherS2 Neo
+feathers2neo.vid.0=0x303a
+feathers2neo.pid.0=0x80B4
+
+feathers2neo.upload.tool=esptool_py
+feathers2neo.upload.maximum_size=1310720
+feathers2neo.upload.maximum_data_size=327680
+feathers2neo.upload.flags=
+feathers2neo.upload.extra_flags=
+feathers2neo.upload.use_1200bps_touch=true
+feathers2neo.upload.wait_for_upload_port=true
+
+feathers2neo.serial.disableDTR=false
+feathers2neo.serial.disableRTS=false
+
+feathers2neo.build.tarch=xtensa
+feathers2neo.build.bootloader_addr=0x1000
```

chegewara, 2021-10-01:
Keep trying, you are on `v2.0.0` tag, not `idf-master` branch:
`commit 44c11981d2159565c551028f2498f65764607f74 (HEAD, tag: 2.0.0)`
use google to find solution.

k3rn3lp4n1c-1337, 2021-10-01:
@chegewara
Sorry i forgot to switch to this branch now i can compile... sorry!!!! but thx for your complete help :)

mbbender, 2021-11-05:
I'm having the same error on the esp32 using master idf and master arduino-esp32. 

> Users/michael/Code/idf/hello_world/components/arduino/cores/esp32/esp32-hal-timer.c:236:16: error: 'timg_dev_t' {aka 'struct timg_dev_t'} has no member named 'int_ena'