espressif/arduino-esp32 #5665: Sometimes after the watchdog takes effect, esp32 cannot start the program（Unable to restart）

houyawei-NO1, 2021-09-14:
Sometimes after the watchdog takes effect, esp32 cannot start the program（Unable to restart）

DOIT esp32 DEVKIT V1
deepin/windows 10
arduino-esp32_1.0.6
Library used：https://github.com/adafruit/Adafruit_SleepyDog

When the conditions are met, the program feeds the dog regularly, otherwise it does not feed the dog.
But sometimes after the watchdog restarts the device, the program cannot be loaded.
The logs printed during normal loading and abnormal loading are the same.

After the watchdog restarts the program：
`E (1690215) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (1690215) task_wdt:  - loopTask (CPU 1)
E (1690215) task_wdt: Tasks currently running:
E (1690215) task_wdt: CPU 0: IDLE0
E (1690215) task_wdt: CPU 1: loopTask
E (1690215) task_wdt: Aborting.
abort() was called at PC 0x40158874 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40088944:0x3ffbfa20 0x40088bc1:0x3ffbfa40 0x40158874:0x3ffbfa60 0x40087279:0x3ffbfa80 0x4016c563:0x3ffbc100 0x4015a123:0x3ffbc120 0x4008b381:0x3ffbc140 0x40089bd2:0x3ffbc160

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
#Now enter the program`

Sometimes the program cannot be loaded，The same log is printed。
`E (410181) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (410181) task_wdt:  - loopTask (CPU 1)
E (410181) task_wdt:  - Task1 (CPU 0)
E (410181) task_wdt: Tasks currently running:
E (410181) task_wdt: CPU 0: IDLE0
E (410181) task_wdt: CPU 1: loopTask
E (410181) task_wdt: Aborting.
abort() was called at PC 0x40158874 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40088944:0x3ffbfa20 0x40088bc1:0x3ffbfa40 0x40158874:0x3ffbfa60 0x40087279:0x3ffbfa80 0x4016c563:0x3ffbc100 0x4015a123:0x3ffbc120 0x4008b381:0x3ffbc140 0x40089bd2:0x3ffbc160

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
#The device freezes here and will not output anything unless it is powered on again`

Another question is, what is the time after rebooting? Why is the time different when compiling with the same environment?
`Rebooting...
ets Jun  8 2016 00:22:57
-----------------------------
Rebooting...
ets Jul 29 2019 12:21:46`

houyawei-NO1, 2021-09-15:
After testing, turn off the power of the router directly, and turn on the power of the router the next day. All esp32 are connected to the network normally.

Keep the router's wifi turned on and unplug the router's lan port network cable. Many esp32 will freeze and cannot be restored unless they are powered on again.

houyawei-NO1, 2021-09-16:
The problem has been solved, use rtc_wdt to solve this problem.

` RTC_WDT_STAGE_ACTION_OFF            = RTC_WDT_STG_SEL_OFF,          /*!< Disabled. This stage will have no effects on the system. */
    RTC_WDT_STAGE_ACTION_INTERRUPT      = RTC_WDT_STG_SEL_INT,          /*!< Trigger an interrupt. When the stage expires an interrupt is triggered. */
    RTC_WDT_STAGE_ACTION_RESET_CPU      = RTC_WDT_STG_SEL_RESET_CPU,    /*!< Reset a CPU core. */
    RTC_WDT_STAGE_ACTION_RESET_SYSTEM   = RTC_WDT_STG_SEL_RESET_SYSTEM, /*!< Reset the main system includes the CPU and all peripherals. The RTC is an exception to this, and it will not be reset. */
    RTC_WDT_STAGE_ACTION_RESET_RTC      = RTC_WDT_STG_SEL_RESET_RTC     /*!< Reset the main system and the RTC. */`

/*first*/
// esp_task_wdt_init(WDT_TIMEOUT, true); //enable panic so ESP32 restarts
// esp_task_wdt_add(NULL); //add current thread to WDT watch
// esp_task_wdt_reset();

/*second*/
//enableCore0WDT();
//enableCore1WDT();
//enableLoopWDT();

/*third*/
//https://github.com/adafruit/Adafruit_SleepyDog
//Watchdog.enable(4000);milliseconds
// Watchdog.reset();

/*fourth*/
//#include "soc/rtc_wdt.h"
// rtc_wdt_set_length_of_reset_signal(RTC_WDT_SYS_RESET_SIG, RTC_WDT_LENGTH_3_2us);
// rtc_wdt_set_stage(RTC_WDT_STAGE0, RTC_WDT_STAGE_ACTION_RESET_RTC);
// rtc_wdt_set_time(RTC_WDT_STAGE0, 2000);
// rtc_wdt_feed();

Capiten, 2022-03-18:
Hi @[houyawei-NO1](https://github.com/houyawei-NO1) 
I'm having the same problem, but I'm not really sure about what the answer is.
I don't get what the text before "//first/" is suppost to be, but is the following (on firest..second...third...and fourth) the full configuration of both task watchdogs and rtc watchdogs?
The feedings ( esp_task_wdt_reset(); , Watchdog.reset(); and   rtc_wdt_feed(); are the only ones that required to be regularly used in to avoid reboot correct?
Finally, the timeouts...I get the WDT_TIMEOUT (in seconds correct?) , 4000ms...and 2000 is milliseconds for the rtc watchdof timeout?

I now its a lot of questions, but please help me out.

Cheers, Luís Pereira.

VojtechBartoska, 2022-04-07:
Hello folks, are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? 

Take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

closed 2022-04-21 by VojtechBartoska