espressif/arduino-esp32 #5666: Are the ESP32-PICO-V3-02 chip or the ESP32-PICO-MINI-02 board supported?

brahmajit, 2021-09-14:
# Problem

I'm trying to use SPI for communicating with other modules connected to the board (for example a nrf24 and a SD card module) but it none of it works. I've tested both the modules on other development boards (like pico-d4 and wroom-32) and there the modules work flawlessly. I've tried setting custom SPI pins, that too didn't work.

It also seems like the board can configure the SPI pins by itself, if I don't `#define` the SPI pins, the board falls into a bootloop.

# OS Configuration
I'm using Arduino IDE with the esp32 boards plugin

# Hardware
ESP32-PICO-MINI-02 board

# What is failing
Anytime I try to use modules that require SPI connection, the device/module could not be initialized

# Sketch
```cpp
#include "FS.h"
#include "SD.h"
#include "SPI.h"

#define MOSI 34
#define MISO 38
#define CS 32
#define SCLK 35
#define CE 37

SPIClass hspi = SPIClass(HSPI);

void
setup()
{
	Serial.begin(115200);
	hspi.begin(SCLK, MISO, MOSI, CS); // sclk, miso, mosi, ss
	if (!SD.begin(CS, hspi)) {
		Serial.println("Card Mount Failed");
		return;
	}
}

void
loop()
{
}
```

or

```cpp
#include "FS.h"
#include "SD.h"
#include "SPI.h"

#define MOSI 34
#define MISO 38
#define CS 32
#define SCLK 35
#define CE 37

void
setup()
{
	Serial.begin(115200);
	SPI.begin(SCLK, MISO, MOSI, CS); // sclk, miso, mosi, ss
	if (!SD.begin(CS)) {
		Serial.println("Card Mount Failed");
		return;
	}
}

void
loop()
{
}
```

Both of the above code results in **Card Mount Failed**.

atanisoft, 2021-09-14:
pins 34-39 are input only, you can not use them for output.

VojtechBartoska, 2022-03-30:
Hello @brahmajit, is this still valid?

brahmajit, 2022-04-01:
Hi Vojtěch Bartoška.

I was able to fix this. May be it was something wrong with that perticular
version. But after a few updates, everything is working fine.

VojtechBartoska, 2022-04-01:
@brahmajit Perfect, congrats you figured that out. I'm closing the issue then.

closed 2022-04-01 by VojtechBartoska

rsramki, 2022-12-03:
> Hi Vojtěch Bartoška. I was able to fix this. May be it was something wrong with that perticular version. But after a few updates, everything is working fine.
> […](#)
> -- Regards *Brahmajit*

Let me know the fix as i am also facing the same issue.. Planning to configure SD CARD in VSPI. Anyone can guide please.

brahmajit, 2022-12-04:
@rsramki As I've said in my last comment, I don't exactly know what the issue was, but from arduino-esp32 2.0.3 I no longer face this issue. The following code (change the pins) has been working with my PICO-v3-03 device.

```c
#include "FS.h"
#include "SD.h"
#include "SPI.h"

#define MOSI 34
#define MISO 38
#define CS 32
#define SCLK 35

void
setup()
{
	Serial.begin(115200);
	SPI.begin(SCLK, MISO, MOSI, CS); // sclk, miso, mosi, ss
	if (!SD.begin(CS)) {
		Serial.println("Card Mount Failed");
		return;
	}
}

void
loop()
{
}
```

rsramki, 2022-12-23:
Thanks got it working by using VSPI with the following code.
HSPI was not being allowed as being used and was crashing if used.

#define SCK  22
#define MISO  19
#define MOSI  21
#define CS  5

SPIClass spi = SPIClass(VSPI);

spi.begin(SCK, MISO, MOSI, CS);

  if (!SD.begin(CS,spi,80000000)) {
    Serial.println("Card Mount Failed");
    return;
  }
  uint8_t cardType = SD.cardType();

  if(cardType == CARD_NONE){
    Serial.println("No SD card attached");
    return;
  }

misiek303, 2023-08-18:
I can't figure this out. 

**This is what I use:** 
-ESP32 WROOM-32 Board
-ESP32 Dev Moduel ( Arduino driver )
-Pico-ResTych-LCD-28 with SD

This is my ESP32 board
[image]

This is the Pico-ResTouch LCD
[image]

**Pinout Mapping:**
MISO - VSPI_MISO ( 19 )
MOSI - VSP_MOSI ( 23 )
CLK - VSP_CLK ( 18 )
SD_CS - VSP_CS
GND
3v2

**The Arduino Code**

#define SD_SCLK 18 
#define SD_MISO 19 
#define SD_MOSI 23 
#define SD_CS   5

Serial.begin(115200);
while (!Serial) {}

// Trying VSPI
SPIClass spi = SPIClass(VSPI);
spi.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);

if (!SD.begin(SD_CS,spi,80000000)) {
  Serial.println("Card Mount Failed VSPI");
} else {
  Serial.println("VSPI");
}

// trying SPI
SPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
if (!SD.begin(SD_CS)) {
  Serial.println("Card Mount Failed SPI");
} else {
  Serial.println("SPI");
}

I tried also HSPI and its pins, but the ESP32 hard boots for some reason.

I am able to use the LCD, and that works with no problem. But can't figure out the SD.
Can you help, any ideas?

Thanks