espressif/arduino-esp32 #5667: Possibility of Serialx based on uart_driver_install (ESP-IDF API)

imwhocodes, 2021-09-14:
Hi

I would like to use advance serial function to Arduino Serial

Reading through the source code any usage of `uart_driver_install` and seem to me that the serial is initialised in a different way from the one described in the [API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html)

Is the are any motivation to that?

I would like to use a custom configuration of RS485 mode, but Arduino Serial use `uart->dev->fifo.rw_byte` to write data to the ring buffer and not the `uart_write_bytes` from the API

Can you give some explanation about that?

Is possible to write an implementation using API reference functions?

Thanks,
Luca

bertmelis, 2021-09-14:
https://github.com/espressif/arduino-esp32/pull/5549

Isn't this already implemented (not released yet though)?

SuGlider, 2021-09-14:
> Hi
>
> I would like to use advance serial function to Arduino Serial
>
> Reading through the source code any usage of `uart_driver_install` and seem to me that the serial is initialised in a different way [...]

@imwhocodes 
I guess, from your post, that you may be using Arduino Core 1.0.6 - which uses ```uart->dev->fifo.rw_byte``` to write data to UART. Please move to recently released Arduino Core version 2.0.0 as this version is using IDF to implement Arduino Serial.
Maybe this new version will run as you need.

VojtechBartoska, 2021-11-18:
@imwhocodes Did you test your code with the latest version 2.0.1? :)

imwhocodes, 2021-12-09:
Yes I tested the code, it is working and it is running in production right now

I didn't answer before because i still want to run some test,
there is a strange behaviour were I receive data from serial in chunk
(Maybe it is excepted behaviour having the interrupt moving data from _harware RX buffer_ to _userspace RX buffer_  )

I would also like to modify some UART conf that are only exposed by `uart_struct_t` (mostly the size of the TX buffer) but i'm not able to compile having this error: `error: invalid use of incomplete type 'uart_t' {aka 'struct uart_struct_t'}`

Anyway the main quest of this issue seem woking correctly both in the old master _branch_ and the now released `2.0.1 tag`

Thanks for the help!

closed 2021-12-13 by SuGlider