espressif/arduino-esp32 #5668: v2.0.0 crash on xSemaphoreTake()

reppr, 2021-09-14:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:              v2.0.0 git 92ce408
IDE name:						Arduino IDE 1.8.15
Flash Frequency:				80Mhz
PSRAM enabled:                               no
Upload Speed:					921600
Partition Scheme:                            HUGE APP
Computer OS:                                   Debian

### Description:
I try to run my code from https://github.com/reppr/pulses.git
commit b43279a (specially configured to show the problem without additional hardware)
This code runs fine up to esp32-arduino v1.0.6
but crashes in v2.0.0 with a HW TIMER NEVER INIT ERROR
when calling xSemaphoreTake(MC_mux, portMAX_DELAY);
See file pulses/ePaper_GxEPD2.h line 379

Unfortunately I was not (yet) able to reproduce the crash with simplified code, but will show some relevant code below. See full code from https://github.com/reppr/pulses.git
I have got error messages and found information that looks related when trying to git-bisect the problem. Please see below.

Some explanation of the code:
The code is used to display on ePaper from the other core to avoid clicks in the music played from the app core. (I use similar code for other things too and get the same problem on v2.0.0)

```
// global code in multicore_display_common.h
SemaphoreHandle_t MC_mux = NULL;

// code from setup_ePaper_GxEPD2() called during setup()
  if(MC_mux == NULL)
    MC_mux = xSemaphoreCreateMutex();

// basic 1line ePaper output (can be called directly or from other core as shown here)
void ePaper_print_1line_at(uint16_t row, const char* text, int16_t offset_y=0) {
  ePaper.setRotation(1);
  ePaper.setTextColor(GxEPD_BLACK);
  ePaper.setFont(used_font_p);
  int16_t x = 0;
  int16_t y = row2y(row);
  y += offset_y;

  ePaper.fillRect(x, y - used_font_yAdvance + 4, ePaper.width(), used_font_yAdvance, GxEPD_WHITE);	// TODO test&trimm: -4
  ePaper.setCursor(x, y);
  ePaper.print(text);
  ePaper.display(true);
} // ePaper_print_1line_at()

#if defined MULTICORE_DISPLAY   // yes, do display from other core
TaskHandle_t ePaper_1line_at_handle;

// wrap the display code in a task that will be called from the other core
// crashes on v2.0.0!
void ePaper_1line_at_task(void* data_) {
  print_descrpt_t* data = (print_descrpt_t*) data_;   // will hold text and position data

  // possible workaround on v2.0.0 *rc1 only*
  //    add a delay(150);   here on the *very first time only* this is called
  //    but does *not* help on 2.0.0 version any more...

  xSemaphoreTake(MC_mux, portMAX_DELAY);	// line 379     esp32-arduino v2.0.0 CRASHES HERE!

  // set_used_font(used_font_p);
  ePaper_print_1line_at(data->row, data->text, data->offset_y);
  free_text_buffer(data);

  vTaskDelay(MC_DELAY_MS / portTICK_PERIOD_MS);
  xSemaphoreGive(MC_mux);
  vTaskDelete(NULL);
} // ePaper_1line_at_task()

// function to prepare text and position data for ePaper_1line_at_task() then create and start a one shot task:
void multicore_ePaper_1line_at(int16_t row, const char* text, int16_t offset_y) {
  print_descrpt_t* txt_descrpt_p = (print_descrpt_t*) malloc(sizeof(print_descrpt_t));
  if(txt_descrpt_p == NULL) {
    MENU.error_ln(F("txt_descript malloc()"));
    return;	// ERROR
  }
  copy_text_to_text_buffer(text, txt_descrpt_p);
  txt_descrpt_p->row = row;
  txt_descrpt_p->offset_y = offset_y;

  BaseType_t err = xTaskCreatePinnedToCore(ePaper_1line_at_task,		// function
					   "ePaper_1line_at",			// name
					   2000,				// stack size
					   txt_descrpt_p,			// task input parameter
					   MONOCHROME_PRIORITY,			// task priority
					   &ePaper_1line_at_handle,		// task handle
					   0);					// core 0
  if(err != pdPASS) {
    MENU.out(err);
    MENU.space();
    MENU.error_ln(F("ePaper_1line_at_task"));
  }
} // multicore_ePaper_1line_at()
#endif // MULTICORE_DISPLAY

```

### Debug Messages:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
E (417) timer_group: time⸮⸮u⸮⸮}⸮⸮չѕ⸮}م⸮Օ⸮54): HW TIMER NEVER INIT ERROR
assertion "( pxQueue )" failed: file "IDF/components/freertos/queue.c", line 1447, function: xQueueSemaphoreTake

abort() was called at PC 0x40180df3 on core 0

Backtrace:0x40102e01:0x3ffd1b200x4008dd29:0x3ffd1b40 0x400938a9:0x3ffd1b60 0x40180df3:0x3ffd1be0 0x4008edee:0x3ffd1c10 0x400d7298:0x3ffd1c50 

ELF file SHA256: 0000000000000000

exception decoder:

Decoding stack results
0x40102e01: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x4008dd29: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 129
0x400938a9: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c line 46
0x40180df3: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c line 62
0x4008edee: xQueueSemaphoreTake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1447
0x400d7298: ePaper_1line_at_task(void*) at /home/t2/Arduino/pulses/ePaper_GxEPD2.h line 379

```

reppr, 2021-09-14:
Frustrated not to be able to show the problem with short, simplified code I have tried to git bisect.
The result was unexpected, but might give a hint anyway.

checking out 1.0.6 as known good version:
```
git checkout 46d5afb1
```

Trying to compile does not work, but gives the following compile error:
```
In file included from /home/t2/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr.h:151,
                 from /home/t2/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/ext/atomicity.h:35,
                 from /home/t2/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/ios_base.h:39,
                 from /home/t2/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/ios:42,
                 from /home/t2/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/ostream:38,
                 from /home/t2/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/iostream:39,
                 from /home/t2/Arduino/pulses/pulses.ino:109:
/home/t2/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr-default.h: In function 'int __gthread_mutex_timedlock(__gthread_mutex_t*, const __gthread_time_t*)':
/home/t2/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr-default.h:781:12: error: 'pthread_mutex_timedlock' was not declared in this scope
     return __gthrw_(pthread_mutex_timedlock) (__mutex, __abs_timeout);
            ^~~~~~~~
/home/t2/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr-default.h:781:12: note: suggested alternative: 'pthread_mutex_trylock'
```

Robert Epprecht

chegewara, 2021-09-14:
The problem may be somewhere else (not saying it is).
Today i spent few hours chasing ghosts. I got crashes due to few different reasons, like not connected I2C sensor, but code was trying to access that device etc, but there was one thing that gave me a headache.
Very often crash in the same line of code, which was `printf` with watermark. I even tried to check if task handle is `NULL` and that didnt help. You not gonna believe what was the reason.
This 1 function call is causing some weird memory corruption:
`esp_wifi_80211_tx(WIFI_IF_STA, CCX_pkt, sizeof(CCX_pkt), false);`
As soon as i commented it out most of my issues gone.

Now your backtrace definitely is pointing into some problems with queue. I know you have this code:
```
  if(MC_mux == NULL)
    MC_mux = xSemaphoreCreateMutex();
```
but do you check if that mutex is actually created? If for some reason it is not then it is still `NULL` and you will try to take it anyway.

reppr, 2021-09-15:
> do you check if that mutex is actually created?
No I did not. And yes, things tend to go wrong in places you'd think it couldn't ever go wrong...

Unfortunately this was not the case here, so the issue is still open.

me-no-dev, 2021-09-15:
Error points [here](https://github.com/espressif/esp-idf/blob/b41d4b0a9c7162050da7ca4b961b68ca8d56ea30/components/freertos/queue.c#L1447). Your semaphore is NULL. You should init it in the task instead.

closed 2021-09-15 by me-no-dev