espressif/arduino-esp32 #5670: Can't connect to ESP32 STA_MODE while in a local network.

GeorgeFlorian, 2021-09-14:
I have an ESP32 and a laptop connected to a switch, making a local network with no internet access.

I have set the ESP in STA Mode, with an Ethernet connection:
`ETH.config(local_IP_STA, gateway_STA, subnet_STA, primaryDNS);`
The gateway is the same as the switch address.
The subnet is 255.255.255.0
And the DNS is 8.8.8.8, but it shouldn't matter because it won't go out onto the Internet.

**I can't connect to the ESP32 address.**
The same ESP32 works just fine in an office environment when it connects to a router with internet access.

Does the ESP require Internet Access ? Shouldn't it work on Local Network as well ?
The Laptop and ESP are both connected to the same switch.

VojtechBartoska, 2022-04-20:
Hello,

as there was no answer in more than 14 days, I'm closing the issue as expired to keep our backlog manageable. 

If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-20 by VojtechBartoska

GeorgeFlorian, 2022-04-20:
> Hello,
> 
> as there was no answer in more than 14 days, I'm closing the issue as expired to keep our backlog manageable.
> 
> If it's still needed, please reopen the issue.
> 
> Thanks for understanding.

I don't use ESP-IDF and it's harder for me to test development versions.
I use PlatformIO Core to program ESPs.

I've fixed this issue by simply adding a `delay(50);` after `ETH.begin();`. Basically wait a little until the network interface initializes.