espressif/arduino-esp32 #5672: Extend String to print 64-bit integers

drws, 2021-09-14:
### Description:

Even in the latest v2.0.0 `String()` is still unable to print 64-bit integers. A minimal non-working example would be:

### Sketch:
```cpp
#include <Arduino.h>

void setup() {
  String(1LL); // String(1L) compiles
}

void loop() {
}
```

Which produces an error `call of overloaded 'String(long long int)' is ambiguous`.

Rotzbua, 2021-09-17:
That is correct, because there is a core definition for all Arduino Frameworks which does not defines `String(long long int)`.
If you want that you have to create an issue here: https://github.com/arduino/ArduinoCore-API/

drws, 2021-09-22:
I have reported it upstream but it already has [fifteen](https://github.com/arduino/ArduinoCore-API/issues?q=is%3Aopen+is%3Aissue+string) up to 9-year-old unresolved String-related issues out of (currently) 88, majority of whom are at least a few **years** old. I'm leaving this issue open since this case provides a starting point to a possible discussion of Arduino Core reliability and more importantly, possible solutions for `arduino-esp32`.

Also, Arduino-ESP32 already extends the original Arduino Core ([an example, see discussion](https://github.com/arduino/ArduinoCore-samd/issues/457)), so manually extending String would actually not deviate from current practice at all. Besides that, String already has at least one deviation from Arduino Core, which are the filenames itself. If this project really inherits 100% of Arduino Core, `WString.*` would need to be renamed to `String.*` for example. And please don't do it! Besides unnecessarily breaking things, it introduces bugs such as [this](https://github.com/arduino/ArduinoCore-API/issues/37).

Unfortunately this is the current state of Arduino Core, where devs apparently rather introduce new bugs than fix at least the obvious old ones, such as [[1]](https://github.com/arduino/ArduinoCore-API/issues/40), [[2]](https://github.com/arduino/ArduinoCore-API/issues/48) and [[3]](https://github.com/arduino/ArduinoCore-API/issues/112).

SuGlider, 2022-05-18:
@drws - PR to add String(long long)/String(unsigned long long) submitted.

PR #6768 (2022-05-18): Extends String to print 64-bit integers
## Description of Change
This PR adds support to long long (64 bits) numbers to Arduino String Class.

## Tests scenarios
Tested with this sketch:

``` cpp
void setup()
{
  Serial.begin(115200);
  Serial.println();

  // highest signed long long 2^63 - 1 = 9223372036854775807
  // highest unsigned long long 2^64 = 18446744073709551615
  signed long long ll = -9223372036854775807;
  unsigned long long ull = 18446744073709551615;
  
  String ss(ll);
  Serial.println(ss);
  String uss(ull);
  Serial.println(uss);

  String txt("This is the highest signed lo [... cut]
closed 2022-05-18 by me-no-dev