espressif/arduino-esp32 #5675: Serial2 recieve garbage when SPI device conected

Gargantuanman, 2021-09-15:
What you are trying to do?
     Use all 3 serial ports, the Serial1 is mapped to pin 13 and 14. and Serial0 and 2 are standard mapped.

### Hardware:
Board:							node32
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Hello, I'm trying to use all 3 serial ports with the following code, I have been using this code, and worked perfectly, but today I connected an SPI Ethernet module that doesn't use any of the Serial pins but simply having the SPI device connected make Serial2 receive a lot of garbage. I checked the PCB and there are no hardware issues no short circuits or similar, If I Unplug the SPI module Serial works flawlessly, but if connected garbage returns only to Serial 2. I have look for a ESP32 limitation but havent found if that's the case.

Ethernet Module uses the following pins:
MISO 19
MOSI 23
ClK    18
Reset 26
CS      5

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
void setup() {
  Serial.begin(115200);
  Serial1.begin(57600, SERIAL_8N1, 13, 14);
  Serial2.begin(57600, SERIAL_8N1, 16, 17); //Serial2.begin(57600); I used like this originally

}

void loop() {

  if (Serial1.available()) {
    Serial.write(Serial1.read());
  }

  if (Serial2.available()) {
    Serial.write(Serial2.read());
  }
}
###

closed 2021-09-15 by Gargantuanman

SuGlider, 2021-09-15:
By coincidence or not, pins 13 and 14 are default HSPI MOSI and CLK.
Arduino SPI default pins are HSPI pins (12, 13, 14 and 15) 
https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.cpp#L37-L80

Try using other pins for Serial1.
By the way you can use default Serial1 pins since you set Arduino Flash Mode to DIO in the menu.

Gargantuanman, 2021-09-15:
> By coincidence or not, pins 13 and 14 are default HSPI MOSI and CLK.
> Arduino SPI default pins are HSPI pins (12, 13, 14 and 15) [...]

Thank you, I found that an unused reset pin connected to GPIO26 seems to be the one causing the trouble, I don't actually know why but now is working, also I think I will change from 13 14 to 26 27 that actually where the pins I originally used now that I know i don't require the hardware reset of Module.