espressif/arduino-esp32 #5676: ESP32-S2 SD card SPI mode on custom pins fails to mount?

Lawrence-RedFern, 2021-09-15:
### Hardware:
Board:							ESP32-S2 Custom Board
Core Installation version:			2.0.0
IDE name:						Arduino IDE/Visual Studio
Flash Frequency:					80MHz
PSRAM enabled:                                    No
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I am trying to run the [SD Card example](https://github.com/espressif/arduino-esp32/tree/master/libraries/SD/examples/SD_Test) on my ESP32-S2 custom board with custom assigned pins, but it does not seem to work. I have tried multiple variations of setting up the SPI but the SD card always fails to mount. Using SPI2/HSPI causes the program to hang at SPI.begin(), so I am only able to use SPI3. 

I have tried doing: 
```
SPIClass * sd_spi = NULL;
sd_spi = new SPIClass(3);
sd_spi->begin(PIN_SPI_SD_CLK, PIN_SPI_SD_MISO, PIN_SPI_SD_MOSI, -1);
pinMode(PIN_SPI_SD_CS, OUTPUT);
```
```
SPIClass * sd_spi = NULL;
sd_spi = new SPIClass(3);
sd_spi->begin(PIN_SPI_SD_CLK, PIN_SPI_SD_MISO, PIN_SPI_SD_MOSI, PIN_SPI_SD_CS);
```
```
SPIClass sd_spi = SPIClass(3);
sd_spi.begin(PIN_SPI_SD_CLK, PIN_SPI_SD_MISO, PIN_SPI_SD_MOSI, PIN_SPI_SD_CS);
```
```
SPIClass sd_spi = SPIClass(3);
sd_spi.begin(PIN_SPI_SD_CLK, PIN_SPI_SD_MISO, PIN_SPI_SD_MOSI, -1);
pinMode(PIN_SPI_SD_CS, OUTPUT);
```

Here is the whole sketch. I can see in the data sheet that SPI2 and SPI3 can be assigned to any GPIO pins, so I'm a little confused as to why these are not working. This sketch always returns from the line if (!SD.begin(PIN_SPI_SD_CS, sd_spi)), saying "Card Mount Failed". Upon entering this function I can debug and see that it is a mounting error, not an initialising error.

Any help on this would be appreciated.

```cpp

#include "FS.h"
#include "SD.h"
#include "SPI.h"

#define PIN_SPI_SD_CS	        20
#define PIN_SPI_SD_MOSI	21
#define PIN_SPI_SD_MISO	11
#define PIN_SPI_SD_CLK	        26

SPIClass sd_spi = SPIClass(3);
// SPIClass sd_spi = SPIClass(2);
// SPIClass sd_spi = NULL;

void setup() {
    Serial.begin(115200);
    Serial.println(F("SD Card Test"));
    
    sd_spi.begin(PIN_SPI_SD_CLK, PIN_SPI_SD_MISO, PIN_SPI_SD_MOSI, -1);
    pinMode(PIN_SPI_SD_CS, OUTPUT);
    Serial.println(F("SD SPI begin"));

    if (!SD.begin(PIN_SPI_SD_CS, sd_spi))
    {
        Serial.println("Card Mount Failed");
        return;
    }
    uint8_t cardType = SD.cardType();

    if (cardType == CARD_NONE) {
        Serial.println("No SD card attached");
        return;
    }

    Serial.print("SD Card Type: ");
    if (cardType == CARD_MMC) {
        Serial.println("MMC");
    }
    else if (cardType == CARD_SD) {
        Serial.println("SDSC");
    }
    else if (cardType == CARD_SDHC) {
        Serial.println("SDHC");
    }
    else {
        Serial.println("UNKNOWN");
    }

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);

}

void loop() {

}

chegewara, 2021-09-15:
For old esp32 please try to use `HSPI` or `VSPI`, i dont what values are asigned.
For esp32 S2 it is `FSPI`.
Here is full example:
https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino

Lawrence-RedFern, 2021-09-15:
> For esp32 S2 it is `FSPI`.
> Here is full example:
> https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino

``sd_spi = new SPIClass(FSPI); // This would evaluate to SPI1?``
Unfortunately this still results in 'Card Mount Failed' :(

chegewara, 2021-09-15:
I see you dont have SS/CS pin connected, maybe thats the reason?

chegewara, 2021-09-15:
`sd_spi.begin(PIN_SPI_SD_CLK, PIN_SPI_SD_MISO, PIN_SPI_SD_MOSI, -1);`
maybe dont pass `-1` for SS/CS pin? An make sure pin is setup before this line.

chegewara, 2021-09-15:
As you can see it works for me:
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/msc/sdcard.cpp#L104-L125

Lawrence-RedFern, 2021-09-15:
Here is a larger code snippet where I added in ``FSPI``.
``sd_spi = new SPIClass(FSPI);
    Serial.println(F("SPIClass enabled"));
    pinMode(PIN_SPI_SD_CS, OUTPUT);
    sd_spi->begin(PIN_SPI_SD_CLK, PIN_SPI_SD_MISO, PIN_SPI_SD_MOSI, PIN_SPI_SD_CS);
    //pinMode(PIN_SPI_SD_CS, OUTPUT);
    //sd_spi.setFrequency(250000);
    Serial.println(F("SD SPI begin"));

    if (!SD.begin(PIN_SPI_SD_CS, *sd_spi))
    {
        Serial.println("Card Mount Failed");
        return;
    }
``
Could it be the hardware set up and my choice of GPIO pins?
#define PIN_SPI_SD_CS	        20 <-- 47K Pull-up
#define PIN_SPI_SD_MOSI	21 <-- 47K Pull-up
#define PIN_SPI_SD_MISO	11 <-- 47K Pull-up
#define PIN_SPI_SD_CLK	        26 <-- 47K Pull-down

Lawrence-RedFern, 2021-09-15:
For some added information, the SD card I am trying to use is 16 GB

chegewara, 2021-09-15:
`• By default, IO26 is connected to the CS pin of the PSRAM and cannot be used for other functions.`
And pin20 is USB, but if you are not using any of it then should be fine.
I am using SD cards up to 32GB.

Lawrence-RedFern, 2021-09-15:
In my design I don't have it attached to the PSRAM. Here is an idea of how I have it wired 
[image]

Lawrence-RedFern, 2021-09-15:
I don't think this is an SPI issue, but is instead something to do with the SD card... Regardless, I am still confused as to why this isn't working! Please let me know if you have any ideas :)

chegewara, 2021-09-15:
How about verbose log level and some logs?
Its really hard to say. Sometimes pullup resistors are required.

Lawrence-RedFern, 2021-09-16:
Here is the verbose log:
```
SD SPI begin
[ 10591][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 10691][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 10791][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 10891][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 10891][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[ 10892][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 10901][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 11006][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 11106][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 11206][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
Card Mount Failed
```

chegewara, 2021-09-16:
There is few similar issues, maybe some will help:
https://github.com/espressif/arduino-esp32/issues?q=is%3Aissue+GO_IDLE_STATE+failed+is%3Aclosed
https://github.com/espressif/arduino-esp32/issues/505

lbernstone, 2021-09-16:
What does your SD module look like?  If it has an LDO on board, you will need to feed it 5V.

Lawrence-RedFern, 2021-09-16:
[image]
The signals feed directly into the ESP32-S2

Lawrence-RedFern, 2021-09-16:
Searching the internet for a few of the lines in my output log shows that a few people have had similar issues, but I am yet to read one that has a solid solution.

Unfortunately my software has started working, and I don't have a real solution.

The final sketch snippet can be seen below, and works with SPI2 or SPI1 (FSPI & HSPI). Adding in ``pinMode(PIN_SPI_SD_CS, OUTPUT);`` does not effect the sketch
```
sd_spi = new SPIClass(2);
Serial.println(F("SPIClass enabled"));
//pinMode(PIN_SPI_SD_CS, OUTPUT);
sd_spi->begin(PIN_SPI_SD_CLK, PIN_SPI_SD_MISO, PIN_SPI_SD_MOSI, PIN_SPI_SD_CS);
// sd_spi.setFrequency(250000);
Serial.println(F("SD SPI begin"));

if (!SD.begin(PIN_SPI_SD_CS, *sd_spi))
{
    Serial.println("Card Mount Failed");
    return;
}
```
It does not work with SPI3, and I'd love a solution as to why. The output is as follows
```
SD SPI begin
[ 10892][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 10892][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 10892][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[ 10897][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 11206][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 11206][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount Failed
```

Thank you @chegewara for your help, I'm sorry I can't give an answer as to why this has started working.

closed 2021-09-16 by Lawrence-RedFern

chegewara, 2021-09-16:
I dont remember, does S2 has 3 SPI ports?

Glad it is working for you.

Lawrence-RedFern, 2021-09-16:
It has 4 - SPI0/1, SPI2, SPI3

quikue, 2022-01-27:
I've spent all afternoon trying to plug in an SD module and haven't been able to get it to work.
Does anyone know how to do it?
I have used:
#define PIN_NUM_MISO 37
#define PIN_NUM_MOSI 36
#define PIN_NUM_CLK 35
#define PIN_NUM_CS 34

And both on the arduino platform, and the Espressif example in esp.idf do not work!
Thanks!

justp7, 2022-01-30:
> 我整个下午都在尝试插入 SD 模块，但无法让它工作。 有谁知道该怎么做？ 我用过： #define PIN_NUM_MISO 37 #define PIN_NUM_MOSI 36 #define PIN_NUM_CLK 35 #define PIN_NUM_CS 34
> 
> 在 arduino 平台上和 esp.idf 中的 Espressif 示例都不起作用！ 谢谢！
Did you succeed? I have the same problem now, also can't solve. I would be very grateful if you could make any progress

wibenjamin, 2022-02-12:
> I've spent all afternoon trying to plug in an SD module and haven't been able to get it to work. Does anyone know how to do it? I have used: #define PIN_NUM_MISO 37 #define PIN_NUM_MOSI 36 #define [...]

This code was working for me ( its (slower) software-spi):

`  // init SD-Card and check for Firmwareupdate

   SPIClass spi = SPIClass(HSPI);
   spi.begin(36 /* SCK */, 35 /* MISO */, 37 /* MOSI */, 38 /* SS */);
   uint8_t cardType;
   Serial.begin(115200);
   Serial.println("Welcome to the SD-Update example!");

   //first init and check SD card
   if (!SD.begin(38, spi, 40000000)) {
      rebootEspWithReason("Card Mount Failed");
   }`

quikue, 2022-05-13:
Thank you
I'm sorry I didn't answer earlier, I went to another project and forgot to check the answers.
I will try your instructions.