espressif/arduino-esp32 #5679: Garbage on Serial.read() if there are concurrent task

WiseroOrb, 2021-09-18:
PROGRAM:
In this program I'm sending data, the string "123456789" from PC to esp32 on Serial0
On the esp32 there are 2 task:
barTask: reads from serial and on newline sends the line back to the PC
fooTask: just a secondary task that increments a counter and then waits for 50ms

On the PC there is a little python program that
sends data to esp32
waits the response
check if it's equal
if not prints the response on the console

SETUP:
I'm using Arduino IDE v1.8.15 with https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json package
I'm using Ubuntu 21.04

ISSUE:
Every now and then I get garbage reading the serial on the esp32
what is send: "123456789"
what i receive: "\xff123456789"

I isolated the problem to this section
```cpp
static portMUX_TYPE my_mutex;
void barTask(void *p){
  vPortCPUInitializeMutex(&my_mutex);
  char gcode[100];
  int i=0;
  for(;;){
    vTaskDelay(1/portTICK_PERIOD_MS);
    while(Serial.available()){
      //portENTER_CRITICAL(&my_mutex);
      char c = Serial.read();
      //portEXIT_CRITICAL(&my_mutex);
      if(c == '\n'){
        gcode[i] = '\0';
        Serial.println(gcode);
        i=0;
        break;
      }
      gcode[i] = c;
      i++;
    }
  }
}
```
The incriminated line is:  char c = Serial.read();

CODE:
[code.zip](https://github.com/espressif/arduino-esp32/files/7189889/code.zip)

SOLUTION:
You can see that I've inserted in the code some mutex code. Decommenting that solves completely the problem.
For me this is very strange, because in the source code I can see that Serial.read already uses some mutex to lock access to the uart.

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Ubuntu

### Debug Messages:
No debug messages are sent

chegewara, 2021-09-18:
> what is send: "123456789"
> what i receive: "\xff123456789"

It looks to me it is send, but you dont know it is send with leading `0xff`. What terminal app are you using on PC?

WiseroOrb, 2021-09-18:
I'm using gnome terminal v3.38.1
However most of the response are correct, that is I receive : "123456789\r\n" , without \xff

WiseroOrb, 2021-09-18:
To clarify, the error occurs on the esp32, in the reading phase.
The send/println phase is always correct
If i send back to the pc the line two times, both are corrupted, meaning that the error is in the receiving phase

chegewara, 2021-09-18:
could you try with putty or even better with arduino IDE Serial console?

chegewara, 2021-09-18:
> To clarify, the error occurs on the esp32, in the reading phase.
> The send/println phase is always correct
> If i send back to the pc the line two times, both are corrupted, meaning that the error is in the receiving phase

I understand, but some terminal apps can send hidden bytes, depending how you set it up.

WiseroOrb, 2021-09-18:
Ok, I will try that.
However it does not explain why the problem is solved if I use mutex for the Serial.read()

WiseroOrb, 2021-09-18:
And I'm not using the terminal to send bytes to the esp32 but the serial library of python
Try to use the Arduino IDE is next to impossible  because the error happens every 1000 commands

WiseroOrb, 2021-09-18:
Ok it happens also with the Arduino IDE
I sent: 111111111111111111111111111111111111111111111
i rece: 111111111111111111111111111111111111⸮111111111
                                                                                   ^
Notice the question mark
By the way, this error happened after 68 lines, so it's quite frequent

chegewara, 2021-09-18:
Its really hard to help in situation where you cant provide enough data what is wrong. It can be problem with python library you are using.
Like i said, simplest test would be to use other terminal or app to confirm that esp32 has bug or even to use logic analyzer.
I am not saing there is no bug, but its not possible to find it with those information.

WiseroOrb, 2021-09-18:
> Ok it happens also with the Arduino IDE
> I sent: 111111111111111111111111111111111111111111111
> i rece: 111111111111111111111111111111111111⸮111111111
> ^
> Notice the question mark
> By the way, this error happened after 68 lines, so it's quite frequent

As I said here, the error occurs with the Arduino IDE

WiseroOrb, 2021-09-18:
> Ok, I will try that.
> However it does not explain why the problem is solved if I use mutex for the Serial.read()

And I've already found the solution, use mutex to block every other task running on both cores.
My problem is that is unclear to my why is necessary to do that, because in Hardware.cpp it already uses mutexes

chegewara, 2021-09-18:
Do you maybe have by any chance enabled some log level?
You are using Serial0 which is logging port.

WiseroOrb, 2021-09-18:
Can i use Serial0 both to receive data and send log?
If i enable log and i keep using serial0 in the program what happens?

chegewara, 2021-09-18:
What i meant is to not enable logs if you are using Serial0 to communicate with other devices. And it would be best to use Serial1, just in case to confirm it is issue with HardwareSerial library.

WiseroOrb, 2021-09-18:
Sorry I can't access uart1 from my PC

SuGlider, 2021-09-20:
@WiseroOrb,
(char) 0xff is actually (int8) -1, returned by Serial.read() when there is no data available yet.
This is due to the fact that SerialHardware isn't threadsafe.
That also explains that using portENTER_CRITICAL / portEXIT_CRITICAL around Serial.read solves the problem you are having.

WiseroOrb, 2021-09-28:
> @WiseroOrb, (char) 0xff is actually (int8) -1, returned by Serial.read() when there is no data available yet. This is due to the fact that SerialHardware isn't thread-safe, but it should occur only when more than 1 task is reading the same UART. That may explain that using portENTER_CRITICAL / portEXIT_CRITICAL around Serial.read solves the problem you are having.

Interesting, however in the code I've uploaded only one task is accessing the UART. Maybe Arduino IDE compiles the final code with some hidden task? I don't have the knowledge to troubleshoot this.

Maybe it should be added a note in the documentation.

SuGlider, 2021-09-29:
ESP32 Arduino always creates a task for loop().

SuGlider, 2021-09-29:
@WiseroOrb 

Could you try running this modified task and check if the issue still occurs?

```
void barTask2(void *p) {
  char gcode[100];
  int i = 0;
  for (;;) {
    vTaskDelay(1 / portTICK_PERIOD_MS);
    while (Serial.available()) {
      int c = Serial.read();
      if (c != -1)  {
        if (c == '\n') {
          gcode[i] = '\0';
          Serial.println(gcode);
          i = 0;
          break;
        }
        gcode[i] = c;
        i++;
      }
    }
  }
}

```

WiseroOrb, 2021-09-30:
> @WiseroOrb,
> 
> ESP32 Arduino always creates a separated task for loop().
> 
> I tested your sketch, as is, with the python program running on the PC. I've never got an error for almost 4 hours running.
> 
> Such issue will only happen when more than one task reads the same UART port. Otherwise if read() is executed only in one task, as you reported, that issue shall never happen.
> 
> Right after loop() it may also call serialEvent(). If Serial is read inside serialEvent() it can compete with another task reading the same UART.
> 
> **Question: What is the core version you are running this sketch? Everything I wrote is regarding Core 2.0.0.**

I'm running 1.0.6. Is it safe to upgrade to 2.0.0? Do I have to migrate code?

> @WiseroOrb
> 
> Could you try running this modified task and check if the issue still occurs?
> 
> ```c++
> void barTask2(void *p) {
>   char gcode[100];
>   int i = 0;
>   for (;;) {
>     vTaskDelay(1 / portTICK_PERIOD_MS);
>     while (Serial.available()) {
>       int c = Serial.read();
>       if (c != -1)  {
>         if (c == '\n') {
>           gcode[i] = '\0';
>           Serial.println(gcode);
>           i = 0;
>           break;
>         }
>         gcode[i] = c;
>         i++;
>       }
>     }
>   }
> }
> ```

With this code I get no more problem.

SuGlider, 2021-10-01:
> I'm running 1.0.6. Is it safe to upgrade to 2.0.0? Do I have to migrate code?

No, you don't have to migrate, but 2.0.0 works better than 1.0.6 regarding the issue you have posted.

There are many changes from 1.0.6 to 2.0.0. I can't tell you if it is safe or not to upgrade to 2.0.0.
It'll depend on your application. 
You can try it and test it. If 2.0.0 breaks your application, all you need to do is to downgrade back to 1.0.6.

WiseroOrb, 2021-10-02:
Thank you for being so helpful!

closed 2021-10-02 by WiseroOrb