espressif/arduino-esp32 #5680: RAM allocation issue when using BT + MQTT

olicooper, 2021-09-19:
### Hardware:
Board:							Lolin 32 Lite / Heltec Wireless Stick
Core Installation version:			1.0.6
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:

I am unable to get Bluetooth or BLE running at the same time as MQTT due to RAM running out. The amount of memory used seems excessive. I created a simple sketch to illustrate the issue.

I am using Arduino libraries except the `mqtt_client` implementation. I would have edited the buffer allocations for WiFi/Bluetooth but ESP-IDF is precompiled for Arduino. I have a much larger project written in Arduino so I can't just switch to IDF at this point without some serious headaches.

Thanks 👍 

### Sketch:
See: https://github.com/olicooper/arduino-bt-mqtt-test

killgroove, 2021-09-23:
Hello,

I was confronted with exactly the same problem recently, and I was forced to switch to the NimBLE library to better manage this memory problem. 
If you don't have any particular constraint that prevents you from changing your BLE library, you should know that I can perfectly use MQTT and BLE now.
https://github.com/h2zero/NimBLE-Arduino

Fortunately, the process of switching from one library to another is relatively easy, and there is a guide for that exact purpose.
https://github.com/h2zero/NimBLE-Arduino/blob/master/docs/Migration_guide.md

Regards,

Alex

olicooper, 2021-09-24:
Thanks for the tip! I am also switch to full BT for serial logging, but I might be able to use the BLE serial implementation instead. 👍

chegewara, 2021-09-24:
Yes, if you cant handle problems with memory shortage it is advised to use NimBLE instead.
I am working now on bigger project and here is memory usage statistics:
```
Printing StackHighWaterMark
        Task: TaskBLE 1492
        Task: TaskLED 1072
        Task: TaskButton 1488
        Task: TaskMQTT 1280
        Task: TaskConfiguration 460
        Task: TaskUWB 1556on 
        Task: TaskIMU 1328
        Task: TaskUpdateRealTime 1248
        Task: TaskMobilaris 988
        Task: TaskMAIN 1176 
Heap summary for capabilities 0x00000006:
  At 0x3ffb2730 len 15448 free 132 allocated 12732 min_free 128
    largest_free_block 0 alloc_blocks 35 free_blocks 0 total_blocks 35
  At 0x3ffcee04 len 32767 free 215 allocated 30100 min_free 111
    largest_free_block 52 alloc_blocks 24 free_blocks 1 total_blocks 25
  At 0x3ffb6388 len 7288 free 212 allocated 4348 min_free 140
    largest_free_block 0 alloc_blocks 47 free_blocks 2 total_blocks 49
  At 0x3ffb9a20 len 16648 free 176 allocated 13804 min_free 112
    largest_free_block 0 alloc_blocks 42 free_blocks 1 total_blocks 43
  At 0x3ffcd6a0 len 76128 free 568 allocated 71716 min_free 568
    largest_free_block 0 alloc_blocks 140 free_blocks 1 total_blocks 141
  At 0x3ffe0440 len 15072 free 72 allocated 12596 min_free 60
    largest_free_block 0 alloc_blocks 20 free_blocks 0 total_blocks 20
  At 0x3ffe4350 len 113840 free 34616 allocated 75248 min_free 28268
    largest_free_block 16372 alloc_blocks 151 free_blocks 8 total_blocks 159
  Totals:
    free 34147 allocated 222364 min_free 29387 largest_free_block 16372
```

As you can see there is 10 running tasks with:
- WIFI + LTE (pppos) + BLE + MQTT, some I2C master with 2 devices, I2C slave and dont know what else
- code is platformio, arduino-esp32 v2.0.0+
- mqtt library is plain esp-idf

To make it all works i had to move to nimble, and do a lot of optimization, but it was worth it. There is still a lot of work to do.

olicooper, 2021-09-24:
I'm now in the process of moving over to Nimble and it's really shocking to see how much memory Bluedroid uses in comparison!

closed 2021-10-15 by olicooper