espressif/arduino-esp32 #5684: WiFiEvent event not called on WiFi.onEvent(WiFiEvent) using Lan8720 Ethernet.

rjjrbatarao, 2021-09-20:
Im trying to migrate from old 1.0.6 Board to this 2.0 version but found out WiFi.onEvent(WiFiEvent) not working anymore on the same sketch that is perfectly working on 1.0.6, and even with the minimal example included with Lan8720 doesnt work either. 

### Hardware:
Board:							TTGO T-POE Esp32 Wroom
Core Installation version:			2.0
IDE name:						Arduino IDE
Flash Frequency:					40Mhz?
PSRAM enabled:                                    no
Upload Speed:					115200?
Computer OS:                                        Windows 10

### Description:
Debug show its working but none of the serial prints showing from the sketch. WiFiEvent callback not called.

```cpp
#include <ETH.h>
#include <WiFi.h>
#include <SPI.h>
#include <SD.h>
#include <WiFiClientSecure.h>

#define SD_MISO         2
#define SD_MOSI         15
#define SD_SCLK         14
#define SD_CS           13
/*
     ETH_CLOCK_GPIO0_IN   - default: external clock from crystal oscillator
     ETH_CLOCK_GPIO0_OUT  - 50MHz clock from internal APLL output on GPIO0 - possibly an inverter is needed for LAN8720
     ETH_CLOCK_GPIO16_OUT - 50MHz clock from internal APLL output on GPIO16 - possibly an inverter is needed for LAN8720
     ETH_CLOCK_GPIO17_OUT - 50MHz clock from internal APLL inverted output on GPIO17 - tested with LAN8720
*/
// #define ETH_CLK_MODE    ETH_CLOCK_GPIO0_OUT          // Version with PSRAM
#define ETH_CLK_MODE    ETH_CLOCK_GPIO17_OUT            // Version with not PSRAM

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   -1

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720

// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        0

// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     23

// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    18

#define NRST            5
static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testSslClient(IPAddress host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);
  Serial.print("connecting to ");
  Serial.println(host);
  WiFiClientSecure client;
  client.setInsecure();
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  } else {
    Serial.println("connected");
  }
  client.stop();
}

void setup()
{
  Serial.begin(115200);

  WiFi.onEvent(WiFiEvent);

  SPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
  if (!SD.begin(SD_CS)) {
    Serial.println("SDCard MOUNT FAIL");
  } else {
    uint32_t cardSize = SD.cardSize() / (1024 * 1024);
    String str = "SDCard Size: " + String(cardSize) + "MB";
    Serial.println(str);
  }

  pinMode(NRST, OUTPUT);
  digitalWrite(NRST, 0);
  delay(200);
  digitalWrite(NRST, 1);
  delay(200);
  digitalWrite(NRST, 0);
  delay(200);
  digitalWrite(NRST, 1);

  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);

}
```

### Debug Messages:
```
[   412][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[   412][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[   412][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[   417][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
SDCard MOUNT FAIL
[  2718][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 18 - ETH_START
[  2718][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 20 - ETH_CONNECTED
[  5526][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 22 - ETH_GOT_IP
[  5526][D][WiFiGeneric.cpp:893] _eventCallback(): ETH IP: 10.10.10.154, MASK: 255.255.255.0, GW: 10.10.10.1
```

me-no-dev, 2021-09-20:
Have a look here: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino

closed 2021-09-20 by me-no-dev

brandonjbates, 2024-08-14:
Just so it's clear, you have to change SYSTEM_EVENT_xxxxx to ARDUINO_EVENT_xxxxx when migrating from 1.x ESP board package to 2.x ESP board package.