espressif/arduino-esp32 #5689: How to troubleshoot and fix Guru Meditation Error: Core 1 panic'ed (StoreProhibited). Exception was unhandled. Arduino?

djornfevrier, 2021-09-21:
### Hardware:
Board:							?ESP32 Dev Module Tiny Pico?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE?
Flash Frequency:					?80Mhz?
PSRAM enabled:                                    ?yes?
Upload Speed:					?921600?
Computer OS:                                        ?Windows 10?

### Description:
I am trying to use the sparkfun AS7265X sensor with a serial or wifi connection to the board. I am using the latest version of Arduino 1.8.15 running on an ESP32 Tiny Pico. I get this error message (Guru Meditation Error: Core 1 panic'ed (StoreProhibited). Exception was unhandled.) every time I try to utilize the sensor with Serial or Wifi communication.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include "SparkFun_AS7265X.h"
AS7265X sensor;

void setup() {
  Serial.begin(115200);
  Serial1.begin(115200);
}

void loop() {
  sensor.takeMeasurements(); //This is a hard wait while all 18 channels are measured

  Serial.print(sensor.getCalibratedA());
  Serial.print(",");

  Serial.println();

    if (Serial.available()) {      // If anything comes in Serial (USB),
    Serial1.write(Serial.read());   // read it and send it out Serial1 (pins 0 & 1)
  }

  if (Serial1.available()) {     // If anything comes in Serial1 (pins 0 & 1)
    Serial.write(Serial1.read());   // read it and send it out Serial (USB)
  }
}
```

### Debug Messages:
```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4014bca0  PS      : 0x00060e30  A0      : 0x800d7f4d  A1      : 0x3ffb1ec0  
A2      : 0x00000000  A3      : 0x00000049  A4      : 0x3ffb8218  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000100  A9      : 0x00000001  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060c20  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000126  LBEG    : 0x4008c67d  LEND    : 0x4008c68d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x4014bca0:0x3ffb1ec0 0x400d7f4a:0x3ffb1ee0 0x400d8035:0x3ffb1f00 0x400d8088:0x3ffb1f20 0x400d809b:0x3ffb1f40 0x400d80e9:0x3ffb1f60 0x400d0b05:0x3ffb1f80 0x400da7b1:0x3ffb1fb0 0x4008e1fe:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4 
```

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
`PC: 0x4014bce4: TwoWire::beginTransmission(int) at C:\Users\\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\Wire\src\Wire.cpp line 156
EXCVADDR: 0x00000126

Decoding stack results
0x4014bce4: TwoWire::beginTransmission(int) at C:\Users\\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\Wire\src\Wire.cpp line 156
0x400d7f8e: AS7265X::readRegister(unsigned char) at C:\Users\\Documents\Arduino\libraries\SparkFun_Spectral_Triad_AS7265X\src\SparkFun_AS7265X.cpp line 539
0x400d8079: AS7265X::virtualWriteRegister(unsigned char, unsigned char) at C:\Users\\Documents\Arduino\libraries\SparkFun_Spectral_Triad_AS7265X\src\SparkFun_AS7265X.cpp line 516
0x400d80cc: AS7265X::selectDevice(unsigned char) at C:\Users\\Documents\Arduino\libraries\SparkFun_Spectral_Triad_AS7265X\src\SparkFun_AS7265X.cpp line 400
0x400d80df: AS7265X::getCalibratedValue(unsigned char, unsigned char) at C:\Users\\Documents\Arduino\libraries\SparkFun_Spectral_Triad_AS7265X\src\SparkFun_AS7265X.cpp line 269
0x400d812d: AS7265X::getCalibratedA() at C:\Users\\Documents\Arduino\libraries\SparkFun_Spectral_Triad_AS7265X\src\SparkFun_AS7265X.cpp line 208
0x400d0b85: loop() at C:\Users\\AppData\Local\Temp\arduino_modified_sketch_674402/sketch_sep06a.ino line 56
0x400da7f5: loopTask(void*) at C:\Users\\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 23
0x4008e1fe: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143`

VojtechBartoska, 2022-04-07:
Hello @djornfevrier, sorry for replying so late. Is this still valid issue for you?

If yes, can you please test this on v2.0.3.-RC1?

djornfevrier, 2022-04-20:
Hi Yes, This is still an issue,

I have done, the issue is still present

Kind regards
DJorn Fevrier

VojtechBartoska, 2022-04-20:
Can you take a look on this issue: https://github.com/espressif/arduino-esp32/issues/207 can be relevant.

VojtechBartoska, 2022-05-18:
@djornfevrier Any news?

VojtechBartoska, 2022-06-14:
Closing as dupe of mentioned issue above. If needed, please reopen the issue.

closed 2022-06-14 by VojtechBartoska