espressif/arduino-esp32 #5692: Combining USBMSC with SD - results in crash when USB plug is connected to PC.

oswaldtoma, 2021-09-21:
### Hardware:
Board:							ESP32-S2 Saola WROVER
Core Installation version:			2.0.0
IDE name:						Arduino IDE 2.0-beta.11
Flash Frequency:					80 Mhz
PSRAM enabled:                                      no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I wanted to connect SD card with USBMSC. Separately these two are working in scenario: USBMSC with ramdisk and SD with listing directories and files, creating, deleting etc. When I wanted to "connect" these two features it all resulted in crashes when USB starts talking. I've provided MSC object callbacks for reading and writing. These callbacks use the "readRAW/writeRAW" methods. I don't know what I'm doing wrong. The code below is everything I've done so far.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include "Defines.h"
#include "USB.h"
#include "USBHIDKeyboard.h"
#include "SD.h"
#include "SPI.h"
#include "USBMSC.h"

#define SD_MISO  37
#define SD_MOSI  35
#define SD_SCK   36
#define SD_CS    34

#define BOOT_BUTTON 0

USBHIDKeyboard keyboard;
USBMSC msc;

static int32_t onWrite(uint32_t lba, uint32_t offset, uint8_t* buffer, uint32_t bufsize)
{
    log("write");
    SD.writeRAW((uint8_t*)buffer, lba);
    return bufsize;
}

static int32_t onRead(uint32_t lba, uint32_t offset, void* buffer, uint32_t bufsize)
{
    log("read");
    SD.readRAW((uint8_t*)buffer, lba);
    return bufsize;
}

static bool onStartStop(uint8_t power_condition, bool start, bool load_eject){
  return true;
}

void log(const char* str)
{
    Serial.println(str);
}

bool isBootButtonClicked()
{
    return digitalRead(BOOT_BUTTON) == LOW;
}

void setup()
{
    pinMode(BOOT_BUTTON, INPUT_PULLUP);

    Serial.begin(115200);

    SPIClass spi(FSPI);
    spi.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);
    if(SD.begin(SD_CS, spi, 40000000))
    {
        log("SD begin");
    }

    // keyboard.begin();
    msc.vendorID("ESP32");
    msc.productID("USB_MSC");
    msc.productRevision("1.0");
    msc.onRead(onRead);
    msc.onWrite(onWrite);
    msc.onStartStop(onStartStop);
    msc.mediaPresent(true);
    msc.begin(SD.cardSize() / 512, 512); //?
    USB.begin();

    char buff[50];
    sprintf(buff, "Total bytes: %d, Card Size: %d", SD.totalBytes() / 1024, SD.cardSize());
    log(buff);
}

void loop()
{
    // put your main code here, to run repeatedly:
    delay(200);

    if(isBootButtonClicked())
    {
        keyboard.write('a');
        log("klik!");
    }
}

```

### Debug Messages:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x498
load:0x4004c000,len:0xa88
load:0x40050000,len:0x25a8
entry 0x4004c19c
[     0][D][esp32-hal-tinyusb.c:580] tinyusb_enable_interface(): Interface HID enabled
[     0][D][USBHID.cpp:59] tinyusb_enable_hid_device(): Device[0] len: 6�
[   569][W][sd_diskio.cpp:186] sdCommand(): token error [59] 0x5
SD begin
[   573][D][esp32-hal-tinyusb.c:441] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 64, if_mask: 0x5
Total bytes: 962304, Card Size: 0
[ 51662][D][USBHID.cpp:174] tinyusb_load_enabled_hid_devices(): Loaded HID Desriptor with the following reports:
[ 51662][D][USBHID.cpp:182] tinyusb_load_enabled_hid_devices():   ID:   1, Type:   INPUT, Size:  8, Usage: KEYBOARD
[ 51671][D][USBHID.cpp:182] tinyusb_load_enabled_hid_devices():   ID:   1, Type:  OUTPUT, Size:  1, Usage: KEYBOARD
read
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400aee51  PS      : 0x00060b30  A0      : 0x80084e8e  A1      : 0x3ffc9de0  
A2      : 0x80081a99  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffce8c0  
A6      : 0x00000001  A7      : 0x3ffc1c40  A8      : 0x8002b8d8  A9      : 0x3ffc9dd0  
A10     : 0x00060b23  A11     : 0x00000000  A12     : 0x00060b20  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x80081a99  LBEG    : 0x00060b20  LEND    : 0x00000000  LCOUNT  : 0x40027205  

Backtrace:0x400aee4e:0x3ffc9de00x40084e8b:0x3ffc9e00 0x4008305a:0x3ffc9e30 0x400838f7:0x3ffc9e60 0x400828c0:0x3ffc9e80 0x4008197f:0x3ffc9ea0 0x40087a29:0x3ffc9ec0 0x40097b21:0x3ffc9ee0 0x400980ee:0x3ffc9f00 0x40099525:0x3ffc9f80 0x400862b3:0x3ffc9fe0 

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026324
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x498
load:0x4004c000,len:0xa88
load:0x40050000,len:0x25a8
entry 0x4004c19c
[     0][D][esp32-hal-tinyusb.c:580] tinyusb_enable_interface(): Interface HID enabled
[     0][D][USBHID.cpp:59] tinyusb_enable_hid_device(): Device[0] len: 67
[ 51877][W][sd_diskio.cpp:186] sdCommand(): token error [59] 0x5
SD begin
[ 51881][D][esp32-hal-tinyusb.c:441] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 64, if_mask: 0x5
Total bytes: 962304, Card Size: 0
```

chegewara, 2021-09-22:
I would add some code to check if sd card is really initialized properly, like list files or something.

oswaldtoma, 2021-09-22:
Just did that. It lists files properly. I used the code from SD example to list them.

chegewara, 2021-09-22:
Could you please check SD card example from my library, just to confirm `readRAW` is not broken?

Also decoding backtrace could help.

chegewara, 2021-09-22:
Try SPI clock like 20MHz or something. Its probably not this, because API should adjust that value, but who knows. Most likely 40MHz wont work with SPI anyway.

oswaldtoma, 2021-09-22:
Nope, didn't help.

chegewara, 2021-09-26:
Ok. I checked your example with `idf-master` branch, which i am currently working on, and it didnt crash, but failed to init sd card.
I also checked with sd card example from my lib and also failed to init sd card. Obviously there is something wrong with it.

Your example code:
```
00:37:22.161 -> ESP-ROM:esp32s2-rc4-20191025
00:37:22.161 -> Build:Oct 25 2019
00:37:22.161 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
00:37:22.161 -> SPIWP:0xee
00:37:22.161 -> mode:DIO, clock div:1
00:37:22.161 -> load:0x3ffe6100,len:0x570
00:37:22.161 -> load:0x4004c000,len:0xa50
00:37:22.161 -> load:0x40050000,len:0x28d4
00:37:22.161 -> entry 0x4004c18c
00:37:22.559 -> [   862][W][sd_diskio.cpp:104] sdWait(): Wait Failed
00:37:22.559 -> [   862][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
00:37:22.559 -> [   862][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
00:37:22.559 -> [   867][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
00:37:22.858 -> [  1176][W][sd_diskio.cpp:104] sdWait(): Wait Failed
00:37:22.858 -> [  1176][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
00:37:22.858 -> Total bytes: 0, Card Size: 0
00:37:36.230 -> [   868][W][sd_diskio.cpp:104] sdWait(): Wait Failed
00:37:36.230 -> [   868][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
00:37:36.230 -> [   868][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
00:37:36.230 -> [   873][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
00:37:36.529 -> [  1182][W][sd_diskio.cpp:104] sdWait(): Wait Failed
00:37:36.529 -> [  1182][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
00:37:36.529 -> Card Mount Failed
00:37:36.562 -> Failed to init SD
00:37:36.562 -> SD Card Type: UNKNOWN
00:37:36.562 -> SD Card Size: 0MB
00:37:36.562 -> Listing directory: /
00:37:36.562 -> [  1192][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:36.562 -> Failed to open directory
00:37:36.562 -> Creating Dir: /mydir
00:37:36.562 -> [  1205][E][vfs_api.cpp:154] mkdir(): File system is not mounted
00:37:36.562 -> mkdir failed
00:37:36.562 -> Listing directory: /
00:37:36.562 -> [  1215][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:36.596 -> Failed to open directory
00:37:36.596 -> Removing Dir: /mydir
00:37:36.596 -> [  1226][E][vfs_api.cpp:183] rmdir(): File system is not mounted
00:37:36.596 -> rmdir failed
00:37:36.596 -> Listing directory: /
00:37:36.596 -> [  1226][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:36.596 -> Failed to open directory
00:37:36.596 -> Writing file: /hello.txt
00:37:36.596 -> [  1242][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:36.596 -> Failed to open file for writing
00:37:36.596 -> Appending to file: /hello.txt
00:37:36.596 -> [  1252][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:36.628 -> Failed to open file for appending
00:37:36.628 -> Reading file: /hello.txt
00:37:36.628 -> [  1263][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:36.628 -> Failed to open file for reading
00:37:36.628 -> Deleting file: /foo.txt
00:37:36.628 -> [  1274][E][vfs_api.cpp:121] remove(): File system is not mounted
00:37:36.628 -> Delete failed
00:37:36.628 -> Renaming file /hello.txt to /foo.txt
00:37:36.628 -> [  1284][E][vfs_api.cpp:87] rename(): File system is not mounted
00:37:36.661 -> Rename failed
00:37:36.661 -> Reading file: /foo.txt
00:37:36.661 -> [  1295][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:36.661 -> Failed to open file for reading
00:37:36.661 -> [  1305][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:36.661 -> Failed to open file for reading
00:37:36.661 -> [  1305][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:36.694 -> Failed to open file for writing
00:37:36.694 -> Total space: 0MB
00:37:36.694 -> Used space: 0MB
[... 38 lines cut ...]
00:37:51.430 -> [  1236][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:51.430 -> Failed to open file for writing
00:37:51.430 -> Appending to file: /hello.txt
00:37:51.430 -> [  1246][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:51.464 -> Failed to open file for appending
00:37:51.464 -> Reading file: /hello.txt
00:37:51.464 -> [  1257][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:51.464 -> Failed to open file for reading
00:37:51.464 -> Deleting file: /foo.txt
00:37:51.464 -> [  1268][E][vfs_api.cpp:121] remove(): File system is not mounted
00:37:51.464 -> Delete failed
00:37:51.464 -> Renaming file /hello.txt to /foo.txt
00:37:51.464 -> [  1278][E][vfs_api.cpp:87] rename(): File system is not mounted
00:37:51.497 -> Rename failed
00:37:51.497 -> Reading file: /foo.txt
00:37:51.497 -> [  1289][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:51.497 -> Failed to open file for reading
00:37:51.497 -> [  1299][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:51.497 -> Failed to open file for reading
00:37:51.497 -> [  1299][E][vfs_api.cpp:22] open(): File system is not mounted
00:37:51.497 -> Failed to open file for writing
00:37:51.497 -> Total space: 0MB
00:37:51.497 -> Used space: 0MB
```
With espressif MSC + SD card i can see even more problems, which needs to be reported in separate issue.

chegewara, 2021-09-26:
Actually i am not going to open new issue, because SD library is broken and there is no point to do it, but here is output from linux logs when your example is running:
```
[297692.228125] usb 1-10: new full-speed USB device number 14 using xhci_hcd
[297692.556322] usb 1-10: New USB device found, idVendor=303a, idProduct=0002, bcdDevice= 7.23
[297692.556332] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[297692.556508] usb 1-10: Product: ESP32-S2
[297692.556512] usb 1-10: Manufacturer: Espressif
[297692.556515] usb 1-10: SerialNumber: 0
[297692.573035] cdc_acm 1-10:1.0: ttyACM0: USB ACM device
[297712.010693] usb 1-10: USB disconnect, device number 14
[297713.151758] usb 1-10: new full-speed USB device number 15 using xhci_hcd
[297713.489738] usb 1-10: New USB device found, idVendor=303a, idProduct=0002, bcdDevice= 1.00
[297713.489745] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[297713.489747] usb 1-10: Product: ESP32S2_DEV
[297713.489749] usb 1-10: Manufacturer: Espressif Systems
[297713.489751] usb 1-10: SerialNumber: 0
[297713.519529] usb-storage 1-10:1.0: USB Mass Storage device detected
[297713.519698] scsi host9: usb-storage 1-10:1.0
[297713.535917] input: Espressif Systems ESP32S2_DEV as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-10/1-10:1.1/0003:303A:0002.001F/input/input44
[297713.596080] hid-generic 0003:303A:0002.001F: input,hidraw3: USB HID v1.11 Keyboard [Espressif Systems ESP32S2_DEV] on usb-0000:02:00.0-10/input1
[297714.547108] scsi host9: scsi scan: INQUIRY result too short (5), using 36
[297714.547118] scsi 9:0:0:0: Direct-Access     ESP32    USB_MSC          1.0  PQ: 0 ANSI: 2
[297714.547432] sd 9:0:0:0: Attached scsi generic sg2 type 0
```
From now on logs are flooded with logs below until i wont turn off S2:
```
[297714.557863] sd 9:0:0:0: [sdc] Read Capacity(10) failed: Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE
[297714.557869] sd 9:0:0:0: [sdc] Sense Key : Not Ready [current] 
[297714.557872] sd 9:0:0:0: [sdc] Add. Sense: Logical unit not ready, cause not reportable
[297714.557876] sd 9:0:0:0: [sdc] 0 512-byte logical blocks: (0 B/0 B)
[297714.557878] sd 9:0:0:0: [sdc] 0-byte physical blocks
[297714.560702] sd 9:0:0:0: [sdc] Write Protect is off
[297714.560709] sd 9:0:0:0: [sdc] Mode Sense: 03 00 00 00
[297714.563687] sd 9:0:0:0: [sdc] No Caching mode page found
[297714.563693] sd 9:0:0:0: [sdc] Assuming drive cache: write through
[297714.633882] sd 9:0:0:0: [sdc] Read Capacity(10) failed: Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE
[297714.633891] sd 9:0:0:0: [sdc] Sense Key : Not Ready [current] 
[297714.633895] sd 9:0:0:0: [sdc] Add. Sense: Logical unit not ready, cause not reportable
[297714.636899] sd 9:0:0:0: [sdc] Read Capacity(10) failed: Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE
[297714.636907] sd 9:0:0:0: [sdc] Sense Key : Not Ready [current] 
[297714.636911] sd 9:0:0:0: [sdc] Add. Sense: Logical unit not ready, cause not reportable
[297714.645702] sd 9:0:0:0: [sdc] Attached SCSI removable disk
[297714.659885] sd 9:0:0:0: [sdc] Read Capacity(10) failed: Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE
[297714.659894] sd 9:0:0:0: [sdc] Sense Key : Not Ready [current] 
[297714.659898] sd 9:0:0:0: [sdc] Add. Sense: Logical unit not ready, cause not reportable
[297714.679871] sd 9:0:0:0: [sdc] Read Capacity(10) failed: Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE
[297714.679880] sd 9:0:0:0: [sdc] Sense Key : Not Ready [current] 
```
That app also "broke" my OS, because i cant unmount disks (both are from S2):
[image]

ShigemoriHakura, 2021-09-28:
Modify the code in setup to:

  static SPIClass* spi = NULL;
  spi = new SPIClass(FSPI);
  spi->begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);
  if (SD.begin(SD_CS, *spi, 40000000))
  {
    log("SD begin");
  }

And the code should work.

oswaldtoma, 2021-10-02:
> Modify the code in setup to:
> 
> static SPIClass* spi = NULL; spi = new SPIClass(FSPI); spi->begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS); if (SD.begin(SD_CS, *spi, 40000000)) { log("SD begin"); }
> 
> And the code should work.

THAT'S IT! Hell of a rookie error here. Thanks for help.

closed 2021-10-02 by oswaldtoma