espressif/arduino-esp32 #5694: Unable to compile using ESP32 1.0.5 / 1.0.6 and 2.0.0 - Error “& was unexpected at this time”

mycbtec, 2021-09-22:
Since updating ESP32 in "Boards Manager" from 1.0.4 to 1.0.X/2.00 I can no longer Verify/compile successfully in the IDE 1.8.16 for Windows. I can downgrade back to ESP32 v1.0.4 and sketches compile fine. Problem can be replicated by choosing "File/New" and saving the new sketch (without modification!) to my custom Sketchbook folder and try to Verify it. I then save the new sketch to C:\Temp folder and the Verify/compile is now fine. I save the same sketch to C:\Temp\a&b folder and now the compile fails again.

When the compile fails I just get file build.options.json in the build folder and the error message in the IDE is

& was unexpected at this time.
exit status 1
Error compiling for board WEMOS LOLIN32.
I looked at the build.options.json file and it pointed me to a slightly weird folder notation of my temporary sketch location of C:\Temp\a&b\sketch_mar06

sketchLocation": "C:\\Temp\\a\u0026b\\sketch_mar06c\\sketch_mar06c.ino
So it looks like the ampersand & in the sketch folder name is causing a compile problem in ESP32 v1.0.5 / v1.0.6 and v2.0.0 and was fine in previous versions of ESP32 boards. I don't really want to change my sketchbook folder (by removing the ampersand &) .

Can anyone help with the above reported issue?

mycbtec, 2022-03-30:
> Hello @mycbtec, do you still need help or can I close this as expired?

Unfortunately, no solution yet. If you have ampersand & in the sketch folder name it will cause a compile problem!

VojtechBartoska, 2022-03-30:
Can you then provide more info? Your sketch and so on?

mycbtec, 2022-03-30:
> Can you then provide more info? Your sketch and so on?

It's easy to simulate the bug:

1- Create a blank sketch 
2- Save it into a folder containing &, for example: C:\Bugtest&\sketchdemo
3- Try to compile
4- Remove & and try to compile the sketchdemo again... it should works. 

The problem here is if the PC user has "&" in it's name, the sketch will never compile... this bug did not happen in version 1.0.4

VojtechBartoska, 2022-03-31:
@mycbtec I see, ampersand is often forbidden in Windows as it can cause problems in some apps having collision, so it's safer to not use it
I would say we won't fix this and gladly ask you to not use this character in your filename or a path.

closed 2022-03-31 by VojtechBartoska