espressif/arduino-esp32 #5695: WiFi Issue, Device automatically disconnect

naakhtar, 2021-09-23:
### Hardware:
Board:							ESP32-Wroom-32D
IDE name:						Arduino IDE latest version

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))

What you are trying to do?
While working on Wi-Fi, device automatically disconnect.

Describe what is failing.
While working with this module we are facing range issues. 
For example when we are connected to the device the device some time automatically disconnects from the mobile/laptop and we get nothing. 
But after some time without any changes it again starts the same communication and showing the Wi-Fi Hotspot.

Show the EXACT error message(it doesn't work is not enough).
There is no exact error message. It is only the signal strength coming or some time it does not
and therefore it is not showing the Hotspot to connect.

YordanYanakiev, 2021-11-15:
It's not mentioned the SDK version, the device type/name, the code which is used for connecting..

VojtechBartoska, 2022-04-07:
Hello, I'm closing this issue as expired due it's old date and also no more info provided. If needed, please open a new issue.

closed 2022-04-07 by VojtechBartoska