espressif/arduino-esp32 #5697: [Question] Core 2.0 increased in size so much my code won't fit with (default) OTA Partition? (was using 1.0.6)

Frtrillo, 2021-09-23:
### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32 Dev Module|
|<B>Version/Date</B>|1.0.6|2.0.0|
|<B>IDE name</B>|Arduino IDE||VSCode|
|<B>Flash Frequency</B>|80Mhz|
|<B>PSRAM enabled</B>|no|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|

### Description:
I ran my code with core 1.0.6 (94% size use) with default partition (as im using OTA HTTPUpdates). 

So I went and updated core to 2.0 and with its increase in size I exceeded a bit the max. partition size.

I can upload my code of course if I change partition size to Huge APP (No OTA) but then I wouldn't be able to use HTTPUpdate( https://github.com/espressif/arduino-esp32/tree/master/libraries/HTTPUpdate)? Or is that only for Arduino IDE Updates and its different from HTTUpdates... 

And If I try to use the other partition schemes my LittleFS code won't work

### Debug Messages:
```
text section exceeds available space in board
The Sketch uses 1313501 bytes (100%) of the program storage space. The maximum is 1310720 bytes.
Global variables use 39536 bytes (12%) of dynamic memory, leaving 288144 bytes for local variables. The maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error compilando para la tarjeta ESP32 Dev Module.
```

alorbach, 2021-09-24:
Facing the same problem, build with 1.0.4 before no problem with 2.0.0 I get this: 

```
Sketch uses 2092697 bytes (106%) of program storage space. Maximum is 1966080 bytes.
Global variables use 62968 bytes (19%) of dynamic memory, leaving 264712 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
text section exceeds available space in board
```

Jason2866, 2021-09-24:
A solution could be to use OTA partitions with different sizes. The bigger for the running firmware and the smaller to store a gzip version of the uploaded new firmware. This firmware is extracted and flashed on the fly over the old firmware.

alorbach, 2021-09-24:
Interesting approach, any sample code or working samples?

SuGlider, 2021-10-21:
This is solved by https://github.com/espressif/esp32-arduino-lib-builder/pull/47
New sizes  are better than in V1.0.6. 
A draft core version for testing is in https://github.com/espressif/arduino-esp32/tree/mem-optimized
Next release 2.0.1RC1 will fix this binary size issue.

commit 67c99142 (2021-10-21): Allows spiram malloc with wifi dynamic buffers - better free heap (#5791)

Summary

Modifies WiFi lib to allow dynamic buffer allocation along with SPIRAM MALLOC enabled
This gives more heap space to the users

Related PR in Arduino Lib Builder: espressif/esp32-arduino-lib-builder#47

Impact

WiFi will work the same as it was in version 1.0.6, restoring free heap.

close #5630
close #5474
close #5699
close #5697