espressif/arduino-esp32 #5698: WiFi incorrect password give 4-way Handshake Timeout message, does not refer to password

RalphBacon, 2021-09-23:
Connecting to WiFi, typed in incorrect password. this is the resultant trace:

14:48:53.270 --> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:48:53.279 --> configsip: 0, SPIWP:0xee
14:48:53.279 --> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:48:53.288 --> mode:DIO, clock div:2
14:48:53.288 --> load:0x3fff0018,len:4
14:48:53.288 --> load:0x3fff001c,len:1044
14:48:53.288 --> load:0x40078000,len:10124
14:48:53.295 --> 
14:48:53.295 --> load:0x40080400,len:5828
14:48:53.295 --> entry 0x400806a8
14:48:53.726 --> Setting ESP32 as Station, no power savings, SSID Greypuss
14:48:53.726 --> I (35) wifi:nvs_log_init, erase log key successfully, reinit nvs log
14:48:53.726 --> I (39) wifi:wifi driver task: 3ffb5724, prio:23, stack:3584, core=0
14:48:53.757 --> I (312) wifi:wifi firmware version: dc30037
14:48:53.757 --> I (312) wifi:config NVS flash: enabled
14:48:53.757 --> I (312) wifi:config nano formating: disabled
14:48:53.757 --> I (313) wifi:Init data frame dynamic rx buffer num: 32
14:48:53.757 --> I (317) wifi:Init management frame dynamic rx buffer num: 32
14:48:53.765 --> I (323) wifi:Init management short buffer num: 32
14:48:53.765 --> I (327) wifi:Init dynamic tx buffer num: 32
14:48:53.774 --> I (331) wifi:Init static rx buffer size: 1600
14:48:53.774 --> I (335) wifi:Init static rx buffer num: 16
14:48:53.781 --> I (339) wifi:Init dynamic rx buffer num: 32
14:48:53.895 --> I (436) wifi:mode : sta (c4:4f:33:67:07:99)
14:48:53.895 --> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
14:48:53.895 --> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
14:48:53.895 --> I (441) wifi:Set ps type: 0
14:48:53.895 --> 
14:48:56.014 --> WiFi begin success.Connecting to WiFi .I (2559) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
14:48:56.751 --> I (3306) wifi:state: init -> auth (b0)
14:48:56.751 --> I (3314) wifi:state: auth -> assoc (0)
14:48:56.751 --> I (3320) wifi:state: assoc -> run (10)
14:49:00.006 --> I (6558) wifi:state: run -> init (fc0)
14:49:00.006 --> I (6559) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
14:49:00.006 --> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
14:49:00.006 --> [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
14:49:05.948 --> WiFi Status: WL_DISCONNECTED, exiting

Code as follows:

`   printf("Setting ESP32 as Station, no power savings, SSID %s\n", SSID);

    // WiFI debugging
    esp_log_level_set("wifi", ESP_LOG_VERBOSE);
    esp_wifi_internal_set_log_level(WIFI_LOG_VERBOSE);
    esp_wifi_internal_set_log_mod(WIFI_LOG_MODULE_ALL, WIFI_LOG_SUBMODULE_ALL, true);

    WiFi.mode(WIFI_STA);
    WiFi.persistent(false);
    WiFi.setAutoReconnect(true);
    WiFi.setSleep(false);
    if (WiFi.begin(SSID, PASSWORD))
    {
        printf("WiFi begin success.");
    }
    WiFi.setTxPower(WIFI_POWER_5dBm);
    delay(2000);

    printf("Connecting to WiFi .");
    uint8_t wifiStatus = WiFi.waitForConnectResult();

    // Successful connection?
    if ((wl_status_t)wifiStatus != WL_CONNECTED)
    {
        printf("WiFi Status: %s, exiting\n", wl_status_to_string((wl_status_t)wifiStatus));
        while (1)
            ;
    }`

Should this not tell me the password was wrong instead of 4-way Handshake Timeout?

RalphBacon, 2021-09-23:
How does this correlate to the fact that when I correct the password it connects? (The router is about 1m away, hence the reduced power request). Want me to re-test without that line in it?

atanisoft, 2021-09-23:
Distance to the AP is not entirely relevant, especially when you change the power configuration during the handshake process. If you want to change power levels do so before calling `WiFi.begin()`. Testing without that line may fix it as well.

RalphBacon, 2021-09-23:
I've commented out the line that controls the power. Snippet below:
`  
    WiFi.mode(WIFI_STA);
    WiFi.persistent(false);
    WiFi.setAutoReconnect(true);
    WiFi.setSleep(false);
    if (WiFi.begin(SSID, PASSWORD))
    {
        printf("WiFi begin success.\n");
    }
    //WiFi.setTxPower(WIFI_POWER_5dBm);
    delay(2000);`

Uploaded the code with **correct** password; all connected as expected.
Changed password to **incorrect** password; failed as before (see trace below):

`16:19:16.942 --> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:19:16.942 --> configsip: 0, SPIWP:0xee
16:19:16.942 --> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:19:16.942 --> mode:DIO, clock div:2
16:19:16.942 --> load:0x3fff0018,len:4
16:19:16.971 --> load:0x3fff001c,len:1044
16:19:16.971 --> load:0x40078000,len:10124
16:19:16.971 --> load:0x40080400,len:5828
16:19:16.971 --> entry 0x400806a8
16:19:17.386 --> Setting ESP32 as Station, no power savings, SSID Greypuss
16:19:17.386 --> I (36) wifi:nvs_log_init, erase log key successfully, reinit nvs log
16:19:17.386 --> I (40) wifi:wifi driver task: 3ffb5724, prio:23, stack:3584, core=0
16:19:17.408 --> I (314) wifi:wifi firmware version: dc30037
16:19:17.408 --> I (314) wifi:config NVS flash: enabled
16:19:17.408 --> I (315) wifi:config nano formating: disabled
16:19:17.430 --> I (315) wifi:Init data frame dynamic rx buffer num: 32
16:19:17.430 --> I (319) wifi:Init management frame dynamic rx buffer num: 32
16:19:17.430 --> I (325) wifi:Init management short buffer num: 32
16:19:17.430 --> I (329) wifi:Init dynamic tx buffer num: 32
16:19:17.430 --> I (333) wifi:Init static rx buffer size: 1600
16:19:17.455 --> I (338) wifi:Init static rx buffer num: 16
16:19:17.455 --> I (341) wifi:Init dynamic rx buffer num: 32
16:19:17.538 --> I (435) wifi:mode : sta (c4:4f:33:67:07:99)
16:19:17.538 --> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
16:19:17.538 --> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
16:19:17.538 --> I (440) wifi:Set ps type: 0
16:19:17.538 --> 
16:19:17.561 --> WiFi begin success.
16:19:19.642 --> Connecting to WiFi .I (2518) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
16:19:20.378 --> I (3265) wifi:state: init -> auth (b0)
16:19:20.378 --> I (3273) wifi:state: auth -> assoc (0)
16:19:20.378 --> I (3280) wifi:state: assoc -> run (10)
16:19:23.654 --> I (6554) wifi:state: run -> init (fc0)
16:19:23.654 --> I (6555) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
16:19:23.654 --> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
16:19:23.675 --> [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
16:19:29.573 --> WiFi Status: WL_DISCONNECTED, exiting
`
Perhaps someone needs to just try this out and make sure it's not me. BTW this happens on Arduino IDE 2.0 and PlatformIO (as they're both using the same framework, I guess).

atanisoft, 2021-09-23:
1. Which specific boards are you using? Not all third-party boards are created equally or are reliable.
2. How are you powering it?
3. If USB, have you tried a different USB cable/port?

RalphBacon, 2021-09-23:
I power via USB but have also tried powering using a 2A 5v supply and a different cable. No difference.

The board that worked initially but gave me the problem (after an upload) was the TTGO v1.7.1 with 16Mb PSRAM. This is the board I re-tried today but it fails most of the time. I can't determine a pattern when it connects and when it gets this 4-way Handshake error.

I then switched to a no-name board (looks very similar to Wemos/Lolin boards) without PSRAM; that's the one that works until I get the password wrong. As long as the password is correct it seems to work (over the last 24 hours, at least).

This issue has only occurred since the last update of the Espressif ESP32 Arduino framework (in PlatformIO) which I did about 1 week ago.

If I knew how to do it, I'd regress the ESP32 platform in PlatformIO to prove to myself that there was a change.

Background: I have a "live" project using a TTGO v1.7.1 with **4**Mb of PSRAM (in use) and that has never failed to connect to Wi-Fi over the 12 months or so that the project was developed. I dare not recompile / upload that code now in case I get this error.

An (unrelated?) error I have found (I can raise it separately) is that up until very recently, since this last framework update, the parameter for the ESP32 board always included `flashmode = qio` but that now just causes continual reboots on both the above boards. Changing it to "dio" resolves the issue but, once again, what has changed? The "live" project is still running in "qio" mode. If this is not relevant just let me know and I'll raise as a separate error if it is not already resolved.

RalphBacon, 2021-09-25:
After several more experiments, on various ESP32 platforms, I get the 4-way Handshake error whenever an incorrect password is submitted.

 I wonder if the 4-way Handshake Timeout is due to my Router/WiFi simply not responding to incorrect password presentation? Security through obfuscation; it won't say what the error was, not even that there was an error, but just quietly ignores the request - leading to a real 4-way Handshake Timeout?

My Router/WiFi AP is one supplied by PlusNet, a subsidiary of British Telecom in the UK. It identifies itself only as "Plusnet Hub One | Software version 4.7.5.1.83.8.289.1.1 | Last updated 09/06/21"

TD-er, 2021-09-27:
It could very well be that the access point deliberately shows this response to avoid a DDoS-like attack.
Otherwise it is very easy to let the access point jam the ether with responses and no longer being able to handle legitimate traffic.

Another reason can be that responding with an error may cause issues when 2 access points in an area use the same SSID but different passwords.
No idea if this behavior is according to the standards, but at least it is understandable why it is done like this.

At least, when a node can connect reliable with the correct password, then I guess it is not really a problem.
What may make it harder for you to debug, is when a node cannot connect reliably even with the correct password.
I've seen (mainly on cheap ESP8266 boards) that some boards do have a really badly tuned antenna.
This results in packet loss, slow responding web page, frequent disconnects, etc.
I even have a board here which only allows me to connect to an access point when I press my thumb on the PCB antenna.

Another thing I noticed is that the latest code does perform really bad on my ESP32 boards. Packet loss of over 50%.
So I went back to `ESP32 SDK v3.3.5-1-g85c43024c` or whatever is installed when using this in PlatformIO.ini:
```
platform                  = espressif32 @ 3.3.2
platform_packages         = framework-arduinoespressif32
```

RalphBacon, 2021-09-27:
I wondered if the latest framework had anything to do with it so I regressed, upgrade by laborious upgrade, right back to 2.2.x - none of it made any difference to this issue. It always errored with a bad password in the same way.

So I think my router does not give an error (in a timely manner) so the 4-way handshake is true (just very confusing). My router is relatively new (within the last 3 months) which might explain why I have not noticed this behaviour before. Normally I type the password in correctly!

I think this can be closed - but I'll open another issue about the Flash mode qio suddenly not working and now have to use dio (on boards that were previously working fine, for months).

closed 2021-09-27 by RalphBacon

TD-er, 2021-09-27:
> I think this can be closed - but I'll open another issue about the Flash mode qio suddenly not working and now have to use dio (on boards that were previously working fine, for months).

On the same board?
I do know that `ESP32-D0WDQ6` and `ESP32-D0WDQ5` seem to differ regarding flash mode.

RalphBacon, 2021-09-27:
This is on a board that has been in use for at least 12 months. It's a TTGO v1.71 with 4Mb PSRAM.

A recent (and infrequent) upload caused it to reboot continuously. Internet pointed me towards Flash Mode so I changed it to DIO - works again. Since then _ALL_ my ESP32 boards have to use DIO where they were all previously happy with QIO (which I thought was the Espressif recommended/default value).

I'm doing some logging so I can report this with all relevant information.

PaulSchulz, 2023-05-25:
I can confirm that if I am using an incorrect wifi password on my ESP32S3 based board (Heltec Wifi Lora V3)  I get this error.

RalphBacon, 2023-05-25:
> I can confirm that if I am using an incorrect wifi password on my ESP32S3 based board (Heltec Wifi Lora V3) I get this error.

I'm assuming you have a different router to me (an assumption, but seems likely). Therefore, it is not a _specific_ router that just doesn't respond with "Wrong Password", and the 4-way handshake seems to be symptomatic of a silent, but failed, attempt to authenticate.

If the above hypothesis is true, and given this post is now nearly 2 years old with no "progress" as such being made, I guess we can just make a note that 4-way handshake failure message is likely due to a wrong password rather than some esoteric failure within the Espressif framework, the Wi-Fi network itself or the ESP32 chip.