espressif/arduino-esp32 #5701: Upgrading to 2.0 breaks SD function

rayb12, 2021-09-24:
I just upgraded from 1.06 to 2.0 of the ESP32 Arduino core.  Using 1.06 all Sd functions were working properly.  After upgrading to 2.0 SD fails to begin.  This is a custom board and SD is connected as per the standard SD connections (as in the picture in the Sd library folder).

I copied the Sd example from the Sd library here and it does the same thing, fails to mount SD card.
As a test, I switched back to ver# 1.06 and all SD functions work properly again.  Am I missing something?

I searched this forum but could not find anything specifically related to my problem.  I'm using the SD library, not the SDMMC library.

System: Arduino IDE
Board Selected in Arduino IDE: ESP32 Dev Module

Code:
    if(!SD.begin(){
        Serial.println("Card Mount Failed");
        return;
    }
Code above always returns failed.
I also tried the following code:
    if(!SD.begin(5){
        Serial.println("Card Mount Failed");
        return;
    }
I also changed SD cards just in case, but still same problem.

Hope this is enough info.  Any help is greatly appreciated!!!

chegewara, 2021-09-24:
Sorry, but this may be not enough info. There is many people using SD library in 2.0.0, including myself, and we have no problem with it, but I have to admit that recently i am using mostly esp32S2 and SD card with custom `SPIClass` object.
Maybe if you enable verbose logging level, then you will have more info for us?

rayb12, 2021-09-24:
Thank you chegewara!
Here is the verbose output (I apologize if this is not what you were asking for):

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
[0	⸮⸮W+QW⸮Y.\&⸮KZX⸮K,\]⸮,'⸮LLW⸮⸮Y](\]⸮⸮equencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 8000000[  2310][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  2310][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  2310][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[  2315][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  2624][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  2624][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount Failed

Thank you for any help in advance!

chegewara, 2021-09-24:
It looks like hardware problem, but who knows.
Could you try to use `SPIClass` to initialize SD card?

rayb12, 2021-09-26:
@chegewara,
I added code to use SPIClass as per your suggestion.

I downgraded back to ver# 1.06 and turned on verbose logging to see why my SD is failing on ver# 2.0.  I discovered that it does indeed fail on sdWait() and sdSelectCard().  However, it then mounts correctly and the SD seems to function correctly (read/write, etc.).

I forgot to mention above, I'm using an ESP32 WROOM module in a custom board and using a micro SD instead of a mini SD.  I checked all the wiring and the SD is indeed connected properly.

Here is my code partially taken from the SD example:

#define SDM_CLK 18
#define SDM_MISO 19
#define SDM_MOSI 23
#define SDM_CS 5

void setup(){
    Serial.begin(115200);
    SPIClass sdSPI(HSPI);
    sdSPI.begin(SDM_CLK,SDM_MISO,SDM_MOSI,SDM_CS); 
    
      if(SD.begin(SDM_CS, sdSPI)){ //, *hspi)){
          Serial.println("Card Mount Success");
      }
      else{
        Serial.println("SD failed");
      }
      uint8_t cardType = SD.cardType();
      if(cardType == CARD_NONE){
          Serial.println("No SD card attached");
          return;
      }
      Serial.print("SD Card Type: ");
      if(cardType == CARD_MMC){
          Serial.println("MMC");
      } else if(cardType == CARD_SD){
          Serial.println("SDSC");
      } else if(cardType == CARD_SDHC){
          Serial.println("SDHC");
      } else {
          Serial.println("UNKNOWN");
      }
      uint64_t cardSize = SD.cardSize() / (1024 * 1024);
      Serial.printf("SD Card Size: %lluMB\n", cardSize);
      Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
      Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));

    }

And here is the serial logs from the above code using ver# 1.06:

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[W][sd_diskio.cpp:101] sdWait(): Wait Failed
[E][sd_diskio.cpp:123] sdSelectCard(): Select Failed
[W][sd_diskio.cpp:175] sdCommand(): crc error
Card Mount Success
SD Card Type: SDHC
SD Card Size: 14910MB
Total space: 14902MB
Used space: 0MB

Using the same code but upgraded to ver# 2.0 here is the serial logs:

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
[0	⸮⸮W+QW⸮Y.\&⸮KZX⸮K,\]⸮,'⸮LLW⸮⸮Y](\]⸮⸮equencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 800000⸮Turned on 3V3 Aux
[  2309][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  2309][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  2309][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[  2314][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  2623][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  2623][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
SD failed

Not sure what the garbage is right after "entry" above, but, as you can see, everything fails using ver# 2.0.  I also tried not using the SPI class to initialize the SD card and I get the same results.

Any help is greatly appreciated!

chegewara, 2021-09-26:
I can confirm. Stock SD card example is giving this result.
I tested with `idf-master` branch and i have the same result.
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x570
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28d4
entry 0x4004c18c
[   854][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[   854][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[   854][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[   859][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  1168][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  1168][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
No SD card attached
```

rayb12, 2021-09-28:
Bump

@me-no-dev 

Any ideas on this issue?

me-no-dev, 2021-09-29:
@P-R-O-C-H-Y please have a look

P-R-O-C-H-Y, 2021-09-30:
@me-no-dev 
For me the SD card is working fine on version 2.0.
Tested with `master` branch both SD examples, no errors or warnings.
 
System: Arduino IDE / VS code
Board: ESP32 Wrover kit v4.1

rayb12, 2021-10-04:
@P-R-O-C-H-Y 
Did you get a chance to try on more ESP boards yet?  We are really stuck over here.  We have tried several different micro SD cards.  Formatted with SD-Formatter as per other suggestions.  I'm wondering if it could be a timing issue?  In reading other posts, some have mentioned issues with needing to slow the SPI bus down to around 10MHZ.  Can you tell me how to do that with ver# 2.0?  I'm just grasping at anything at the moment.

Any help is greatly appreciated!

P-R-O-C-H-Y, 2021-10-04:
@rayb12 
Now builded on esp32s2 with same result. Everything is working. IDK where is the problem.
SPI speed is 80MHz and working perfectly.

**CODE:**
```

#define SD_CS 34
#define SD_SCK 36
#define SD_MISO 37
#define SD_MOSI 35

SPIClass sd_spi;

void setup() 
{
  Serial.begin(115200);
  while (!Serial) {}

  sd_spi.begin(SD_SCK,SD_MISO,SD_MOSI,SD_CS); // The bultin SD card SPI for WROVER
 
  if (!SD.begin(SD_CS,sd_spi)) 
  {
        Serial.println("initialization failed!");
        while(1);
  }
  else
  {
        Serial.println("initialization done.");
  }

  listDir(SD,"/",0);
}
```

**OUTPUT:**
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x17e4
load:0x4004c000,len:0xa48
load:0x40050000,len:0x2e58
entry 0x4004c204
I (21) boot: ESP-IDF v4.4-dev-2875-g5f38b766a8-dirty 2nd stage bootloader
I (21) boot: compile time 14:46:50
I (22) boot: chip revision: 0
I (26) boot.esp32s2: SPI Speed      : 80MHz
I (31) boot.esp32s2: SPI Mode       : DIO
I (35) boot.esp32s2: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0ab34h ( 43828) map
I (100) esp_image: segment 1: paddr=0001ab5c vaddr=3ffbe0a0 size=01f04h (  7940) load
I (102) esp_image: segment 2: paddr=0001ca68 vaddr=40022000 size=035b0h ( 13744) load
I (112) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=280fch (164092) map
I (150) esp_image: segment 4: paddr=00048124 vaddr=400255b0 size=08af0h ( 35568) load
I (159) esp_image: segment 5: paddr=00050c1c vaddr=50000000 size=00010h (    16) load
I (165) boot: Loaded app from partition at offset 0x10000
I (165) boot: Disabling RNG early entropy source...
I (180) ca〞f�~���F�Սѥ���cache  : size 8KB, 4Ways, cache line size 32Byte
I (181) cpu_start: Pro cpu up.
I (192) cpu_start: Pro cpu start user code
I (192) cpu_start: cpu freq: 160000000
I (192) cpu_start: Application information:
I (197) cpu_start: Project name:     sd_test
I (202) cpu_start: App version:      1
I (206) cpu_start: Compile time:     Oct  4 2021 15:48:02
I (212) cpu_start: ELF file SHA256:  83f6db9b1d7f3379...
I (218) cpu_start: ESP-IDF:          v4.4-dev-2875-g5f38b766a8-dirty
I (225) heap_init: Initializing. RAM available for dynamic allocation:
I (233) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (239) heap_init: At 3FFC09F8 len 0003B608 (237 KiB): DRAM
I (245) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (252) spi_flash: detected chip: generic
I (256) spi_flash: flash io: dio
I initialization done.
Listing directory: /
  FILE: TEST.TXT  SIZE: 1048576
  FILE: FOO.TXT  SIZE: 0
  FILE: FOOO.TXT  SIZE: 0
```

P-R-O-C-H-Y, 2021-10-04:
@rayb12 
**EDIT:** got same issue like you get, when building in Arduino IDE, when in VS code its working.
Gonna find the problem.

rayb12, 2021-10-04:
@P-R-O-C-H-Y 
Thanks for looking into it.  Look forward to you finding the problem.  As side note, can you tell me where to change the SPI speed?
Thanks!

P-R-O-C-H-Y, 2021-10-05:
@rayb12 
Can you print your idf version?

Insert this code:
```
Serial.println("ESP-IDF version is: " + String(esp_get_idf_version()));
```

P-R-O-C-H-Y, 2021-10-05:
Updated Arduino IDE with latest version of arduino-esp32 ```master``` branch and everything working correctly on ESP32 and ESP32S2. Can you try that?

rayb12, 2021-10-05:
@P-R-O-C-H-Y 

That worked! Here is the output including idf version:

ESP-IDF version is: v4.4-dev-2313-gc69f0ec32
Card Mount Success
SD Card Type: SDHC
SD Card Size: 14910MB
Total space: 14902MB
Used space: 1MB

Thank you soooooo much!!!!

P-R-O-C-H-Y, 2021-10-06:
@rayb12  No problem :) So I can close this issue right?

closed 2021-10-06 by rayb12

paoloinverse, 2022-01-25:
Hello,

this issue is still valid and present. I'm using latest stable 2.0.2 on Arduino, upgraded from 1.0.6 (with which, SD card reader works flawlessly), and I get this with SD.begin(): 

[  1088][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  1088][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  1088][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[  1093][E][sd_diskio.cpp:795] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  1401][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  1401][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed

idf version as reported, is: 

v4.4-beta1-189-ga79dc75f0a

The wiring is correct. The SD card test example sketch fails too, so it's not my code the source of the problem. 
It tried deleting the ESP32 board and reinstalling again: same result. 
Tried upgrading the Arduino IDE to arduino-1.8.19 and the SD still fails. 

If I rollback to ESP32 board ver. 1.0.6 the SD card works perfectly again.

Could it be the chip select pin of the VSPI definition changed? Or perhaps the internal pullup in VMISO is not enabled? Because otherwise I'm at a loss. 

@P-R-O-C-H-Y  what do you mean with "...update Arduino IDE with latest version of arduino-esp32 master branch" ? 

Arduino IDE does not allow me to install a board definition outside the File -> Preferences -> paste the  .json URL procedure ...or does it? 

Do you mean I have to download the arduino-esp32 master branch in a zip file and unzip it somewhere in the Arduino IDE data folder? 
Thanks in advance.

cyberman54, 2022-01-30:
This code works on my ESP32 (TTGO v21 old version) without errors;

```
SPIClass sd_spi;
...
digitalWrite(SDCARD_CS, HIGH);
sd_spi.begin(SDCARD_SCLK, SDCARD_MISO, SDCARD_MOSI, SDCARD_CS);
digitalWrite(SDCARD_CS, LOW);
useSDCard = SD.begin(SDCARD_CS, sd_spi);
```