espressif/arduino-esp32 #5702: ESP32 Wire.requestFrom() stop parameter not working

caternuson, 2021-09-24:
### Hardware:
Board:							Adafruit Feather ESP32
Core Installation version:			1.0.6, 2.0.0, 2.0.0 w/ PR #5683
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    n/a
Upload Speed:					921600
Computer OS:                                        Ubuntu

### Description:
Board does not generated expected I2C traffic when attempting to suppress the I2C stop on a read. This is a weird use case, but is needed for implementing chuckified reads when total length exceeds board's read buffer.

### Sketch:
```cpp
#include <Wire.h>

#define WIRE Wire
#define I2C_ADDRESS (0x69)
#define READ_SIZE (32)

void setup() {
  Serial.begin(9600);
  while(!Serial);

  uint8_t buffer[READ_SIZE];
  
  WIRE.begin();
  WIRE.requestFrom(I2C_ADDRESS, 32, false);
  uint32_t count = 0;
  while (WIRE.available() and count < READ_SIZE) {
    buffer[count++] = WIRE.read();
  }
}

void loop() {
}
```

### Debug Messages:
None. Code compiles and runs without errors. Issue is seen in actual I2C traffic. See traces below.

**BSP 1.0.6 and 2.0.0**
**stop = true** (expected results)
[image]

**stop = false** (not expected results)
[NO TRAFFIC]

**BSP 2.0.0 with PR #5683**
**stop = true** (expected results)
[image]

**stop = false** (not expected results - should be no stop at end)
[image]

See here for expected outputs on a SAMD21 based board (Adafruit Qt Py):
https://github.com/adafruit/Adafruit_BusIO/issues/64#issuecomment-926894559

me-no-dev, 2021-09-27:
I am not sure if we will be able to support this. Do you have hardware that requires this feature? Or is this something custom?

caternuson, 2021-09-27:
Here's one example sensor that has a long read back to get all the sensor data:
https://github.com/adafruit/Adafruit_AMG88xx

The main goal is to abstract all the I2C details into this library:
https://github.com/adafruit/Adafruit_BusIO
including breaking up long read/writes into "chunks" sized per platform.

The necessity for actually needing this (the suppressed stop) can vary from sensor to sensor though. For example, the AMG8833 seems to be OK with our without it. But general concern would be other sensors may see it as repeated start and reset their internal data buffer and essentially terminate the long data transfer.

caternuson, 2021-09-27:
@me-no-dev FWIW - things seem to end up working with the #5683 updates, despite it still producing the stop (as seen in scope trace above). Here's an example with AMG8833:
https://github.com/adafruit/Adafruit_BusIO/pull/65

The biggest issue with current BSP is it just gives up and doesn't even output anything on the I2C bus.

VojtechBartoska, 2022-04-07:
Hello, can you please retest this with v2.0.3-rc1? thanks!

me-no-dev, 2022-04-27:
@caternuson final resolution seems to be that we can not support this on ESP32. The peripheral will not send anything until STOP is queued. You PR to Adafruit was the correct action to take.

closed 2022-04-27 by me-no-dev

caternuson, 2022-04-27:
@me-no-dev OK. Thanks for responding back here with info on final resolution.