espressif/arduino-esp32 #5704: Update Timeout Flawed

rdammkoehler, 2021-09-26:
https://github.com/espressif/arduino-esp32/blob/317be68cefe4678e49f712e263da8064b75c7d68/libraries/Update/src/Updater.cpp#L357

Yes, I agree, 100*300 = 30000 = 30s, but you forgot the stream timeout below which is also contributing. I'm seeing about 8s on my device right now (oddly unaffected by setTimeout) and that is getting me 8000*30000 or 66 2/3h before this is going to timeout.

Jaredkmacho, 2021-11-01:
>I also discovered that even if the socket comes back it won't recover...so all these retries (in my case) don't help with the problem
Im trying to make OTA work in my project and have noticed the same thing. When writeStream is timing out I wait a consistent 6 minutes untill the abort function is called. This makes it really time consuming to debug the problem Im having (I can do OTA with small binaries ~300kB but little over 1MB fails most of the time).

VojtechBartoska, 2022-04-07:
Hello, are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? 

Take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-21:
Hello,

as there was no answer in more than 14 days, I'm closing the issue as expired to keep our backlog manageable.

If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-21 by VojtechBartoska

rdammkoehler, 2022-04-22:
@VojtechBartoska I have found a reasonable work around for now but will try to test using 2.0.3-RC1 however we are very busy with a number of other activities and might not get back to OTA for another month.

cyberman54, 2023-07-29:
@VojtechBartoska Could we please reopen this issue? It's still existing in 2.0.9.
Another, related, problem is that Update on a Wifi socket stream often fails with file sizes > 1MB, as stated above.

cyberman54, 2023-07-29:
> > I also discovered that even if the socket comes back it won't recover...so all these retries (in my case) don't help with the problem
> 
> Im trying to make OTA work in my project and have noticed the same thing. When UpdateClass::writeStream() is timing out I wait a consistent 6 minutes untill the abort function is called. This makes it really time consuming to debug the problem Im having (I can do OTA with small binaries ~300kB but little over 1MB fails most of the time).

Did you find a solution to update bigger files? I ran in the same problem.

Jaredkmacho, 2023-07-30:
@cyberman54 In the end I switched libraries to HTTPUpdater instead