espressif/arduino-esp32 #5705: BLE_client not looping RSSI in serial monitor.

sssuun, 2021-09-26:
### Hardware:
|||
|:---|---|
|<B>Board</B>|DOIT ESP32 DevKit V1|
|<B>IDE name</B>|Arduino IDE|
|<B>Flash Frequency</B>|80Mhz|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|

### Description:
I try to use ESP32 as client and multiple server. now they can connect but when Client connect to Server A and B. it not keep showing RSSI value in serial monitor. I don't know what wrong with my code.

### Sketch:  
```cpp

#include "BLEDevice.h"
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>
#include <iostream>
#include <string>

static BLEUUID serviceUUID_A("7cddf5af-453f-40fa-808e-37ae6ad8facd");
static BLEUUID charUUID_A("3383a686-a53c-42e4-a88c-e0e1de4c4bda");

static boolean doConnect_A = false;
static boolean connected_A = false;
static boolean doScan_A = false;
static BLERemoteCharacteristic* pRemoteCharacteristic_A;
static BLEAdvertisedDevice* myDevice_A;
static BLEClient*  pClient_A;

static BLEUUID serviceUUID_B("69e2c127-7db7-469c-8852-b0a542e15623");
static BLEUUID charUUID_B("2916187f-e651-4c42-9144-d91262892b22");

static boolean doConnect_B = false;
static boolean connected_B = false;
static boolean doScan_B = false;
static BLERemoteCharacteristic* pRemoteCharacteristic_B;
static BLEAdvertisedDevice* myDevice_B;
static BLEClient*  pClient_B;

static void notifyCallback_A(BLERemoteCharacteristic* pBLERemoteCharacteristic,uint8_t* pData,size_t length,bool isNotify){}
static void notifyCallback_B(BLERemoteCharacteristic* pBLERemoteCharacteristic,uint8_t* pData,size_t length,bool isNotify){}

class MyClientCallback_A : public BLEClientCallbacks {
      void onConnect(BLEClient* pclient_A) {
           }
      void onDisconnect(BLEClient* pclient_A) {
           connected_A = false;
           Serial.println("onDisconnect");
          }
     };

class MyClientCallback_B : public BLEClientCallbacks {
    void onConnect(BLEClient* pclient_B) {
        }
    void onDisconnect(BLEClient* pclient_B) {
         connected_B = false;
         Serial.println("onDisconnect");
        }
    };

bool connectToServer_A() {
      Serial.print("Forming a connection to Server A with MAC Address: ");
      Serial.println(myDevice_A->getAddress().toString().c_str());
      
      pClient_A  = BLEDevice::createClient();
      pClient_A->setClientCallbacks(new MyClientCallback_A());
      
      // Connect to the remove BLE Server.
      pClient_A->connect(myDevice_A);
      
      // Obtain a reference to the service we are after in the remote BLE server.
      BLERemoteService* pRemoteService = pClient_A->getService(serviceUUID_A);
      if (pRemoteService == nullptr) {
          Serial.print("Failed to find our service UUID: ");
          Serial.println(serviceUUID_A.toString().c_str());
          pClient_A->disconnect();
          return false;
      }
      Serial.println("Server A Service found");
      
      //Obtain a reference to the characteristic in the service of the remote BLE server.
      pRemoteCharacteristic_A = pRemoteService->getCharacteristic(charUUID_A);
      if (pRemoteCharacteristic_A == nullptr) {
          Serial.print("Failed to find our characteristic UUID: ");
          Serial.println(charUUID_A.toString().c_str());
          pClient_A->disconnect();
          return false;
      }
      Serial.println("Server A Characterstics found");
      
      //Register the client for notifications
      if (pRemoteCharacteristic_A->canNotify())
          pRemoteCharacteristic_A->registerForNotify(notifyCallback_A);
      
      connected_A = true;
}

class MyAdvertisedDeviceCallbacks_A: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.print("Potential Servers: ");
      Serial.println(advertisedDevice.toString().c_str());
      Serial.print(" RSSI A : ");
      Serial.println(advertisedDevice.getRSSI());

  //We have found a device, let us now see if it contains the service we are looking for.
  if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID_A)) {
      BLEDevice::getScan()->stop();
      myDevice_A = new BLEAdvertisedDevice(advertisedDevice);
      doConnect_A = true;
      doScan_A = true;
    }
  }
};

bool connectToServer_B() {
      Serial.print("Forming a connection to Server B with MAC Address: ");
      Serial.println(myDevice_B->getAddress().toString().c_str());
      
      pClient_B  = BLEDevice::createClient();
      pClient_B->setClientCallbacks(new MyClientCallback_B());
      
      //Connect to the remove BLE Server.
      pClient_B->connect(myDevice_B); // if you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)
      
      //Obtain a reference to the service we are after in the remote BLE server.
      BLERemoteService* pRemoteService = pClient_B->getService(serviceUUID_B);
      if (pRemoteService == nullptr) {
          Serial.print("Failed to find our service UUID: ");
          Serial.println(serviceUUID_B.toString().c_str());
          pClient_B->disconnect();
          return false;
      }
      Serial.println("Server B Service found");
      
      //Obtain a reference to the characteristic in the service of the remote BLE server.
      pRemoteCharacteristic_B = pRemoteService->getCharacteristic(charUUID_B);
      if (pRemoteCharacteristic_B == nullptr) {
          Serial.print("Failed to find our characteristic UUID: ");
          Serial.println(charUUID_B.toString().c_str());
          pClient_B->disconnect();
          return false;
      }
      Serial.println("Server B Characterstics found");
      
      //Register the client for notifications
      if (pRemoteCharacteristic_B->canNotify())
      pRemoteCharacteristic_B->registerForNotify(notifyCallback_B);
      
      connected_B = true;
}

class MyAdvertisedDeviceCallbacks_B: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.print("Potential Servers: ");
      Serial.println(advertisedDevice.toString().c_str());
      Serial.print(" RSSI B : ");
      Serial.println(advertisedDevice.getRSSI());

  //We have found a device, let us now see if it contains the service we are looking for.
  if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID_B)) {
      BLEDevice::getScan()->stop();
      myDevice_B = new BLEAdvertisedDevice(advertisedDevice);
      doConnect_B = true;
      doScan_B = true;
  }
 }
};

void setup() {
    Serial.begin(115200);
    Serial.println("Starting Arduino BLE Client application...");
    BLEDevice::init("esp32_Client");
    
    BLEScan* pBLEScan_A = BLEDevice::getScan();
    pBLEScan_A->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks_A());
    pBLEScan_A->setInterval(1349);
    pBLEScan_A->setWindow(449);
    pBLEScan_A->setActiveScan(true);
    pBLEScan_A->start(5, false);
    
    BLEScan* pBLEScan_B = BLEDevice::getScan();
    pBLEScan_B->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks_B());
    pBLEScan_B->setInterval(1349);
    pBLEScan_B->setWindow(449);
    pBLEScan_B->setActiveScan(true);
    pBLEScan_B->start(5, false);
}

void loop() {
    if (doConnect_A == true) {
        if (connectToServer_A()) {
           Serial.println("We are now connected to Server A");
        } 
        else {
           Serial.println("We have failed to connect to server A");
        }
      doConnect_A = false;
    }

    if (connected_A) {
        int rssi_A = pClient_A->getRssi();
        Serial.println("RSSI A :"+ String(rssi_A));
      } 
    else if(doScan_A) {
        BLEDevice::getScan()->start(0);
      }

    if (doConnect_B == true) {
        if (connectToServer_B()) {
            Serial.println("We are now connected to Server B");
        } 
        else {
           Serial.println("We have failed to connect to Server B");
        }
      doConnect_B = false;
    }

    if (connected_B) {
     
        int rssi_B = pClient_B->getRssi();
        Serial.println("RSSI B :"+ String(rssi_B));
      } 
    else if(doScan_B) {
        BLEDevice::getScan()->start(0);
      }

    delay(1000);
}
```

sssuun, 2021-09-26:
> Starting Arduino BLE Client application...
Potential Servers: Name: ESP32 Server A, Address: 24:6f:28:9e:a5:c2, serviceUUID: 7cddf5af-453f-40fa-808e-37ae6ad8facd, txPower: 3
 RSSI A : -40
Potential Servers: Name: ESP32 Server A, Address: 24:6f:28:9e:a5:c2, serviceUUID: 7cddf5af-453f-40fa-808e-37ae6ad8facd, txPower: 3
 RSSI B : -41
Potential Servers: Name: ESP32 Server B, Address: 24:6f:28:9d:74:56, serviceUUID: 69e2c127-7db7-469c-8852-b0a542e15623, txPower: 3
 RSSI B : -47
Forming a connection to Server A with MAC Address: 24:6f:28:9e:a5:c2
Server A Service found
Server A Characterstics found
We are now connected to Server A
RSSI A :-46
Forming a connection to Server B with MAC Address: 24:6f:28:9d:74:56
Server B Service found
Server B Characterstics found
We are now connected to Server B
RSSI B :-50

> 
serial monitor showing this but not continuous showing RSSI value.

sssuun, 2021-09-26:
### Debug Messages:
```
Starting Arduino BLE Client application...
[D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -49
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x07 (), length: 16, data: cdfad86aae378e80fa403f45aff5dd7c
[D][BLEAdvertisedDevice.cpp:500] setServiceUUID(): - addServiceUUID(): serviceUUID: 7cddf5af-453f-40fa-808e-37ae6ad8facd
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x09 (), length: 14, data: 4553503332205365727665722041
[D][BLEAdvertisedDevice.cpp:461] setName(): - setName(): name: ESP32 Server A
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x0a (), length: 1, data: 03
[D][BLEAdvertisedDevice.cpp:530] setTXPower(): - txPower: 3
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
Potential Servers: Name: ESP32 Server A, Address: 24:6f:28:9e:a5:c2, serviceUUID: 7cddf5af-453f-40fa-808e-37ae6ad8facd, txPower: 3
[D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -48
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x07 (), length: 16, data: 2356e142a5b052889c46b77d27c1e269
[D][BLEAdvertisedDevice.cpp:500] setServiceUUID(): - addServiceUUID(): serviceUUID: 69e2c127-7db7-469c-8852-b0a542e15623
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x09 (), length: 14, data: 4553503332205365727665722042
[D][BLEAdvertisedDevice.cpp:461] setName(): - setName(): name: ESP32 Server B
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x0a (), length: 1, data: 03
[D][BLEAdvertisedDevice.cpp:530] setTXPower(): - txPower: 3
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
Potential Servers: Name: ESP32 Server B, Address: 24:6f:28:9d:74:56, serviceUUID: 69e2c127-7db7-469c-8852-b0a542e15623, txPower: 3
Forming a connection to Server A with MAC Address: 24:6f:28:9e:a5:c2
[I][BLEDevice.cpp:614] addPeerDevice(): add conn_id: 0, GATT role: client
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
Server A Service found
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 42, UUID: 3383a686-a53c-42e4-a88c-e0e1de4c4bda
[E][BLERemoteCharacteristic.cpp:287] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
Server A Characterstics found
We are now connected to Server A
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
RSSI A :-50
Forming a connection to Server B with MAC Address: 24:6f:28:9d:74:56
[I][BLEDevice.cpp:614] addPeerDevice(): add conn_id: 1, GATT role: client
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[... cut ...]
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
Server B Service found
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 42, UUID: 2916187f-e651-4c42-9144-d91262892b22
[E][BLERemoteCharacteristic.cpp:287] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
Server B Characterstics found
We are now connected to Server B
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
RSSI B :-47
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!

```

sssuun, 2021-09-26:
Here my code for 1 ESP32 client and 1 ESP32 server
```cpp
/**
 * A BLE client example that is rich in capabilities.
 * There is a lot new capabilities implemented.
 * author unknown
 * updated by chegewara
 */

#include "BLEDevice.h"
#include "BLEScan.h"
#include "BLEAdvertisedDevice.h"

static BLEUUID serviceUUID("7cddf5af-453f-40fa-808e-37ae6ad8facd");
static BLEUUID charUUID("3383a686-a53c-42e4-a88c-e0e1de4c4bda");

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;
static BLEClient*  pClient;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
    Serial.print("data: ");
    Serial.println((char*)pData);
}

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
  }

  void onDisconnect(BLEClient* pclient) {
    connected = false;
    Serial.println("onDisconnect");
  }
};

bool connectToServer() {
    Serial.print("Forming a connection to ");
    Serial.println(myDevice->getAddress().toString().c_str());
    
    pClient  = BLEDevice::createClient();
    Serial.println(" - Created client");

    pClient->setClientCallbacks(new MyClientCallback());

    // Connect to the remove BLE Server.
    pClient->connect(myDevice);  // if you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)
    Serial.println(" - Connected to server");

    // Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    Serial.println(" - Found our service");

    // Obtain a reference to the characteristic in the service of the remote BLE server.
[... 169 lines cut ...]
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
 - Connected to server
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
 - Found our service
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 42, UUID: 3383a686-a53c-42e4-a88c-e0e1de4c4bda
[E][BLERemoteCharacteristic.cpp:287] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
 - Found our characteristic
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
The characteristic value was: Hello!
We are now connected to the BLE Server.
Setting new characteristic value to "Time since boot: 3"
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
RSSI :-49
Setting new characteristic value to "Time since boot: 4"
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
```
this code keep showing the RSSI value.

VojtechBartoska, 2022-03-30:
Hello @sssuun  , sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

closed 2022-04-20 by VojtechBartoska