espressif/arduino-esp32 #5706: Touchpad wakeup: don't need native ISR?

mahesh2000, 2021-09-26:
Hi, I'm running a **mix** of Arduino and ESP32 native API code. It works. The touchpad wakes the ESP32 from deep sleep by using `esp_sleep_enable_touchpad_wakeup()`. And then `setup()` (Arduino) is executed. I need the mix because the wake must be on `TOUCH_TRIGGER_ABOVE`, which Arduino doesn't have, and I also **need** Arduino libraries.

Therefore, does the touch ISR (`touch_pad_isr_register()`, `touch_pad_intr_enable()`) have any use at all? Since, on wakeup, `setup()` is always executed anyway and I can find the source of the wake.

Thanks!

chegewara, 2021-09-26:
Yes, you are not using touch ISR in wakeup from deep sleep, but maybe you dont know that you can use touch pads in code too.

closed 2022-01-19 by SuGlider

SuGlider, 2022-01-26:
Check #6194