espressif/arduino-esp32 #5712: ESP-PSRAM64H PSRAM not working with ESP32-S2

Lawrence-RedFern, 2021-09-28:
### Hardware:
Board:							ESP32-S2 Custom Board
Core Installation version:			2.0.0
IDE name:						Arduino IDE/Visual Studio
Flash Frequency:					80MHz
PSRAM enabled:                                    YES
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I am trying to run the CameraWebServer example on my custom ESP32-S2 board. I have an ESP-PSRAM64H acting as PSRAM on my board, but it refuses to initialise. The sketch I am using to test the PSRAM functionality can be found below. This is the verbose log when I try to run the sketch.

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x498
load:0x4004c000,len:0xa88
load:0x40050000,len:0x25a8
entry 0x4004c19c
E (91) psram: PSRAM ID read error: 0x00ffffff
[   556][W][esp32-hal-psram.c:61] psramInit(): PSRAM init failed!
VMDPR_Camera Test
psramInit
psramFound
psramFound false
```
Any ideas on where to go from here would be much appreciated. 

```
void setup() {
    Serial.begin(115200);
    Serial.println(F("CritterPic Camera Test"));

    //i2c_0.begin(SDA_0, SCL_0);

    //io_expander.begin(IOE_ADDRESS, &i2c_0);

    //io_expander.setAllPinsLowPower();

    camera_config_t config;
    config.ledc_channel = LEDC_CHANNEL_0;
    config.ledc_timer = LEDC_TIMER_0;
    config.pin_d0 = PIN_CAM_D0;
    config.pin_d1 = PIN_CAM_D1;
    config.pin_d2 = PIN_CAM_D2;
    config.pin_d3 = PIN_CAM_D3;
    config.pin_d4 = PIN_CAM_D4;
    config.pin_d5 = PIN_CAM_D5;
    config.pin_d6 = PIN_CAM_D6;
    config.pin_d7 = PIN_CAM_D7;
    config.pin_xclk = PIN_CAM_MCLK;
    config.pin_pclk = PIN_CAM_PCLK;
    config.pin_vsync = PIN_CAM_VSYNC;
    config.pin_href = PIN_CAM_HSYNC;
    config.pin_sscb_sda = PIN_CAM_SDA;
    config.pin_sscb_scl = PIN_CAM_SCL;
    config.pin_pwdn = -1;
    config.pin_reset = PIN_CAM_RST;
    config.xclk_freq_hz = 20000000;
    config.pixel_format = PIXFORMAT_JPEG;
    Serial.println(F("psramInit"));
    psramInit();
    Serial.println(F("psramFound"));
    if (psramFound()) {
        config.frame_size = FRAMESIZE_UXGA;
        config.jpeg_quality = 10;
        config.fb_count = 2;
        Serial.println(F("psramFound true"));
    }
    else {
        config.frame_size = FRAMESIZE_SVGA;
        config.jpeg_quality = 12;
        config.fb_count = 1;
        Serial.println(F("psramFound false"));
    }
```

Jason2866, 2021-09-28:
Mhh, sdkconfig has a different CS defined for PSRAM https://github.com/espressif/esp32-arduino-lib-builder/blob/master/sdkconfig.esp32s2#L415

Lawrence-RedFern, 2021-09-28:
For some reason I can't see this comment on github, only in emails!

GPIO26 is the same pin that I've used, as well as on the hardware design
guideline.

Lawrence-RedFern, 2021-09-28:
On a different board which has the same wire configuration but uses an ESP32-S2F I get the error message:

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x498
load:0x4004c000,len:0xa88
load:0x40050000,len:0x25a8
entry 0x4004c19c
E (1183) spiram: SPI SRAM error@3f780000:00000000/aaaaaaaa 

E (1183) spiram: SPI SRAM error@3f780020:00000000/aaaaaaa2 

E (1183) spiram: SPI SRAM error@3f780040:00000000/aaaaaaba 

E (1474) spiram: SPI SRAM memory test fail. 262144/262144 writes Camera Test
psramInit
psramFound
psramFound false

```

Lawrence-RedFern, 2021-09-29:
It turns out the PSRAM chips I was using were faulty. I replaced the ones on my board with one off a dev board and it worked first time.

closed 2021-09-29 by Lawrence-RedFern