espressif/arduino-esp32 #5714: Esp32 built in MQTT chunked data receive to PSRAM buffer.

zekageri, 2021-09-28:
I want to place every `event->data` on mqtt data event inside a preallocated `char *`  buffer in PSRAM.

EXCEPTION CODE:

```C
data full length: 47 data chunk length: 47
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.    
Core 1 register dump:
PC      : 0x4008ca18  PS      : 0x00060d30  A0      : 0x800e5545  A1      : 0x3ffeb780 

A2      : 0x00000000  A3      : 0x3ffe8424  A4      : 0x0000002f  A5      : 0x00000000 

A6      : 0x7974227b  A7      : 0x22657079  A8      : 0x6f64223a  A9      : 0x3ffeb6d0 

A10     : 0x0000002c  A11     : 0x00000001  A12     : 0x0000002c  A13     : 0x3ffeb770 

A14     : 0x3ffeb6f0  A15     : 0x00000008  SAR     : 0x00000008  EXCCAUSE: 0x0000001d 

EXCVADDR: 0x00000000  LBEG    : 0x4008ca11  LEND    : 0x4008ca33  LCOUNT  : 0x00000001 

ELF file SHA256: 0000000000000000

Backtrace: 0x4008ca18:0x3ffeb780 0x400e5542:0x3ffeb790 0x400e6b29:0x3ffeba10 0x4013fa81:0x3ffeba30 0x40140026:0x3ffeba50 0x4014016f:0x3ffeba90 0x40140528:0x3ffebac0 0x40090842:0x3ffebaf0
  #0  0x4008ca18:0x3ffeb780 in memcpy at /home/mak/e/p/newlib_old/newlib_xtensa-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:275
  #1  0x400e5542:0x3ffeb790 in onMqttData(esp_mqtt_event_t*) at lib/modbus-esp8266-3.0.6/src/Modbus.h:238
  #2  0x400e6b29:0x3ffeba10 in mqttEvent(esp_mqtt_event_t*) at lib/modbus-esp8266-3.0.6/src/Modbus.h:238
  #3  0x4013fa81:0x3ffeba30 in esp_mqtt_dispatch_event at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1616
  #4  0x40140026:0x3ffeba50 in deliver_publish at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1616   
  #5  0x4014016f:0x3ffeba90 in mqtt_process_receive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1616
  #6  0x40140528:0x3ffebac0 in esp_mqtt_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1616     
  #7  0x40090842:0x3ffebaf0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```
----------------------------- -------------- -----------------------------

### Hardware:
Board:							ESP-WROVER-32 ( 4mb psram ,16mb flash )
IDE name:						Platform.io
Flash Frequency:					80Mhz
PSRAM enabled:                                      yes
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
I have large and small data stream going on MQTT. I can receive like 43 length data or a full 40kb raw file.
I want to buffer every `event->data`  in the callback function, after deserialize the buffer ( because every data is a json string in my case ) and decide what to do whit it.

### Sketch:  

```C
static const inline void setupMQTT(){
    Serial.print("mqtt_setup: "); Serial.println(mqttServer);
    mqtt_cfg.event_handle   =   mqttEvent;
    mqtt_cfg.uri            =   mqttServer;
    mqtt_cfg.port           =   15902;
    mqtt_cfg.task_stack     =   10048;
    //mqtt_cfg.out_buffer_size=   10024;
    mqtt_cfg.buffer_size    =   2048;

    mqtt_cfg.username=(const char *)mqtt_user;
    mqtt_cfg.password=(const char *)mqtt_pwd;

    mqtt_client = esp_mqtt_client_init(&mqtt_cfg);
    if( esp_mqtt_client_start(mqtt_client) == ESP_OK){
        Serial.println("MQTT init success.");
    }else{
        Serial.println("Problem on mqtt INIT");
    }
}

void onMqttConnect(esp_mqtt_client_handle_t client) {
    mqttConnected = true;
    esp_mqtt_client_subscribe(client, receiveTopic, 0);
    Serial.print("subscribed to: "); Serial.println(receiveTopic);
}

size_t mqttBufferSize = 50000;
char * mqttBuffer = (char *) ps_malloc (mqttBufferSize * sizeof (char));
int mqttBufferIndex = 0;

void onMqttData(esp_mqtt_event_handle_t event) {
    boolean dataIsOk = false;
    //String topic = String(event->topic).substring(0, event->topic_len);
    //String msg = String(event->data).substring(0, event->data_len);
    // no problem with the String msgs if the data is not chunked....

    Serial.printf("\ndata full length: %d data chunk length: %d\n",
        event->total_data_len,event->data_len);

    if(event->data_len == event->total_data_len){
        // if the sizes matching, we already got a full data. Copy it to the preallocated buffer.
        dataIsOk = true;
        if(mqttBufferIndex == 0){
            // if the mqttBufferIndex == 0, that means the last data was not chunked at all. So copy to mqttBuffer
            memcpy(mqttBuffer, event->data, sizeof(event->data[0])*event->data_len);
        }
    }else{

        for (size_t i = 0; i < event->data_len; i++){
           mqttBuffer[mqttBufferIndex] = event->data[i];
           mqttBufferIndex++;
        }
    }
    if( dataIsOk ) {
        Serial.println("Got a full message");
        // mqttBuffer[mqttBufferIndex] = "\0"; <-- don't know if this is necessary yet.
        mqttBufferIndex = 0;
        SpiRamJsonDocument doc(mqttBufferSize);
        DeserializationError err = deserializeJson(doc, mqttBuffer);
        memset(mqttBuffer, 0, mqttBufferSize * sizeof (char));
        if( !err ){
          String type = doc["type"];
          // Process data in "doc"
       }
 }
```

The best thing would be to define a preallocated buffer for the mqtt data and don't worry about the remaining heap.

Thank you for your help,

zekageri, 2021-09-28:
**If i go with this approach:**

```C
void onMqttData(esp_mqtt_event_handle_t event) {
    boolean dataIsOk = false;

    if(event->data_len == event->total_data_len){
        dataIsOk = true;
        if(mqttBufferIndex == 0){
            // for loop instead of memcpy.
            for (size_t i = 0; i < event->data_len; i++){
               mqttBuffer[i] = event->data[i];
            }
        }
    }else{
        for (size_t i = 0; i < event->data_len; i++){
           mqttBuffer[mqttBufferIndex] = event->data[i];
           mqttBufferIndex++;
        }
    }
    if( dataIsOk ) {
    // deserialize and things like that...   
   }
}
```

**I got a different error:**

```C
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.    
Core 1 register dump:
PC      : 0x400e555e  PS      : 0x00060530  A0      : 0x800e6b8c  A1      : 0x3ffeb790 

A2      : 0x3ffd20d8  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffc3b78 

A6      : 0x3ffc1fd0  A7      : 0x3ffc1fd4  A8      : 0x0000007b  A9      : 0x3ffeb6f0 

A10     : 0x0000002c  A11     : 0x80a892e9  A12     : 0x80a892e9  A13     : 0x3ffc86f8 

A14     : 0x3ffeb72c  A15     : 0x3ffeb72c  SAR     : 0x00000004  EXCCAUSE: 0x0000001d 

EXCVADDR: 0x00000000  LBEG    : 0x4008c990  LEND    : 0x4008c9ac  LCOUNT  : 0xffffffff 

ELF file SHA256: 0000000000000000

Backtrace: 0x400e555e:0x3ffeb790 0x400e6b89:0x3ffeba10 0x4013fae5:0x3ffeba30 0x4014008a:0x3ffeba50 0x401401d3:0x3ffeba90 0x4014058c:0x3ffebac0 0x40090842:0x3ffebaf0
  #0  0x400e555e:0x3ffeb790 in onMqttData(esp_mqtt_event_t*) at lib/modbus-esp8266-3.0.6/src/Modbus.h:238
  #1  0x400e6b89:0x3ffeba10 in mqttEvent(esp_mqtt_event_t*) at lib/modbus-esp8266-3.0.6/src/Modbus.h:238
  #2  0x4013fae5:0x3ffeba30 in esp_mqtt_dispatch_event at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1616
  #3  0x4014008a:0x3ffeba50 in deliver_publish at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1616   
  #4  0x401401d3:0x3ffeba90 in mqtt_process_receive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1616
  #5  0x4014058c:0x3ffebac0 in esp_mqtt_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1616     
  #6  0x40090842:0x3ffebaf0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jul 29 2019 12:21:46
```

chegewara, 2021-09-28:
`char * mqttBuffer = (char *) ps_malloc (mqttBufferSize * sizeof (char));`
but, did you confirm anywhere in your code that that `mqttBuffer` is not NULL?

zekageri, 2021-10-07:
I ended up doing it with String buffer.

```
const size_t mqttJsonSize = 5000;
String mqttBuffer;
static const inline void setupMQTT(){
  mqttBuffer.reserve(mqttJsonSize);
}

void onMqttData(esp_mqtt_event_handle_t event) {
    mqttBuffer += String(event->data).substring(0, event->data_len);
    if(mqttBuffer.length() >= event->total_data_len){
        // package is ok to process...
        SpiRamJsonDocument doc(mqttJsonSize);
        DeserializationError err = deserializeJson(doc, mqttBuffer);
        mqttBuffer = "";
    }
}
```

closed 2021-10-07 by zekageri