espressif/arduino-esp32 #5715: Proposed addition to Preferences interface

ballen4705, 2021-09-28:
This is a comment/question rather than a bug report, and concerns the Preferences library:
https://github.com/espressif/arduino-esp32/tree/master/libraries/Preferences/src

I propose to add two additional methods to the Preferences class, put() and get().  By overloading, these will call the appropriate putX() or getX() method (see Preferences.h).   For example:

```
put(const char *key, int8_t value) {
      return putChar(name, value); 
}
put(const char *key, uint8_t value) {
      return putUChar(name, value); 
}
put(const char *key, int16_t value) {
      return putShort(name, value); 
}
put(const char *key, uint16_t value) {
      return putUShort(name, value); 
}
...
```
This overloading will make the library simpler to use, because a single method can be called to get and retrieve ALL data types.  This is helpful because otherwise one has to keep in mind sixteen different non-standard function names like 'putLong64' and 'getUlong64'.  If there is support for this, I will provide a patch as a pull request. 

Cheers, Bruce

igrr, 2021-09-28:
Although this is convenient, please note that relying on function overloading may introduce unexpected behavior due to integer promotion rules. The underlying NVS library requires "set" and "get" to be called with the same type, so if the argument type for "get" doesn't match the type of "set", then "get" will fail. Explicitly specifying argument type reduces the chance of accidentally calling set and get with different types.

ballen4705, 2021-09-28:
My suggested patch would provide a prototype that matches every native type.  So unless the user explicitly does a type conversion/cast, the compiler will always apply the correct function.

VojtechBartoska, 2022-03-30:
Hello @ballen4705, we will take a look on this again and check if it's applicable. Thanks for your contribution.

ballen4705, 2022-03-31:
Hi Vojtech, thanks for picking this up.  As I wrote in September last year, if there is support for this, I will provide a patch as a pull request. Please let me know.  Cheers, Bruce

closed 2025-01-16 by Parsaabasi