espressif/arduino-esp32 #5725: setScanResponseData(): esp_ble_gap_config_scan_rsp_data_raw: 258 Unknown ESP_ERR error

mbastida123, 2021-10-01:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 devkit V1
Core Installation version:			
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
IDE name:						Platform.io
Flash Frequency:					240Mhz
PSRAM enabled:                                     no
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
I have tried to test the BLE Ibeacon example but I don't see anything on NRF connect. I then enabled the logging via the platformio.ini file and saw that the code throws an error.
It seems to be the same error as here: https://github.com/espressif/arduino-esp32/issues/2853
But I have checked from the same github issue this: https://github.com/espressif/arduino-esp32/pull/3034

and start() is commented on my code. So that should be resolved, but it isn't.

### Sketch
```cpp
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by pcbreflux
*/

 
/*
   Create a BLE server that will send periodic iBeacon frames.
   The design of creating the BLE server is:
   1. Create a BLE Server
   2. Create advertising data
   3. Start advertising.
   4. wait
   5. Stop advertising.
   6. deep sleep
   
*/
#include "Arduino.h"
#include "sys/time.h"

#include "BLEDevice.h"
#include "BLEUtils.h"
#include "BLEBeacon.h"
#include "esp_sleep.h"

#define GPIO_DEEP_SLEEP_DURATION     10  // sleep x seconds and then wake up
RTC_DATA_ATTR static time_t last;        // remember last boot in RTC Memory
RTC_DATA_ATTR static uint32_t bootcount; // remember number of boots in RTC Memory

#ifdef __cplusplus
extern "C" {
#endif

uint8_t temprature_sens_read();
//uint8_t g_phyFuns;

#ifdef __cplusplus
}
#endif

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/
BLEAdvertising *pAdvertising;
struct timeval now;

#define BEACON_UUID           "8ec76ea3-6668-48da-9866-75be8bc86f4d" // UUID 1 128-Bit (may use linux tool uuidgen or random numbers via https://www.uuidgenerator.net/)

void setBeacon() {

  BLEBeacon oBeacon = BLEBeacon();
  oBeacon.setManufacturerId(0x4C00); // fake Apple 0x004C LSB (ENDIAN_CHANGE_U16!)
  oBeacon.setProximityUUID(BLEUUID(BEACON_UUID));
  oBeacon.setMajor((bootcount & 0xFFFF0000) >> 16);
  oBeacon.setMinor(bootcount&0xFFFF);
  BLEAdvertisementData oAdvertisementData = BLEAdvertisementData();
  BLEAdvertisementData oScanResponseData = BLEAdvertisementData();
  
  oAdvertisementData.setFlags(0x04); // BR_EDR_NOT_SUPPORTED 0x04
  
  std::string strServiceData = "";
  
  strServiceData += (char)26;     // Len
  strServiceData += (char)0xFF;   // Type
  strServiceData += oBeacon.getData(); 
  oAdvertisementData.addData(strServiceData);
  
  pAdvertising->setAdvertisementData(oAdvertisementData);
  pAdvertising->setScanResponseData(oScanResponseData);
  pAdvertising->setAdvertisementType(ADV_TYPE_NONCONN_IND);

}

void setup() {

    
  Serial.begin(115200);
  gettimeofday(&now, NULL);

  Serial.printf("start ESP32 %d\n",bootcount++);

  Serial.printf("deep sleep (%lds since last reset, %lds since last boot)\n",now.tv_sec,now.tv_sec-last);

  last = now.tv_sec;
  
  // Create the BLE Device
  BLEDevice::init("");

  // Create the BLE Server
  // BLEServer *pServer = BLEDevice::createServer(); // <-- no longer required to instantiate BLEServer, less flash and ram usage

  pAdvertising = BLEDevice::getAdvertising();
  
  setBeacon();
   // Start advertising
  pAdvertising->start();
  Serial.println("Advertizing started...");
  delay(100);
  pAdvertising->stop();
  Serial.printf("enter deep sleep\n");
  esp_deep_sleep(1000000LL * GPIO_DEEP_SLEEP_DURATION);
  Serial.printf("in deep sleep\n");
}

void loop() {
}
```

### Debug Messages:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
start ESP32 0
deep sleep (0s since last reset, 0s since last boot)
[I][BLEDevice.cpp:569] getAdvertising(): create advertising
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[V][BLEAdvertising.cpp:153] setAdvertisementData(): >> setAdvertisementData
[V][BLEAdvertising.cpp:161] setAdvertisementData(): << setAdvertisementData
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 4 0x04
[V][BLEAdvertising.cpp:170] setScanResponseData(): >> setScanResponseData
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[E][BLEAdvertising.cpp:175] setScanResponseData(): esp_ble_gap_config_scan_rsp_data_raw: 258 Unknown ESP_ERR error
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEAdvertising.cpp:178] setScanResponseData(): << setScanResponseData
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[V][BLEAdvertising.cpp:187] start(): >> start: customAdvData: 1, customScanResponseData: 1
[D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 4]
[D][BLEAdvertising.cpp:206] start(): - no services advertised
[V][BLEAdvertising.cpp:252] start(): << start
Advertizing started...
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 6 0x06
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 6]
[V][BLEAdvertising.cpp:262] stop(): >> stop
[V][BLEAdvertising.cpp:268] stop(): << stop
enter deep sleep
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 17 0x11
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 17]
[I][BLEAdvertising.cpp:522] handleGAPEvent(): STOP advertising
```

chegewara, 2021-10-01:
Some arguments are invalid. Please print `oScanResponseData`.

mbastida123, 2021-10-01:
> Some arguments are invalid. Please print `oScanResponseData`.

I put:
`log_d("oScanResponseData: %s\n",oScanResponseData.getPayload().c_str());`

And get:

`[D][main.cpp:71] setBeacon(): oScanResponseData: 

␀ing`

chegewara, 2021-10-01:
Not that way. You have to print byte values, not a string.

mbastida123, 2021-10-01:
> Not that way. You have to print byte values, not a string.

Like this?:
```
Serial.printf("OscanResponse: ");
  char const *c=oScanResponseData.getPayload().c_str();
  for (int i=0; i<sizeof(c);i++)
  {
    Serial.printf("%X ", c[i]);
  }
  
  Serial.println();
```

`OscanResponse: 0 0 0 0 `

chegewara, 2021-10-01:
Yes, like this.
This is incorrect payload, because 1st byte is length of some data, 2nd is type and then payload.

mbastida123, 2021-10-03:
> Yes, like this. This is incorrect payload, because 1st byte is length of some data, 2nd is type and then payload.

This is the example taken out of the BLE library. Does this mean that there's a bug or something wrong on the example or did I do something wrong?

mbastida123, 2021-10-05:
@chegewara any idea on how I could continue troubleshooting?

chegewara, 2021-10-05:
Sorry, i dont have time to check  this example/code right now, but i would just start with removing this line:
` pAdvertising->setScanResponseData(oScanResponseData);`

mbastida123, 2021-10-05:
> Sorry, i dont have time to check this example/code right now, but i would just start with removing this line: ` pAdvertising->setScanResponseData(oScanResponseData);`

Ok, no problem.

I removed the line. I paste the output just in case someone else can help:
```
rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
start ESP32 8
deep sleep (90s since last reset, 12s since last boot)
[I][BLEDevice.cpp:569] getAdvertising(): create advertising
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[V][BLEAdvertising.cpp:153] setAdvertisementData(): >> setAdvertisementData
[V][BLEAdvertising.cpp:161] setAdvertisementData(): << setAdvertisementData
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 4 0x04
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEAdvertising.cpp:187] start(): >> start: customAdvData: 1, customScanResponseData: 0
[D][BLEAdvertising.cpp:206] start(): - no services adverti[D][BLEAdvertising.cpp:206] start(): - no services advertised
[D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 4]
[V][BLEAdvertising.cpp:252] start(): << start
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 1 0x01
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 1]
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 6 0x06
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 6]
[V][BLEAdvertising.cpp:262] stop(): >> stop
[V][BLEAdvertising.cpp:268] stop(): << stop
enter deep sleep
```

VojtechBartoska, 2022-03-30:
Hello @mbastida123, is this still relevant?

mbastida123, 2022-03-30:
> Hello @mbastida123, is this still relevant?

Hello @VojtechBartoska,

Yes it is still relevant. I haven't checked since then (I gave up on using Beacon with ESP32 for other reasons). But it would be nice to have it working :)

Will try again just in case some release has fixed the problem.

mbastida123, 2022-03-30:
> Hello @mbastida123, is this still relevant?

> > Hello @mbastida123, is this still relevant?
> 
> Hello @VojtechBartoska,
> 
> Yes it is still relevant. I haven't checked since then (I gave up on using Beacon with ESP32 for other reasons). But it would be nice to have it working :)
> 
> Will try again just in case some release has fixed the problem.
I have just tried and I still get the same error... :(

VojtechBartoska, 2022-03-31:
Adding it to the Roadmap, thanks for testing.

VojtechBartoska, 2022-04-08:
@mbastida123 Are you able to retest your issue on v2.0.3-RC1? Thanks!

italocjs, 2022-05-02:
I have the same issue, im looking for v2.0.3-RC1 but cant find it.  can you maybe send me a link?

mbastida123, 2022-05-02:
> I have the same issue, im looking for v2.0.3-RC1 but cant find it. can you maybe send me a link?
> 
> i had the same issue using the espressif example [https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_iBeacon/BLE_iBeacon.ino](url). This other one its transmitting, but it still reports an error [https://www.filipeflop.com/blog/beacon-com-esp32/](url)

There you have it: https://github.com/espressif/arduino-esp32/tree/2.0.3-RC1

@VojtechBartoska I have tried but I'm struggling with PlatformIO and using a certain framework from github. Will try again and get back to you.

VojtechBartoska, 2022-05-03:
@mbastida123 Thanks!

@italocjs Please take a look here, there is everything explained step by step: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

mbastida123, 2022-05-13:
@VojtechBartoska I think the problem still persists:
```
rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
[␀␘␂␂␂��m␑um����2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
start ESP32 3
deep sleep (33s since last reset, 11s since last boot)
[   668][I][BLEDevice.cpp:577] getAdvertising(): create advertising
[   668][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[   669][V][BLEAdvertising.cpp:153] setAdvertisementData(): >> setAdvertisementData
[   676][V][BLEAdvertising.cpp:161] setAdvertisementData(): << setAdvertisementData
[   680][V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 4 0x04
[   684][V][BLEAdvertising.cpp:170] setScanResponseData(): >> setScanResponseData
[   680][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
**[   699][E][BLEAdvertising.cpp:175] setScanResponseData(): esp_ble_gap_config_scan_rsp_data_raw: 258 Unknown ESP_ERR error**
[   707][V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[   718][V][BLEAdvertising.cpp:178] setScanResponseData(): << setScanResponseData
[   725][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[   732][V][BLEAdvertising.cpp:187] start(): >> start: customAdvData: 1, customScanResponseData: 1
[   738][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 4]
[   747][D][BLEAdvertising.cpp:206] start(): - no services advertised
[   760][V][BLEAdvertising.cpp:252] start(): << start
Advertizing started...
[   773][V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 6 0x06
[   773][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[   783][V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[   791][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[   796][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 6]
[   876][V][BLEAdvertising.cpp:262] stop(): >> stop
[   876][V][BLEAdvertising.cpp:268] stop(): << stop
enter deep sleep
```

patrawi, 2023-04-02:
Is there anyone know how to solve this ? It still persist.

closed 2025-01-16 by Parsaabasi